evcc-io/evcc #13835: ned.nl (Nationaal Energie Dashboard) and co2monitor.nl integrations

gijs007, 2024-05-09:
The Dutch energy network provides free information through an API on: https://ned.nl/nl (free registration required)
It includes predictions for solar and wind energy for the next 7 days. 

Additionally https://co2monitor.nl/ gives data on the current CO2 emissions.
They mention that CO2 predictions may also be available upon request, if there is a valid need.

It would be nice if we can use the current CO2 emissions for the statistics. Additionally NED's solar and wind forecasts could be used to optimize charge planning.

andig, 2024-05-09:
Happy to accept a PR for a new template. See https://github.com/evcc-io/evcc/blob/master/templates/definition/tariff/energy-charts-api.yaml for an example how to do this.

mdkeil, 2024-05-10:
The request relates to another CO2 provider and an energy forecast provider, which is currently not supported in evcc.
(not my buisiness ;D)

andig, 2024-05-10:
This can already be implemented using the custom tariff. Closing here- please feel free to open a PR with a suggested implementation.

closed 2024-05-10 by andig