evcc-io/evcc #13836: Incorrect spellings in the Turkish translation

aerucu, 2024-05-09:
### Describe the bug

1. In the Turkish translation, the percentage sign was written after the value in various places. However, the percentage sign must be written before the value.

[image]
[image]

2. `[footer.community]greenShareSub1` In Turkish, “solar, and battery storage” must come before “Power provided by”. Otherwise the sentence is not correct.

[image]

### Steps to reproduce

1. Set language to Türkçe
2. Click on savings button in the footer.
3.
...

### Configuration details

```yaml
Not relevant
```

### Log details

```text
Not relevant
```

### What type of operating system are you running?

Linux

### Version

v0.126.2

andig, 2024-05-09:
Please fix directly at https://hosted.weblate.org/projects/evcc/evcc/!

andig, 2024-05-09:
Ah, sorry- it's not that easy. @naltatis how could we fix this?

naltatis, 2024-05-09:
@andig yes, we've previously [discussed this in Slack](https://evccgroup.slack.com/archives/C01321PUJAD/p1714978630567399?thread_ts=1714928263.519539&cid=C01321PUJAD). This is more of a general issue (I assume not only for Turkish and not only on this screen) where we have to change UI structure/layout based on language.

@aerucu in your screenshot you only flipped unit and number for the percentage value (`% 88,6`). We have other places in the UI. I assume these have to be adjusted as well, right?

[image]

I'd suggest we collect all these places before making the necessary adjustments. I'll make the title of this issue a bit more general to indicate that.

naltatis, 2024-05-09:
> 2. `[footer.community]greenShareSub1` In Turkish, “solar, and battery storage” must come before “Power provided by”. Otherwise the sentence is not correct.

@aerucu I don't see an issue here. In this place, we've named the placeholders pretty generic deliberately `greenShareSub1` and `greenShareSub2`. The contents of both fields do not have to match the source translation order as long as the message stays correct. Feel free to flip the content if it makes more sense.

aerucu, 2024-05-09:
> @aerucu in your screenshot you only flipped unit and number for the percentage value (`% 88,6`). We have other places in the UI. I assume these have to be adjusted as well, right?

@naltatis I have already flipped all the places that could be flipped in the [Weblate](https://hosted.weblate.org/projects/evcc/evcc/).

aerucu, 2024-05-09:
> @aerucu I don't see an issue here. In this place, we've named the placeholders pretty generic deliberately `greenShareSub1` and `greenShareSub2`. The contents of both fields do not have to match the [...]

In Turkish, the content of the `[footer.community]greenShareSub2` placeholder must be displayed first and then the content of the `[footer.community]greenShareSub1` placeholder. Otherwise the sentence is not correct.

aerucu, 2024-05-09:
Done!

StefanSchoof, 2024-05-10:
Was curious what `toLocaleString` is doing with turkish percent and it works:

```javascript
(3/4).toLocaleString("tr", {style: "percent"})
```

will produce "%75".

PR #13880 (2024-05-17): UI i18n: percentage formatting
fixes #8510, fixes #13836

- format percentage values according to language-specific conventions
- uses `Intl.NumberFormat` for formatting
- uses DIN 5008 for German (see #8510)

overview for all existing languages:

[image]

**English**

[image]

**German**

[image]

**Arabic**

[image]
[image]
[image]
[image]
<img width="740" alt="Bildschirmfoto 2024-05-12 um 16 01 39" src="https://github.com/evcc-io/evcc/assets/152287/17a279f3-9c74-49
closed 2024-05-17 by naltatis