evcc-io/evcc #13839: easee charge Phasenswitch von 1 nach 3 geht nicht

pasqualito79, 2024-05-09:
### Describe the bug

Hallo Leute! Folgendes bei Wechseln von Modus PV nach Schnell (also von 1 Phase nach 3 Phase) festgestellt:

Einstellhalle: charger logic error: phases mismatch (got 1, expected 3)

umgekehrt funktioniert es, also von 3 Phasen auf 1 Phase runterregeln geht automatisch.

Was funktioniert ist, wenn man in der easee App die Ladung pausiert, dann schaltet evcc beim nächsten Intervall auf 3-phasig um und lädt.

Kann es sein, dass der easee Equalizer die automatische Umschaltung während dem Laden unterbindet?
Hier steht etwas darüber, aber eigentlich sollte das nur im "solar mode" relevant sein, oder? Denn der Equalizer bei uns ist nur fürs "normale" Lastenmanagement zwischen den Chargern. 

[](https://support.easee.com/hc/en-gb/articles/14607272585361-Why-is-my-charger-only-charging-6A-with-the-Solar-mode-enabled)

Hier der Abschnitt: 
_To protect the relays in the charger, the charging session will complete on single phase and not revert to three phase. The charging session will be completed on single phase regardless of whether, during the charging session, there will be sufficient capacity on all phases to charge with three phases. If the capacity has increased sufficiently in the system again (for example, solar production has increased) and you want to switch from single-phase to three-phase, you must stop the current charging session and restart charging in the app. If charging on solar power is first started on single phase, the Equalizer will not automatically activate three-phase charging if the production increases to over 6A on all phases._

Wenn das trotzdem das Verhalten bei mir erklärt, wie wäre den Einbau einer Option "Pause bei Phasenwechsel"?

### Steps to reproduce

1. Wechsel von PV Modus nach Schnell um von 1 Phase nach 3 Phase umzuschalten
2. Fehler in Log
3.
...

### Configuration details

```yaml
chargers:
  - name: easee_wallbox
    type: template
    template: easee
    user: ***
    password: ***
    charger: ***
    timeout: 20s # optional
    authorize: #

in UI Ladestrom automatischer Wechsel aktiv, 6A mind, 16A max.
```

### Log details

```text
[site ] DEBUG 2024/05/09 18:01:59 site power: 2829W
[lp-1 ] DEBUG 2024/05/09 18:01:59 detected active phases: 1p
[lp-1 ] DEBUG 2024/05/09 18:01:59 charge total import: 2328.356kWh
[lp-1 ] DEBUG 2024/05/09 18:01:59 charger status: C
[lp-1 ] WARN 2024/05/09 18:01:59 charger logic error: phases mismatch (got 1, expected 3)
[easee ] TRACE 2024/05/09 18:01:59 GET https://api.easee.com/api/sites/644946/circuits/608662/settings
[easee ] TRACE 2024/05/09 18:01:59 {"allowOfflineMaxCircuitCurrent":true,"equalizerId":"QHVTR93Q","masterChargerId":"ECAJ39MJ","useDynamicMaster":false,"dynamicCircuitCurrentP1":16.0,"dynamicCircuitCurrentP2":16.0,"dynamicCircuitCurrentP3":16.0,"enableIdleCurrent":true,"maxCircuitCurrentP1":16.0,"maxCircuitCurrentP2":16.0,"maxCircuitCurrentP3":16.0,"offlineMaxCircuitCurrentP1":0,"offlineMaxCircuitCurrentP2":0,"offlineMaxCircuitCurrentP3":0}
[easee ] TRACE 2024/05/09 18:01:59 POST https://api.easee.com/api/sites/644946/circuits/608662/settings
[easee ] TRACE 2024/05/09 18:01:59 {"dynamicCircuitCurrentP1":16,"dynamicCircuitCurrentP2":16,"dynamicCircuitCurrentP3":16}
[lp-1 ] DEBUG 2024/05/09 18:01:59 switched phases: 3p
[easee ] TRACE 2024/05/09 18:02:26 ProductUpdate ECAJ39MJ: (2024-05-09 16:02:25 +0000 UTC) CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L1 16
[easee ] TRACE 2024/05/09 18:02:26 ProductUpdate ECAJ39MJ: (2024-05-09 16:02:26 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L1 15.107000350952148
[easee ] TRACE 2024/05/09 18:02:27 ProductUpdate ECAJ39MJ: (2024-05-09 16:02:26 +0000 UTC) IN_CURRENT_T3 15.104999542236328
[site ] DEBUG 2024/05/09 18:02:29 ----
[lp-1 ] DEBUG 2024/05/09 18:02:29 charge power: 3554W
[lp-1 ] DEBUG 2024/05/09 18:02:29 charge currents: [15.1 0.021 0.012]A
```

### What type of operating system are you running?

Docker container

### Version

0.126.2

andig, 2024-05-09:
Da stimmt schon vorher etwas nicht, Log dazu fehlt

closed 2024-05-09 by andig

pasqualito79, 2024-05-09:
Kannst du mir sagen nach welchen Keywords ich suchen muss? Im Log finde ich keine fehler, aber hier noch einen grösseren Abschnitt genau beim Wechsel von PV nach Schnell:
[lp-1 ] DEBUG 2024/05/09 19:29:29 detected active phases: 1p
[lp-1 ] DEBUG 2024/05/09 19:29:29 charge total import: 2331.129kWh
[lp-1 ] DEBUG 2024/05/09 19:29:29 charger status: C
[lp-1 ] DEBUG 2024/05/09 19:29:29 pv charge current: 1.86A = 6A + -4.14A (953W @ 1p)
[lp-1 ] DEBUG 2024/05/09 19:29:56 set charge mode: now
[site ] DEBUG 2024/05/09 19:29:56 ----
[easee ] TRACE 2024/05/09 19:29:56 POST https://api.easee.com/api/chargers/ECAJ39MJ/settings
[easee ] TRACE 2024/05/09 19:29:57 {"smartCharging":false} -- [{"device":"ECAJ39MJ","commandId":11,"ticks":638508725971143480}]
[easee ] TRACE 2024/05/09 19:29:58 CommandResponse ECAJ39MJ: {SerialNumber:ECAJ39MJ ID:11 Timestamp:2024-05-09 17:29:57.114348 +0000 UTC DeliveredAt:2024-05-09 17:29:57.77 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638508725971143480}
[lp-1 ] DEBUG 2024/05/09 19:29:58 charge power: 1296W
[lp-1 ] DEBUG 2024/05/09 19:29:58 charge currents: [5.5 0.016 0.012]A
[modbus] TRACE 2024/05/09 19:29:58 modbus: send 81 ce 00 00 00 06 01 04 13 98 00 02
[modbus] TRACE 2024/05/09 19:29:58 modbus: recv 81 ce 00 00 00 07 01 04 04 03 86 00 00
[modbus] TRACE 2024/05/09 19:29:58 modbus: send 81 cf 00 00 00 06 01 04 32 ca 00 02
[modbus] TRACE 2024/05/09 19:29:58 modbus: recv 81 cf 00 00 00 07 01 04 04 79 b6 00 02
[site ] DEBUG 2024/05/09 19:29:58 pv power: 902W
[modbus] TRACE 2024/05/09 19:29:58 modbus: send 81 d0 00 00 00 06 01 04 32 dd 00 01
[modbus] TRACE 2024/05/09 19:29:58 modbus: recv 81 d0 00 00 00 05 01 04 02 03 70
[modbus] TRACE 2024/05/09 19:29:58 modbus: send 81 d1 00 00 00 06 01 04 32 c8 00 01
[modbus] TRACE 2024/05/09 19:29:58 modbus: recv 81 d1 00 00 00 05 01 04 02 00 0d
[modbus] TRACE 2024/05/09 19:29:58 modbus: send 81 d2 00 00 00 06 01 04 32 c8 00 01
[modbus] TRACE 2024/05/09 19:29:58 modbus: recv 81 d2 00 00 00 05 01 04 02 00 0d
[modbus] TRACE 2024/05/09 19:29:58 modbus: send 81 d3 00 00 00 06 01 04 32 e2 00 02
[modbus] TRACE 2024/05/09 19:29:58 modbus: recv 81 d3 00 00 00 07 01 04 04 8d 11 00 00
[modbus] TRACE 2024/05/09 19:29:58 modbus: send 81 d4 00 00 00 06 01 04 32 de 00 01
[modbus] TRACE 2024/05/09 19:29:58 modbus: recv 81 d4 00 00 00 05 01 04 02 03 a8
[site ] DEBUG 2024/05/09 19:29:58 battery soc: 94%
[site ] DEBUG 2024/05/09 19:29:58 battery power: 880W
[modbus] TRACE 2024/05/09 19:29:58 modbus: send 81 d5 00 00 00 06 01 04 32 d1 00 02
[modbus] TRACE 2024/05/09 19:29:58 modbus: recv 81 d5 00 00 00 07 01 04 04 00 00 00 00
[site ] DEBUG 2024/05/09 19:29:58 grid meter: -0W
[modbus] TRACE 2024/05/09 19:29:58 modbus: send 81 d6 00 00 00 06 01 04 32 ec 00 02
[modbus] TRACE 2024/05/09 19:29:58 modbus: recv 81 d6 00 00 00 07 01 04 04 6e a8 00 01
[site ] DEBUG 2024/05/09 19:29:58 site power: 980W
[lp-1 ] DEBUG 2024/05/09 19:29:58 detected active phases: 1p
[lp-1 ] DEBUG 2024/05/09 19:29:58 charge total import: 2331.129kWh
[lp-1 ] DEBUG 2024/05/09 19:29:58 charger status: C
[easee ] TRACE 2024/05/09 19:29:58 GET https://api.easee.com/api/sites/644946/circuits/608662/settings
[easee ] TRACE 2024/05/09 19:29:59 {"allowOfflineMaxCircuitCurrent":true,"equalizerId":"QHVTR93Q","masterChargerId":"ECAJ39MJ","useDynamicMaster":false,"dynamicCircuitCurrentP1":16.0,"dynamicCircuitCurrentP2":16.0,"dynamicCircuitCurrentP3":16.0,"enableIdleCurrent":true,"maxCircuitCurrentP1":16.0,"maxCircuitCurrentP2":16.0,"maxCircuitCurrentP3":16.0,"offlineMaxCircuitCurrentP1":0,"offlineMaxCircuitCurrentP2":0,"offlineMaxCircuitCurrentP3":0}
[easee ] TRACE 2024/05/09 19:29:59 POST https://api.easee.com/api/sites/644946/circuits/608662/settings
[easee ] TRACE 2024/05/09 19:29:59 {"dynamicCircuitCurrentP1":16,"dynamicCircuitCurrentP2":16,"dynamicCircuitCurrentP3":16}
[lp-1 ] DEBUG 2024/05/09 19:29:59 switched phases: 3p
[easee ] TRACE 2024/05/09 19:29:59 POST https://api.easee.com/api/chargers/ECAJ39MJ/settings
[easee ] TRACE 2024/05/09 19:29:59 {"dynamicChargerCurrent":16} -- [{"device":"ECAJ39MJ","commandId":48,"ticks":638508725992320000}]
[easee ] TRACE 2024/05/09 19:29:59 ProductUpdate ECAJ39MJ: (2024-05-09 17:29:57 +0000 UTC) LEDMODE 24
[easee ] TRACE 2024/05/09 19:29:59 ProductUpdate ECAJ39MJ: (2024-05-09 17:29:57 +0000 UTC) SMART_CHARGING false
[easee ] TRACE 2024/05/09 19:29:59 ProductUpdate ECAJ39MJ: (2024-05-09 17:29:57 +0000 UTC) SMART_CHARGING false
[easee ] TRACE 2024/05/09 19:30:00 ProductUpdate ECAJ39MJ: (2024-05-09 17:29:59 +0000 UTC) DYNAMIC_CHARGER_CURRENT 16
[easee ] TRACE 2024/05/09 19:30:00 CommandResponse ECAJ39MJ: {SerialNumber:ECAJ39MJ ID:48 Timestamp:2024-05-09 17:29:59.232 +0000 UTC DeliveredAt:2024-05-09 17:29:59.3 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638508725992320000}
[lp-1 ] DEBUG 2024/05/09 19:30:00 max charge current: 16A
[site ] DEBUG 2024/05/09 19:30:00 ----
[lp-1 ] DEBUG 2024/05/09 19:30:00 charge power: 1296W
[lp-1 ] DEBUG 2024/05/09 19:30:00 charge currents: [5.5 0.016 0.012]A
[modbus] TRACE 2024/05/09 19:30:00 modbus: send 81 d7 00 00 00 06 01 04 13 98 00 02
[modbus] TRACE 2024/05/09 19:30:00 modbus: recv 81 d7 00 00 00 07 01 04 04 03 81 00 00
[modbus] TRACE 2024/05/09 19:30:00 modbus: send 81 d8 00 00 00 06 01 04 32 ca 00 02
[modbus] TRACE 2024/05/09 19:30:00 modbus: recv 81 d8 00 00 00 07 01 04 04 79 b6 00 02
[site ] DEBUG 2024/05/09 19:30:00 pv power: 897W
[... cut ...] uitCurrentP3":16.0,"enableIdleCurrent":true,"maxCircuitCurrentP1":16.0,"maxCircuitCurrentP2":16.0,"maxCircuitCurrentP3":16.0,"offlineMaxCircuitCurrentP1":0,"offlineMaxCircuitCurrentP2":0,"offlineMaxCircuitCurrentP3":0}
[easee ] TRACE 2024/05/09 19:30:59 POST https://api.easee.com/api/sites/644946/circuits/608662/settings
[easee ] TRACE 2024/05/09 19:30:59 {"dynamicCircuitCurrentP1":16,"dynamicCircuitCurrentP2":16,"dynamicCircuitCurrentP3":16}
[lp-1 ] DEBUG 2024/05/09 19:30:59 switched phases: 3p

skakwy, 2024-05-14:
Problem besteht immer noch.

skakwy, 2024-05-16:
> Was funktioniert ist, wenn man in der easee App die Ladung pausiert, dann schaltet evcc beim nächsten Intervall auf 3-phasig um und lädt.

> Kann es sein, dass der easee Equalizer die automatische Umschaltung während dem Laden unterbindet? Hier steht etwas darüber, aber eigentlich sollte das nur im "solar mode" relevant sein, oder? Denn [...]

Denke, nicht, dass das am Equalizer liegt, sonst würde der Wechsel zwischen 1 und 3 phasig per schnell laden auch nicht funktionieren. Aus und Anschalten funktioniert tatsächlich, jedoch nur dann, wenn dann auch ausreichend pv Überschuss zur Verfügung steht, sodass direkt mit drei Phasen geladen werden kann. Sonst wird der Timer zum Wechseln von 1p zu 3p lediglich neu gestartet. Da, wie gesagt, das Wechseln der Phasen an sich kein Problem ist, liegt das Problem wohl nicht unbedingt am charger.

pasqualito79, 2024-05-16:
Schreibst du das, weil du das gleiche Problem hast? Denn bei mir ists ja so, wie oben beschrieben auch beim Wechsel von PV nach Schnell, wenn er dann von 1P nach 3P wechseln müsste. Oder ganz einfach gesagt, er kann nie während dem Laden ohne Pause von 1 Phase nach 3 Phase umschalten. Die einzige Möglichkeit ist eben zu pausieren.

skakwy, 2024-05-16:
Der Wechsel von PV auf schnell hab ich jetzt nicht ausprobiert. Lediglich der Phasen wechsel bei schnell laden, da funktioniert es bei mir. Aber wie bei dir hab ich das Problem, wenn ich den PV Modus benutze, ohne Pause wechselt er nicht auf 3 Phasen.

andig, 2024-05-17:
Das kann kein Massenproblem sein- sonst würden sich Dutzende Anwender melden. Bleibt also die Frage, was bei Euch anders ist. Vielleicht liegts am Equilizer? Auf jeden Fall sollte der Easee Support Auskunft geben können.