evcc-io/evcc #13845: Ford wird nicht mehr erkannt / bei Ford blockiert? #13830

djrepose, 2024-05-10:
### Describe the bug

Hallo

seit vorgestern wird mein Ford nicht mehr erkannt. Debug-Log:

[ford ] TRACE 2024/05/09 10:32:39 POST **https://login.ford.com**/......
[ford ] TRACE 2024/05/09 10:32:39 .....Access Denied You don't have permission to access ... on this server
[main ] ERROR 2024/05/09 10:32:39 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed

Habe Upgrade auf aktuellste evcc Version gemacht, selbes Problem. Aber es sieht ja so aus als ob Ford mich da sperrt oder eine neue API Adresse hat? Und ja, ich kann mich Online im Ford Konto immer noch normal anmelden, die Zugangsdaten stimmen also.

Es scheint tatsächlich ein Ford API Problem zu sein, siehe hier:
https://www.reddit.com/r/homeassistant/comments/1cn2b3r/fordpass_integration_broken_again/
https://github.com/itchannel/fordpass-ha/issues/488

### Steps to reproduce

**sudo evcc -l debug vehicle
[main  ] INFO 2024/05/10 07:38:40 evcc 0.126.2 (2254fedd)
[main  ] INFO 2024/05/10 07:38:40 using config file: /etc/evcc.yaml
[db    ] INFO 2024/05/10 07:38:40 using sqlite database: /root/.evcc/evcc.db
[main  ] ERROR 2024/05/10 07:38:41 creating vehicle ev3 failed: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed
Capacity: 68.0kWh
Features: [Offline Retryable]
**

### Configuration details

```yaml
vehicles:
- type: template
  template: ford 
  title: Mustang Mach E  
  icon: car  
  user: XXXXXXXXXXXXXXXXXXX  
  password: XXXXXXXXXXXXXXXXXX 
  vin: XXXXXXXXXXXXXX  
  capacity: 68  
  mode: pv   
  name: ev3
```

### Log details

```text
sudo evcc -l debug vehicle
[main  ] INFO 2024/05/10 07:38:40 evcc 0.126.2 (2254fedd)
[main  ] INFO 2024/05/10 07:38:40 using config file: /etc/evcc.yaml
[db    ] INFO 2024/05/10 07:38:40 using sqlite database: /root/.evcc/evcc.db
[main  ] ERROR 2024/05/10 07:38:41 creating vehicle ev3 failed: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed
Capacity: 68.0kWh
Features: [Offline Retryable]
```

### What type of operating system are you running?

Linux

### Version

 evcc version 0.126.2 (2254fedd)

andig, 2024-05-10:
Gibts ein anderes Open Source Projekt bei dem der Login noch funktioniert? Welches?

djrepose, 2024-05-10:
leider joch keinen weiteren gefunden, wenn ich was finde melde ichs. 

gruss

andig, 2024-05-10:
Ohne können wir leider nichts tun… daher erstmal wontfix.

TA2k, 2024-05-10:
ford.com auf ford.de umstellen

beathoch, 2024-05-10:
> login.ford.com auf login.ford.de umstellen

wie ist das gemeint? lösst dass das Problem und wie kann man das auf login.ford.de umstellen?

Max112233445566, 2024-05-10:
> Gibts ein anderes Open Source Projekt bei dem der Login noch funktioniert? Welches?

Fordpass funktioniert mit der deutschen Seite:
https://login.ford.de/4566605f-43a7-400a-946e-89cc9fdb0bd7/B2C_1A_SignInSignUp_DE-DE/SelfAsserted?tx={transId}&p=B2C_1A_SignInSignUp_de-DE

beathoch, 2024-05-10:
Bei mir die selbe Meldung: 

[main ] ERROR 2024/05/10 23:46:46 creating vehicle Mustang_Mach-E failed: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed

Was ich nicht verstehe ist, sowohl mein Home Assistant hat Zugriff auf meinen Account und ich kann mich auf der Ford Webseite auch problemlos anmelden, in meinem Fall login.ford.ch Mit HA kann ich das Fahrzeug wie gewohnt bedienen, aber in evcc kommt diese Fehlermeldung.

ifa8537, 2024-05-11:
Danke fürs erfassen als Bug. koehtop hat in meinem [Diskussionsbeitrag](https://github.com/evcc-io/evcc/discussions/13830) einen Link zugesandt. Falls das hilft:

> Hier sind die neuen Endpunkte, je nach locale:
> 
> https://github.com/itchannel/fordpass-ha/blob/master/custom_components/fordpass/fordpass_new.py

Sassar88, 2024-05-14:
Ich hab heute auf die neuste EVCC Version geupdatet und habe das gleiche Problem

```
[main  ] INFO 2024/05/14 15:40:19 evcc 0.126.2
[main  ] INFO 2024/05/14 15:40:19 using config file: /etc/evcc.yaml
[db    ] INFO 2024/05/14 15:40:19 using sqlite database: /home/xxx/.evcc/evcc.db
[main  ] ERROR 2024/05/14 15:40:21 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed
```

Chris591, 2024-05-14:
Hab heute mal die Nightly getestet aber leider das gleiche Problem.
```
[main  ] INFO 2024/05/14 23:27:36 evcc 0.126.2 (c5330fbb)
...
[main  ] ERROR 2024/05/14 23:27:37 creating vehicle Mustang failed: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed
.Soc:        vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed
Capacity:    91.0kWh
Features:    [Offline Retryable]
```

closed 2024-05-16 by andig

beathoch, 2024-05-16:
Hallo zusammen, hat jemand eine Idee wie das Problem zu beheben ist? ich lese was von ford.go Datei welche man anpassen soll. Aber wo findet man diese Dateien? Ich habe evcc in Home Assistant installiert und HA kann auch problemlos auf den Account Zugriefen, respektive das Fahrzeug lässt sich problemlos bedienen. Nur evcc funktioniert nicht.

Und warum bekomme ich keine Antworten auf meine Fragen? beschreibe ich das Problem in einem falschen Issue Ticket? muss ich ein neues eröffnen?

Es wäre wirklich toll wenn man dieses Problem endlich gelöst bekommen würde.

Danke

Max112233445566, 2024-05-16:
@beathoch
Wir müsen warten, bis das Fix in die neue Version aufgenommen wird. HA kann leider nicht mit Nightly Build Versionen zusammen arbeiten. Da die aktuelle Version 0.126.2 bereits 2 Wochen alt ist, nehme ich an, es kommt bald die neue Version.

beathoch, 2024-05-16:
> @beathoch Wir müsen warten, bis das Fix in die neue Version aufgenommen wird. HA kann leider nicht mit Nightly Build Versionen zusammen arbeiten. Da die aktuelle Version 0.126.2 bereits 2 Wochen alt ist, nehme ich an, es kommt bald die neue Version.

Vielen Dank! jetzt weiss ich woran ich bin. Dann kann ich aufhören nach möglichen Bugfixes zu suchen. Dann ist lediglich etwas Geduld gefragt.

Chris591, 2024-05-17:
now it is brocken again - also the Homeassistant plugin isn't working for me

LarsSche, 2024-05-17:
Home Assistant evcc addon is working fine on my site but my ford login isn't working either.
creating vehicle mach-e failed: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed

Chris591, 2024-05-18:
I guess only as long as you don't restart the AddIn or HomeAssistant - that why we see so many inconsistent responses.
as the refresh Token is valid for 90 day and if you refresh in that time you get also a new refresh Token - so as long as you don't restart it's working.

ElementFlo, 2024-05-26:
Hey,
I had to adjust some smal things in the config and did an restart evcc. Now i got this error.. 5mins before it worked. 

Is there some workaround i can get it back to work?

Sassar88, 2024-06-01:
I have upgrade today to the newest version, but problem is still existing

dremde, 2024-08-07:
Hallo zusammen,
leider werde ich aus den ganzen Infos nicht schlau.
Ich habe auch das Problem, das der Ford nicht mehr erkannt wird:
[main  ] INFO 2024/08/07 18:17:39 evcc 0.129.0
[main  ] INFO 2024/08/07 18:17:39 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/07 18:17:39 using sqlite database: /root/.evcc/evcc.db
[main  ] ERROR 2024/08/07 18:17:41 creating vehicle my_car failed: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed
my_car
------
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed
Capacity: 14.0kWh
Features: [Offline Retryable]

Update von evcc habe ich durchgeführt und auch schon mehrfach neu gestartet.
Im Browser habe ich mich angemeldet um zu schauen ob es eine AGB oder ähnliches zu bestätigen gibt, Fehlanzeige.

Kann mir da jemand helfen, werden mehr Infos benötigt ?

Sassar88, 2024-08-07:
Ich hatte das gleiche problem. Ich musste einen neuen Access/ refresh token erzeugen, danach ging es

LarsSche, 2024-08-07:
> Hallo zusammen,
>
> leider werde ich aus den ganzen Infos nicht schlau.
>
> Ich habe auch das Problem, das der Ford nicht mehr erkannt wird:
>
> [main ] INFO 2024/08/07 18:17:39 evcc 0.129.0
>
> [main ] INFO [...]

> Ich hatte das gleiche problem. Ich musste einen neuen Access/ refresh token erzeugen, danach ging es

Dito
Neues Anmeldetoken, seitdem wieder Ruhe.

dremde, 2024-08-07:
> Dito Neues Anmeldetoken, seitdem wieder Ruhe.

Ok, und was ist das Anmeldetoken ? Passwort von Ford-Account scheint es nicht zu sein...das ändert nämlich nichts.

Sassar88, 2024-08-07:
Mit den Client Secret kannst du dann via Postman dir die tokens generieren

Max112233445566, 2024-08-08:
> Ok, und was ist das Anmeldetoken ? Passwort von Ford-Account scheint es nicht zu sein...das ändert nämlich nichts.

Schau mal hier 
https://github.com/evcc-io/evcc/discussions/14117#discussioncomment-9654192