evcc-io/evcc #13848: battery 1 power: read failed: read tcp 192.x.x.x

AndreBreitenbach, 2024-05-10:
### Describe the bug

Hallo zusammen,
grundsätzlich funktioniert evcc bei mir wirklich gut.
Seit ein paar Wochen häufen sich aber Fehlermeldungen bzgl. meinem Varta Akku in der Oberfläche.
Direkte Auswirkungen hat dies nicht, allerdings wollte ich trotzdem mal das Problem melden, evtl. hat sich bei Varta etwas verändert.
[image]
[evcc-20240510-090914-debug.log](https://github.com/evcc-io/evcc/files/15271499/evcc-20240510-090914-debug.log)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
[{
	"resource": "/c:/_myALProjects/M365Plastics/app/app.json",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": {
		"value": "AS0081",
		"target": {
			"$mid": 1,
			"path": "/dynamics365/business-central/dev-itpro/developer/analyzers/appsourcecop-as0081",
			"scheme": "https",
			"authority": "learn.microsoft.com",
			"query": "wt.mc_id=d365bc_inproduct_alextension"
		}
	},
	"severity": 4,
	"message": "The InternalsVisibleTo setting will expose your internal objects to any extension with the given name, publisher, and ID. Access modifiers are not designed to be used as a security boundary, but for API development.",
	"source": "AL",
	"startLineNumber": 17,
	"startColumn": 3,
	"endLineNumber": 17,
	"endColumn": 23
},{
	"resource": "/c:/_myALProjects/M365Plastics/app/app.json",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": {
		"value": "MCP0000",
		"target": {
			"$mid": 1,
			"path": "/apidocs/ModusCop/MCP0000",
			"scheme": "https",
			"authority": "docs.modusconsult.de"
		}
	},
	"severity": 2,
	"message": "MODUS Cop active.",
	"source": "AL",
	"startLineNumber": 2,
	"startColumn": 3,
	"endLineNumber": 2,
	"endColumn": 7
},{
	"resource": "/c:/_myALProjects/M365Plastics/app/app.json",
	"owner": "modus-m365.modus-m365-al_project",
	"severity": 2,
	"message": "Version Build and Revision should be 0 for version in app/app.json.",
	"startLineNumber": 5,
	"startColumn": 4,
	"endLineNumber": 5,
	"endColumn": 11
},{
	"resource": "/c:/_myALProjects/M365Plastics/app/src/codeunit/PMARoutingManagement.Codeunit.al",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": {
		"value": "MCP0025",
		"target": {
			"$mid": 1,
			"path": "/apidocs/ModusCop/MCP0025",
			"scheme": "https",
			"authority": "docs.modusconsult.de"
		}
	},
	"severity": 4,
	"message": "The Access property of an object must be set, if the accessibility of members differs.",
	"source": "AL",
	"startLineNumber": 1,
	"startColumn": 18,
	"endLineNumber": 1,
	"endColumn": 47
},{
	"resource": "/c:/_myALProjects/M365Plastics/app/src/codeunit/PMARoutingManagement.Codeunit.al",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": {
		"value": "MCP0026",
		"target": {
			"$mid": 1,
			"path": "/apidocs/ModusCop/MCP0026",
			"scheme": "https",
			"authority": "docs.modusconsult.de"
		}
	},
	"severity": 2,
	"message": "The public member Method DenyFieldChangesForAttachedRoutingLines should contain a descriptive XML comment.",
	"source": "AL",
	"startLineNumber": 665,
	"startColumn": 15,
	"endLineNumber": 665,
	"endColumn": 54
},{
	"resource": "/c:/_myALProjects/M365Plastics/app/src/codeunit/PMARoutingManagement.Codeunit.al",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": {
		"value": "MCP0026",
		"target": {
			"$mid": 1,
			"path": "/apidocs/ModusCop/MCP0026",
			"scheme": "https",
			"authority": "docs.modusconsult.de"
		}
	},
	"severity": 2,
	"message": "The public member Method OnBeforeDenyFieldChangesForAttachedRoutingLines should contain a descriptive XML comment.",
	"source": "AL",
	"startLineNumber": 838,
	"startColumn": 21,
	"endLineNumber": 838,
	"endColumn": 68
},{
	"resource": "/c:/_myALProjects/M365Plastics/app/src/codeunit/PMARoutingManagement.Codeunit.al",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": {
		"value": "MCP0026",
		"target": {
			"$mid": 1,
			"path": "/apidocs/ModusCop/MCP0026",
			"scheme": "https",
			"authority": "docs.modusconsult.de"
		}
	},
	"severity": 2,
	"message": "The public member Method OnAfterDenyFieldChangesForAttachedRoutingLines should contain a descriptive XML comment.",
	"source": "AL",
	"startLineNumber": 843,
	"startColumn": 21,
	"endLineNumber": 843,
	"endColumn": 67
},{
	"resource": "/c:/_myALProjects/M365Plastics/app/src/codeunit/PMARoutingManagement.Codeunit.al",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": {
		"value": "MCP0026",
		"target": {
			"$mid": 1,
			"path": "/apidocs/ModusCop/MCP0026",
			"scheme": "https",
			"authority": "docs.modusconsult.de"
		}
	},
	"severity": 2,
	"message": "The public member Method OnBeforeInsertStdTaskQualificationToRoutingQualification should contain a descriptive XML comment.",
	"source": "AL",
	"startLineNumber": 848,
	"startColumn": 21,
	"endLineNumber": 848,
	"endColumn": 77
},{
	"resource": "/c:/_myALProjects/M365Plastics/app/src/permissionset/PMAGeneralNew.PermissionSet.al",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": {
		"value": "AL0432",
		"target": {
			"$mid": 1,
			"path": "/dynamics365/business-central/dev-itpro/developer/diagnostics/diagnostic-al432",
			"scheme": "https",
			"authority": "learn.microsoft.com",
			"query": "wt.mc_id=d365bc_inproduct_alextension"
		}
	},
	"severity": 4,
	"message": "Table 'M365 PMA Planning Routing Tool' is marked for removal. Reason: No longer used with plastics 2.0.. Tag: 221102.",
	"source": "AL",
	"startLineNumber": 10,
	"startColumn": 15,
	"endLineNumber": 10,
	"endColumn": 47
},{
	"resource": "/c:/_myALProjects/M365Plastics/app/src/permissionset/PMAGeneralNew.PermissionSet.al",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": {
		"value": "AL0432",
		"target": {
			"$mid": 1,
			"path": "/dynamics365/business-central/dev-itpro/developer/diagnostics/diagnostic-al432",
			"scheme": "https",
			"authority": "learn.microsoft.com",
			"query": "wt.mc_id=d365bc_inproduct_alextension"
		}
	},
	"severity": 4,
	"message": "Table 'M365 PMA Planning Routing Tool' is marked for removal. Reason: No longer used with plastics 2.0.. Tag: 221102.",
	"source": "AL",
	"startLineNumber": 11,
	"startColumn": 19,
	"endLineNumber": 11,
	"endColumn": 51
}]
```

### Log details

```text
lp-1 ] DEBUG 2024/05/10 09:13:46 charger status: A
[site ] DEBUG 2024/05/10 09:13:46 ----
[lp-1 ] DEBUG 2024/05/10 09:13:46 charge power: 0W
[lp-1 ] DEBUG 2024/05/10 09:13:46 charge currents: [0 0 0]A
[site ] DEBUG 2024/05/10 09:13:46 pv power: 3924W
[site ] DEBUG 2024/05/10 09:13:46 battery soc: 85%
[site ] DEBUG 2024/05/10 09:13:46 battery power: -2106W
[site ] DEBUG 2024/05/10 09:13:46 grid meter: -1624W
[site ] DEBUG 2024/05/10 09:13:46 grid powers: [-559 -586 -479]W
[site ] DEBUG 2024/05/10 09:13:46 grid currents: [-2.41 -2.53 -2.17]A
[site ] DEBUG 2024/05/10 09:13:46 battery has priority at soc 85% (< 100%)
[site ] DEBUG 2024/05/10 09:13:46 site power: -1624W
[lp-1 ] DEBUG 2024/05/10 09:13:46 charge total import: 1488.020kWh
[lp-1 ] DEBUG 2024/05/10 09:13:47 charger status: A
[site ] DEBUG 2024/05/10 09:13:50 ----
[lp-1 ] DEBUG 2024/05/10 09:13:50 charge power: 0W
[lp-1 ] DEBUG 2024/05/10 09:13:51 charge currents: [0 0 0]A
[site ] DEBUG 2024/05/10 09:13:51 pv power: 3944W
[site ] ERROR 2024/05/10 09:14:06 battery 1 power: read failed: read tcp 192.168.2.72:48002->192.168.2.44:502: read: connection reset by peer
[site ] DEBUG 2024/05/10 09:14:06 ----
[lp-1 ] DEBUG 2024/05/10 09:14:06 charge power: 0W
[lp-1 ] DEBUG 2024/05/10 09:14:06 charge currents: [0 0 0]A
[site ] DEBUG 2024/05/10 09:14:06 pv power: 3949W
[site ] DEBUG 2024/05/10 09:14:06 battery soc: 85%
[site ] DEBUG 2024/05/10 09:14:06 battery power: -2105W
[site ] DEBUG 2024/05/10 09:14:06 grid meter: -1673W
[site ] DEBUG 2024/05/10 09:14:06 grid powers: [-559 -603 -511]W
[site ] DEBUG 2024/05/10 09:14:06 grid currents: [-2.42 -2.58 -2.28]A
[site ] DEBUG 2024/05/10 09:14:06 battery has priority at soc 85% (< 100%)
[site ] DEBUG 2024/05/10 09:14:06 site power: -1673W
[lp-1 ] DEBUG 2024/05/10 09:14:06 charge total import: 1488.020kWh
[lp-1 ] DEBUG 2024/05/10 09:14:06 charger status: A
[site ] DEBUG 2024/05/10 09:14:10 ----
[lp-1 ] DEBUG 2024/05/10 09:14:10 charge power: 0W
[lp-1 ] DEBUG 2024/05/10 09:14:11 charge currents: [0 0 0]A
```

### What type of operating system are you running?

Linux

### Version

0.126.2

VolkerK62, 2024-05-10:
Der Fehler kommt ja nicht durchgehend.
Welches `interval` hast du in der evcc.yaml definiert?

andig, 2024-05-10:
Bei nicht reagier nden Geräten können wir wenig tun. Bitte an Varta wenden.

closed 2024-05-10 by andig

AndreBreitenbach, 2024-05-10:
> Der Fehler kommt ja nicht durchgehend. Welches `interval` hast du in der evcc.yaml definiert?

interval steht auf 10s

VolkerK62, 2024-05-10:
>interval steht auf 10s

dann erhöhe das mal auf 30s.
Auffällig ist im Log, dass die Antwortblöcke 4s bzw 16s auseinanderliegen.
U.U. sind 10s zu schnell für Varta.

AndreBreitenbach, 2024-05-10:
> > interval steht auf 10s
> 
> dann erhöhe das mal auf 30s. Auffällig ist im Log, dass die Antwortblöcke 4s bzw 16s auseinanderliegen. U.U. sind 10s zu schnell für Varta.

Habe mal auf interval 30s gestellt. Sieht erstmal gut aus. Danke, ich werde es beobachten.

zoomy01, 2025-02-20:
Hört sich ebenfalls nach diesem Probelm und der Lösung an...

https://github.com/evcc-io/evcc/issues/12828#issuecomment-2103139315