evcc-io/evcc #13854: SMA Tripower SME Entladesperre verhindert aufladen.

oskarfessel, 2024-05-10:
### Describe the bug

Wenn ich beim meinem EVCC 1.26.2 jetzt im Sommer verhindern möchte, dass bei gplantem Laden die Hausbatterie entladen wird, dann wird die Batterie auch immer **nicht** geladen.  Das möchte ich nicht so gerne, weil, wenn genug Sonne da ist, soll die bitte ins Auto und was das nicht wegschafft soll in die Batterie.

### Steps to reproduce

1. Setting "Verhindere Entladung im Schnell-Modus und bei geplantem Laden." anstellen
2. fast leeres Fahrzeug verbinden
3. Fahrzeug wird im Modus "Schnell" aufgeladen bis MinSoC, bzw. mauell auf schnell stellen.
4. Fahrzeug lädt Solar, Rest wird ins Netz eingespeist.  Die Hausbaterie bleibt auf dem Stand zu Ladebeginn
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

interval: 10s # control cycle interval

log: info

mqtt:
  broker: 192.168.100.138:1883
  topic: evcc # root topic for publishing, set empty to disable publishing

meters:
- name: sma_grid
  type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.100.253 # IP-Adresse oder Hostname
  interface: xennet1

- type: template
  template: sma-hybrid
  name: pv2bat
  host: 192.168.100.99
  usage: battery
  modbus: tcpip
  capacity: 16.6
  id: 3 # Configured Modbus Device ID (SMA default ID 3 = 126)

- type: template
  template: sma-hybrid
  id: 3
  host: 192.168.100.99
  port: 502
  usage: pv
  modbus: tcpip
  name: pv3

hems:
  type: sma
  deviceid: *****
  allowcontrol: false

chargers:
- type: template
  template: wallbe-meter
  host: 192.168.0.8
  port: 502
  name: wallbox2

vehicles:
- type: template
  template: volvo
  title: UNIX43E
  icon: car
  user: *****
  password: *****
  vin: *****
  mode: pv
  capacity: 11
  phases: 1
  maxcurrent: 14
  name: ev1

loadpoints:
- title: Garage
  charger: wallbox2
  mode: pv

site:
  title: Mein Zuhause
  residualPower: 100
  meters:
    grid: sma_grid
    battery:
      - pv2bat
    pv:
      - pv3
```

### Log details

```text
n/a
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.126.2 (86b32070)

premultiply, 2024-05-10:
Die Einflussmöglichkeiten sind leider sehr hersteller- und modellabhängig beschränkt.
Die Alternative wäre die Funktion einfach zu deaktivieren, sie ganz aus dem Code auszubauen oder einen anderen Weg zu finden wie das Intendierte umgesetzt werden kann.

oskarfessel, 2024-05-10:
Tja, das stimmt.  Deswegen könnte es helfen, wenn ich verstehen würde, wo die Logik für den SMA Tripower SE hierfür programmiert ist um das zu verbessern.  Das habe ich noch nicht rausbekommen. Ich erinnere mich, dass da speziell mit Leistungsvorgabe gearbeitet wird, weil die Methode für die SBS nicht funktioniert hat.
Leider gibt es kein MQTT interface für den API-Call "batterydischargecontrol", dann könnte das ja sehr einfach mit der Heimautomatisierung erschlagen werden - ich möchte nämlich da nicht mischen zwischen HTTP und MQTT.

andig, 2024-05-10:
> ich möchte nämlich da nicht mischen zwischen HTTP und MQTT.

warum auch immer?

oskarfessel, 2024-05-11:
Weil meine Hausautomatisierung dafür zwei Geräte angelegt haben möchte. Und alles andere über MQTT läuft…

andig, 2024-05-11:
Na wenn das kein starkes Argument für evcc ist ;). Aber Scherz beiseite- das API fehlt und sollte ergänzt werden. Issue selbst ist erledigt solange wir keine besseren Informationen haben wie die Entladesperre umgesetzt werden könnte.

closed 2024-05-11 by andig