evcc-io/evcc #13856: invalid message template

TobiasHuber1980, 2024-05-10:
### Describe the bug

Messaging  template invalid message

### Steps to reproduce

1. Ladung starten/stoppen
2. invalid message template for soc oder
3. invalid message template for stop
...

### Configuration details

```yaml
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:  
    - type: telegram 
      token: xxx
      chats:
      - xxx
```

### Log details

```text
[push ] ERROR 2024/05/10 15:01:42 invalid message template for soc: wanted: [vehicleSoc], got: [sessionEnergy pvAction gridPower connectedDuration effectivePriority batteryPower effectivePlanSoc homePower auth effectiveMinCurrent chargeDuration chargerIcon title vehicleMinSoc chargeTotalImport chargeCurrent sessionCo2PerKWh limitEnergy minCurrent limitSoc siteTitle vehiclePlanTime effectiveLimitSoc batterySoc planProjectedStart mode batteryCapacity maxCurrent prioritySoc gridConfigured phasesConfigured batteryEnergy batteryMode pvRemaining chargedEnergy gridEnergy vehicleIcon planOverrun connected phaseRemaining pvPower chargePower vehicles effectiveMaxCurrent chargerPhases1p3p chargeRemainingEnergy smartCostActive phasesActive battery priority phaseAction vehicleOdometer planTime effectivePlanTime charging vehicleClimaterActive bufferStartSoc vehicleSoc vehicleTitle fatal vehicleLimitSoc vehicleRange enabled vehiclePlanSoc chargerFeatureHeating sessionSolarPercentage phasesEnabled vehicleName interval vehicleDetectionActive batteryDischargeControl disableThreshold vehicleCapacity chargerFeatureIntegratedDevice currency sponsorTokenExpires sessionPricePerKWh residualPower smartCostType chargeCurrents chargerPhysicalPhases planEnergy enableThreshold pvEnergy loadpoint pv version sessionPrice bufferSoc sponsor]
```

### What type of operating system are you running?

Linux

### Version

0.126.2 (2254fedd)

andig, 2024-05-10:
Gibts zu dem Zeitpunkt in /api/state einen soc am Ladepunkt?

TobiasHuber1980, 2024-05-10:
Während der Ladung:
```
{
  "result": {
    "auth": {
      "vehicles": {

      }
    },
    "battery": [
      {
        "power": -1510.68395996094,
        "soc": 48.4797388315201,
        "capacity": 11.5,
        "controllable": false
      }
    ],
    "batteryCapacity": 11.5,
    "batteryDischargeControl": false,
    "batteryEnergy": 0,
    "batteryMode": "unknown",
    "batteryPower": -1510.68395996094,
    "batterySoc": 48.4797388315201,
    "bufferSoc": 80,
    "bufferStartSoc": 0,
    "currency": "EUR",
    "fatal": null,
    "greenShareHome": 1,
    "greenShareLoadpoints": 0.995081152885694,
    "gridConfigured": true,
    "gridEnergy": 0,
    "gridPower": 6.74324750900269,
    "homePower": 445.84654340744,
    "interval": 30,
    "loadpoints": [
      {
        "chargeCurrent": 6,
        "chargeCurrents": [6.039, 0, 0],
        "chargeDuration": 103000000000,
        "chargePower": 1370.9,
        "chargeRemainingDuration": 13378000000000,
        "chargeRemainingEnergy": 5094.33333333335,
        "chargeTotalImport": 1188.088,
        "chargedEnergy": 38.9999999999873,
        "chargerFeatureHeating": false,
        "chargerFeatureIntegratedDevice": false,
        "chargerIcon": null,
        "chargerPhases1p3p": true,
        "chargerPhysicalPhases": null,
        "charging": true,
        "connected": true,
        "connectedDuration": 0,
        "disableThreshold": 0,
        "effectiveLimitSoc": 100,
        "effectiveMaxCurrent": 16,
        "effectiveMinCurrent": 6,
        "effectivePlanSoc": 0,
        "effectivePlanTime": "0001-01-01T00:00:00Z",
        "effectivePriority": 0,
        "enableThreshold": 0,
        "enabled": true,
        "limitEnergy": 0,
        "limitSoc": 0,
        "maxCurrent": 16,
        "minCurrent": 6,
        "mode": "pv",
        "phaseAction": "inactive",
        "phaseRemaining": 0,
        "phasesActive": 1,
        "phasesConfigured": 0,
[... 202 lines cut ...]
        "chargedKWh": 1878.64624366258,
        "solarPercentage": 79.4139431311494
      }
    },
    "tariffFeedIn": 0.08,
    "tariffGrid": 0.3094,
    "tariffPriceHome": 0.08,
    "tariffPriceLoadpoints": 0.08,
    "vehicles": {
      "my_car_1": {
        "title": "VW ID.5",
        "icon": "car",
        "capacity": 77
      },
      "my_car_2": {
        "title": "BMW i3",
        "icon": "car",
        "capacity": 42,
        "limitSoc": 100
      }
    },
    "version": "0.126.2 (2254fedd)"
  }
}
```

TobiasHuber1980, 2024-05-10:
es kommen SOC und Stop

`[push ] ERROR 2024/05/10 15:01:42 invalid message template for soc: wanted: [vehicleSoc], got: ....`
und 
`[push ] ERROR 2024/05/10 16:36:09 invalid message template for stop: wanted: [chargedEnergy chargeDuration], got: ....`

andig, 2024-05-11:
Ist das ein neuer Fehler und tritt der jetzt immer bei den Events auf? Könntest Du mal bitte probieren, oder der in d9b3d46633ccbf730ed43ac45ec4424c284fa658 auch schon auftritt? Evtl. liegts irgendwie an https://github.com/evcc-io/evcc/pull/13804, auch wenn ich nicht wüsste wieso.

TobiasHuber1980, 2024-05-11:
fehler mit 0.126.2+1715393653 (c8b3e24f)
fehler mit 0.126.2+1715221045 (86b32070)
kein fehler mit 0.126.2+1715150383 (52ca35f2)

hilft das so?

hrpv, 2024-05-15:
Ich sehe diese Fehler auch auf zwei verschiedenen Systemen, bei start stop disconnect 
Im April waren diese Fehler noch nicht da, mit Version 0.125.0+1712628851 war alles o.k.

evcc version 0.126.2 (c8b3e24f)
[evcc-20240515-171134-debug.log](https://github.com/evcc-io/evcc/files/15324023/evcc-20240515-171134-debug.log)

evcc version 0.126.2 (c5330fbb)
[evcc-20240515-171858-debug.log](https://github.com/evcc-io/evcc/files/15324030/evcc-20240515-171858-debug.log)

```
[lp-2  ] INFO 2024/05/15 17:06:06 stop charging <-
[lp-2  ] DEBUG 2024/05/15 17:06:06 wake-up timer: start
[lp-2  ] DEBUG 2024/05/15 17:06:06 charge total import: 36005.190kWh
[lp-2  ] DEBUG 2024/05/15 17:06:06 !! session: chargedEnergy=15.4 > chargedEnergy=12.7
[lp-2  ] DEBUG 2024/05/15 17:06:06 !! session: start=35981.044 stop=36005.190 chargedEnergy=15.424
[push  ] ERROR 2024/05/15 17:06:06 invalid message template for stop: wanted: [chargedEnergy chargeDuration], got: [greenShareHome sessionPrice gridCurrents vehicleTitle planEnergy limitEnergy chargerFeatureHeating vehiclePlanSoc batteryPower chargeCurrent vehicleSoc interval batteryMode siteTitle bufferStartSoc planOverrun pvEnergy sessionCo2PerKWh smartCostType priority vehicleMinSoc connectedDuration mode chargeRemainingEnergy chargerFeatureIntegratedDevice battery batteryDischargeControl effectivePlanSoc phasesActive bufferSoc fatal effectivePlanTime vehicleDetectionActive effectiveMaxCurrent chargeTotalImport chargedEnergy tariffGrid planTime loadpoint chargerIcon greenShareLoadpoints phasesConfigured enabled vehiclePlanTime chargeRemainingDuration gridPowers effectiveMinCurrent vehicleClimaterActive connected enableThreshold minCurrent gridPower gridEnergy currency phaseRemaining batteryCapacity vehicleIcon tariffPriceLoadpoints vehicleOdometer batteryEnergy sessionPricePerKWh residualPower smartCostActive title disableThreshold tariffFeedIn maxCurrent chargerPhysicalPhases vehicleLimitSoc version gridConfigured limitSoc pvPower phasesEnabled pvRemaining effectivePriority vehicleName prioritySoc vehicleRange charging pvAction pv planProjectedStart chargerPhases1p3p chargePower batterySoc statistics sessionEnergy effectiveLimitSoc auth homePower sessionSolarPercentage vehicleCapacity chargeDuration tariffPriceHome phaseAction vehicles]
[lp-2  ] INFO 2024/05/15 17:06:06 car disconnected
```

hrpv, 2024-05-15:
update:
mit reduziertem Messaging fast ohne Variablen sind die Fehler weg, aber das ist definitiv keine Lösung

```yaml
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started ${title} in ${mode} mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished ${title} charging
    connect: # vehicle connect event
      title: Car connected
      msg: Car connected
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected
  services:
  - type: telegram

```

andig, 2024-05-15:
Danke für die hilfreiche Analyse und Sorry, dass ich den Case im Test übersehen habe...

closed 2024-05-15 by andig

hrpv, 2024-05-16:
Gerade getestet, messaging läuft wieder, Danke!
evcc (0.126.2+1715825801) 
evcc version 0.126.2 (940ce81a)
❤️