evcc-io/evcc #13862: OCPP: Vehicle that had been manually assigned in the UI is assigned to different charger during identification phase

jduar, 2024-05-10:
### Describe the bug

On a setup with multiple vehicles and multiple chargers - but only one configured vehicle connected, the rest being guests - I had an issue where shortly after a restart, I had to manually assign the configured vehicle to its correct loadpoint.

Shortly after, evcc assigned that same vehicle to a different charger even though I had already manually assigned  it using the UI. In the UI, the vehicle appeared as assigned to two chargers simultaneously. I unassigned it from the wrong charger - while leaving it assigned in the correct one - and shortly afterwards it was automatically assigned to another charger. Again, it appeared as assigned to two chargers in the UI.

There seem to actually be two errors in this behaviour: 
* the manual vehicle assignment I did in the UI was ignored during vehicle identification phase, so the vehicle was reassigned to a different charger despite me having already assigned it manually;
* on the side of the UI, the vehicle was treated as if it was assigned to two chargers simultaneously.

I'll leave a better description below along with the logs, as it's easier to explain the chain of events.

### Steps to reproduce

1. Have multiple chargers
2. Have one connected configured vehicle, and other connected guest vehicles
3. After manually assigning the configured vehicles, evcc still assigned it to other chargers.

### Configuration details

```yaml
network:
  schema: http
  host: evcc
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/en/docs/reference/configuration/interval/

log: debug
levels:
  site: debug
  lp-1: trace
  lp-2: trace
  cache: error
  db: error
  ocpp: trace
  flobz: trace

site:
  - title: Something
    meters:
      grid: grid
    residualPower: 0
    maxGridSupplyWhileBatteryCharging: 0

meters:
  - name: grid
    type: template
    template: huawei-smartlogger
    usage: grid

    modbus: tcpip
    id: 11
    host: 10.10.6.8
    port: 502

  - name: pv
    type: template
    template: huawei-sun2000-rs485
    usage: pv

    modbus: tcpip
    id: 1
    host: 10.10.6.8
    port: 502

chargers:
  - name: charger1
    type: template
    template: pulsarplus
[... 357 lines cut ...]
[lp-5  ] DEBUG 2024/05/10 11:35:31 charge power: 7360W
[lp-6  ] DEBUG 2024/05/10 11:35:31 charge power: 0W
[site  ] DEBUG 2024/05/10 11:35:31 grid meter: 17077W
[site  ] DEBUG 2024/05/10 11:35:31 grid currents: [33.5 17.4 20.5]A
[site  ] DEBUG 2024/05/10 11:35:31 pv power: 0W
[site  ] DEBUG 2024/05/10 11:35:31 site power: 17077W
[lp-2  ] DEBUG 2024/05/10 11:35:31 charger status: C
[lp-2  ] DEBUG 2024/05/10 11:35:31 vehicle api refresh

// blc is once again automatically assigned to another charger, charger2, despite showing as already assigned to charger 5 in the UI - like before, it now appears as assigned to both loadpoint 2 and 5 in the UI

[main  ] DEBUG 2024/05/10 11:35:31 vehicle status: C (blc)
[main  ] DEBUG 2024/05/10 11:35:31 vehicle status: A (sfr)
[lp-2  ] INFO 2024/05/10 11:35:31 vehicle updated: unknown -> blc
[lp-2  ] DEBUG 2024/05/10 11:35:31 vehicle soc: 87%
[lp-2  ] DEBUG 2024/05/10 11:35:31 vehicle range: 298km
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.126.2

andig, 2024-05-11:
> Again, it appeared as assigned to two chargers in the UI.

That would be a bug, but unfortunately it's not visible from the logfile.

closed 2024-06-06 by github-actions[bot]

andig, 2024-06-08:
@jduar if this is still an issue, a look at /api/state when this happens could help.