evcc-io/evcc #13867: Load MAnagement - charging oscillates if current is close to maxCurrent

OneLineTwoBugs, 2024-05-11:
### Describe the bug

This setup has two circuits: grid @ 125A and emob @ 10A with parent: grid. There are 7 loadpoints all connected to emob. Only one loadpoint has a connected car. the car limits charging to 5A. Charging starts, but is then stopped again during the next regulation.
the issue disappears if emob is configured with 63A maxCurrent.

### Steps to reproduce

see description

### Configuration details

```yaml
see attached config file
```

### Log details

```text
see attached log file
```

### What type of operating system are you running?

Linux

### Version

0.126.2

premultiply, 2024-05-11:
Current signs are missing.

@andig Maybe we should add an additional flag to compatible meter templates and reject LM usage with non-flagged unknown meters at all?

closed 2024-05-11 by premultiply

OneLineTwoBugs, 2024-05-11:
What do you mean by "Current signs are missing"?

premultiply, 2024-05-11:
The currents you provide with custom meter do not include the correct sign (current flow direction). They are all positive (unsigned) but power is negative.

OneLineTwoBugs, 2024-05-11:
ABB meters (which are used in the background) don't provide current signs. I'm not sure if requiring something that such a major mamnufacturer does not have is a sensible requirement. I can guess them, though

premultiply, 2024-05-11:
LM does require signed currents.
So short so easy.

Maybe use the integrated templates instead of your custom meter. It should provide signed currents as requested.

Check with `evcc meter`.

OneLineTwoBugs, 2024-05-11:
I can't use evcc's implementation because I need the meter values for other things as well (monitoring and billing), i.e. I have my own code communicating with the meter and then a script that provides values to evcc from the cache.

I've looked at evcc's implementation of ABB meters and I don't see a current sign correction in the code, hence I assume evcc would have the exact same issue.

 Current sign could be correctly determined by looking at the sign of ApparentPower. One can also use effective power (i.e. the "normal" power value), as it will be correct in 99.99% of cases, however it needs to be checked for all three phases rather than just one. I'll add this to my script and can probably test again tomorrow.

However, the current signs would not explain why setting the emob circuit to 63A makes the behaviour disappear permanently. The circuit that seems to be limiting is emob, not grid, and emob never sees negative power.

premultiply, 2024-05-11:
The correction is done internally based on the sign of the phase powers that need to be provided if the currents are unsigned.

OneLineTwoBugs, 2024-05-11:
```
root@m8pi:/home/lorin# evcc meter
[main  ] INFO 2024/05/11 13:42:10 evcc 0.126.2
[main  ] INFO 2024/05/11 13:42:10 using config file: /etc/evcc.yaml
[db    ] INFO 2024/05/11 13:42:11 using sqlite database: /root/.evcc/evcc.db
gridmeter
---------
Power:          -19439W
Current L1..L3: 26.6A 26.1A 28.7A

pvmeter
-------
Power:          23628W
Current L1..L3: 32.7A 32.9A 32.8A

emobmeter
---------
Power:          48W
Current L1..L3: 0.25A 0.17A 0.17A
```

apparently not for all meters? I can change the current sign in the script input though

premultiply, 2024-05-11:
Provide 
a) signed currents or
b) unsigned currents AND signed phase powers

The included templated should work out of the box in most cases.

SolarPower2024, 2024-05-12:
> Current signs are missing.
> 
> @andig Maybe we should add an additional flag to compatible meter templates and reject LM usage with non-flagged unknown meters at all?

@premultiply ich weiß ihr müsst die Anwendung natürlich so programmieren, dass die Funktionen von "nicht technikaffinen Personen" ebenso ohne Schwierigkeiten genutzt werden können und das geht am einfachsten, wenn man Sachen ausschließt.

Ich würde dich aber bitten trotzdem eine Möglichkeit zu belassen, die LM Funktion per custom Meter zu nutzen. Gerne auch mit einem Hinweis / Verweis auf die Doku was die Mindestanforderungen dafür sind, dass das custom meter xyz erfüllen muss, etc.

Ich habe bei mir einen recht speziellen Anwendungsfall, weshalb ich beim grid meter nicht mit einem Template arbeiten kann und würde mich daher sehr freuen, wenn das weiterhin funktioniert. 
Ansonsten muss ich mir die Funktion selber im Smarthome nachbauen, hätte es aber schon gerne alles zusammen in evcc. 
Vielen Dank.

pdonn, 2024-06-16:
habe ein ähnlich gelagertes Problem mit Hager Zählern. Auch dort sind die Phasenströme unsigned. 
Wie kann ich die signed Phasenleistungen am besten zur Verfügung stellen? Als Meter ist es standardmäßig nicht vorgesehen, oder?

andig, 2024-06-16:
@pdonn you can configure `powers` on your custom meter. Might be nice to create a template for that.

pdonn, 2024-06-16:
thanks! in the generic template I only see "power" and therefore only the overall power consumption. Can I use "powers" in the same way as "currents"?

andig, 2024-06-16:
What generic template?

pdonn, 2024-06-16:
this is working well! thanks!
I mean the generic template here: [https://docs.evcc.io/docs/devices/meters#](url)