evcc-io/evcc #13869: Kleiner Bug im UI - iPhone12 Mini

michiproep, 2024-05-11:
### Describe the bug

Bei mehreren aktiven Ladepunkten + Batterie + Einspeisung sieht das meistens so aus:
[image]

### Steps to reproduce

siehe Beschreibung

### Configuration details

```yaml
#not relevant
```

### Log details

```text
#not relevant
```

### What type of operating system are you running?

Linux

### Version

0.126.2

naltatis, 2024-05-11:
Bitte mit Log oder ausgeklappten Detailwerten damit ich das nachstellen kann.

michiproep, 2024-05-12:
Jetzt muss ich natürlich warten bis das wieder auftritt.
Aber bis dahin ca Werte:
Haus 3
Auto 6
WP 0,1
PoolWP 1,5
InfrHeizung 1,1
Batterie 2,2
Einsp. 0,5

closed 2024-05-13 by naltatis