evcc-io/evcc #13871: Mercedes hinzufügen: * '' has invalid keys: user

Nele02, 2024-05-11:
### Describe the bug

Beim Generieren der config für meinen Mercedes mit 'evcc configure --advanced' tritt nach der Eingabe aller Daten folgender Fehler auf. Es wirkt so, als wäre die generierte config ungültig. Beim Starten vom evcc tritt derselbe Fehler auf.

Testing the configuration of Mercedes-Benz ...
   Error: cannot create vehicle type 'mercedes': 1 error(s) decoding:

* '' has invalid keys: user

Ich nutze die Version: evcc version 0.126.2 (c8b3e24f)

### Steps to reproduce

1. evcc configure --advanced
2. Daten eingeben
3. Bei der automatischen Prüfung tritt der Fehler auf

### Configuration details

```yaml
- type: template
  template: mercedes
  user: xxx@t-online.de
  title: EQA
  icon: car
  region: EMEA
  accessToken: xxx
  refreshToken: xxx
  vin: W1N2XXXX
  capacity: 66.5
  phases: 3
  mode: minpv
  minCurrent: 6
  maxCurrent: 16
  name: ev2
```

### Log details

```text
Testing the configuration of Mercedes-Benz ...
   Error: cannot create vehicle type 'mercedes': 1 error(s) decoding:

* '' has invalid keys: user
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.126.2 (c8b3e24f)

simon2022, 2024-05-11:
I am currently running into the same problem. It seems as if the generator is faulty.

RTTTC, 2024-05-11:
What happens if you comment it out?

Nele02, 2024-05-11:
> What happens if you comment it out?

I still get the same error after deleting the line from the config file.

RTTTC, 2024-05-11:
Did you restart evcc after changing yaml?

Nele02, 2024-05-11:
> Did you restart evcc after changing yaml?

Yes, I did.

andig, 2024-05-12:
Should be fixed in nightly.

closed 2024-05-12 by andig

Nele02, 2024-05-12:
@andig unfortunately I still get the same error with the latest nightly version:

`evcc version 0.126.2 (df4b8baf)`

[image]

andig, 2024-05-15:
@Nele02 bitte probiere morgen nochmal das nightly. Den Parameter "account" hätten wir nie erlauben dürfen- an jeder anderen Stelle in evcc heisst er user...