evcc-io/evcc #13872: Tibber und PV: Plan berücksichtigt nicht korrekt, wenn der eingekaufte Strom günstiger ist

foto-andreas, 2024-05-11:
### Describe the bug

Zielladen ist auf 100% eingestellt. Der Plan steht auf vollladen bis morgen um 18 Uhr. Aktuell liefert die Sonne etwa 6kW. Es wären damit noch 2h zu laden. Der Einspeisepreis steht auf 12.3ct. Morgen nachmittag kostet der Strom im Einkauf 4ct.

Statt nun auf den günstigen Strom morgen zu warten, wird aktuell mit Solarstrom geladen. 

Damit wird nun für 12.3ct statt morgen für 4ct geladen.

Also muss ich jetzt ausschalten und morgen wieder passend einschalten.

### Steps to reproduce

1. Zielladen 100% morgen Abend
2. Sonne einschalten
3. Billigen Strom morgen einkaufen
4. PV-Stellung in EVCC

### Configuration details

```yaml
network:
  schema: https
  host: xxx
  port: 7070

log: debug
levels:
  cache: debug
  lp-1: debug
  hyundai: info
  renault: debug
  vehicle: debug
  easee:  debug
  tibber: debug

interval: 30s # control cycle interval

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: tibber
    token: xxx
    homeid: xxx
  feedin:
    type: fixed
    price: 0.1231 # [currency]/kWh

meters:
- name: tibber
  type: template
  template: tibber-pulse
  usage: grid
  token: xxx
  homeid: xxx

- name: pv1
  type: template
  template: solaredge-hybrid
  usage: pv
  modbus: tcpip
  host: 192.168.2.50 # Hostname
  port: 502 # Port
  timeout: 30s # Akzeptiere keine Daten die älter als dieser Wert sind # Optional 

vehicles:
- name: zoe
  type: template
  template: renault
  title: Zoe
  user: axxx
[... 849 lines cut ...]
[lp-1  ] DEBUG 2024/05/11 11:29:07 plan: charge 2h26m9s between 2024-05-12 10:33:51 +0000 UTC until 2024-05-12 14:00:00 +0000 UTC (power: 11040W, avg cost: 0.035)
[site  ] DEBUG 2024/05/11 11:29:22 ----
[lp-1  ] DEBUG 2024/05/11 11:29:22 charge power: 0W
[lp-1  ] DEBUG 2024/05/11 11:29:22 charge currents: [0.011 0.008 0.009]A
[site  ] DEBUG 2024/05/11 11:29:22 pv power: 6672W
[site  ] DEBUG 2024/05/11 11:29:22 grid meter: -6416W
[site  ] DEBUG 2024/05/11 11:29:22 grid currents: [0 0 0]A
[site  ] DEBUG 2024/05/11 11:29:22 site power: -6416W
[lp-1  ] DEBUG 2024/05/11 11:29:22 charge total import: 7444.268kWh
[lp-1  ] DEBUG 2024/05/11 11:29:22 charger status: B
[lp-1  ] DEBUG 2024/05/11 11:29:22 next soc poll remaining time: 19m1s
[lp-1  ] DEBUG 2024/05/11 11:29:22 plan: charge 2h26m9s between 2024-05-12 10:33:51 +0000 UTC until 2024-05-12 14:00:00 +0000 UTC (power: 11040W, avg cost: 0.035)
[site  ] DEBUG 2024/05/11 11:29:52 ----
[lp-1  ] DEBUG 2024/05/11 11:29:52 charge power: 0W
[lp-1  ] DEBUG 2024/05/11 11:29:52 charge currents: [0.011 0.008 0.009]A
```
```

### What type of operating system are you running?

Linux

### Version

0.126.2

naltatis, 2024-05-11:
Aktuell ist das, das erwartete Verhalten:

>  Selbst produzierte Sonnenenergie wird weiterhin bevorzugt genutzt

Siehe https://docs.evcc.io/docs/features/plans

foto-andreas, 2024-05-11:
Ah schade. Wäre aber hübsch, wenn es demnächst mal erweitert würde.

andig, 2024-05-12:
Hier gehts doch nicht um eingekauften Strom sondern um die Einspeisung, oder? Die wiederrum hängt von PV Prognose ab und ist daher aktuell nicht im Planner enthalten. Dazu gehören dann auch die Kosten.

Duplikat zu https://github.com/evcc-io/evcc/issues/12746

closed 2024-05-12 by andig