evcc-io/evcc #13887: Garage: max charge current 4A: invalid current 4.0

jomuench, 2024-05-13:
### Describe the bug

in the ui, I try to modify the value for Min. Ladestrom. It is set to 6A (default), after setting it to e.g. 4A, this settings doesn't take effect and I am getting log Message as provided in the title.

### Steps to reproduce

no further details.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: [...]]

interval: 30s # control cycle interval

sponsortoken: [...]]

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
#- type: template
#  template: sonnenbatterie 
#  usage: grid  
#  host: sonnenbatterie  
#  port: 8080  
#  name: grid
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: sonnenbatterie  
  port: 8080  
  name: pv
- type: template
  template: sonnenbatterie 
  usage: battery
  host: sonnenbatterie
  port: 8080
  token: [...]]
  name: battery
- type: template
  template: sma-inverter-modbus 
  id: 3  
  host: inverter  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: inverter
- type: template
  template: sma-energy-meter 
  usage: grid  
  host: energymeter  
  name: grid

chargers:
- type: template
  template: etrel 
  host: sonnencharger  
  port: 502  
  name: wallbox

vehicles:
- type: template
  template: vw 
  title: id.3  
  user: [...]  
  password: [...]
  vin: [...]  
  capacity: 58  
  timeout: 10s  
  name: id.3

loadpoints:
- title: Garage
  charger: wallbox
  vehicle: id.3
  mode: pv

site:
  title: Zuhause
  meters:
    grid: grid
    pv:
    - pv
#    - inverter
#    - energymeter
    battery:
    - battery
  residualPower: 300
```

### Log details

```text
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [site  ] DEBUG 2024/05/13 11:08:11 ----
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [lp-1  ] DEBUG 2024/05/13 11:08:11 charge power: 4288W
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [lp-1  ] DEBUG 2024/05/13 11:08:11 charge currents: [6.32 6.18 6.21]A
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [site  ] DEBUG 2024/05/13 11:08:11 pv power: 4429W
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [site  ] DEBUG 2024/05/13 11:08:11 battery soc: 71%
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [site  ] DEBUG 2024/05/13 11:08:11 battery power: 4059W
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [site  ] DEBUG 2024/05/13 11:08:11 grid meter: 230W
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [site  ] DEBUG 2024/05/13 11:08:11 grid powers: [-3732 48 3913]W
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [site  ] DEBUG 2024/05/13 11:08:11 grid currents: [-15.4 0.994 16.5]A
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [site  ] DEBUG 2024/05/13 11:08:11 site power: 4589W
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [lp-1  ] DEBUG 2024/05/13 11:08:11 detected active phases: 3p
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [lp-1  ] DEBUG 2024/05/13 11:08:11 charger status: C
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [lp-1  ] DEBUG 2024/05/13 11:08:11 soc estimated: 51.13% (vehicle: 50.00%)
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [lp-1  ] DEBUG 2024/05/13 11:08:11 vehicle soc: 51%
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [lp-1  ] DEBUG 2024/05/13 11:08:11 vehicle soc limit: 100%
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [lp-1  ] DEBUG 2024/05/13 11:08:11 vehicle range: 217km
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [lp-1  ] DEBUG 2024/05/13 11:08:11 pv charge current: 0A = 6A + -6.65A (4589W @ 3p)
Mai 13 11:08:11 ThinkCentre-KODI evcc[67066]: [lp-1  ] ERROR 2024/05/13 11:08:11 max charge current 4A: invalid current 4.0
```

### What type of operating system are you running?

Linux

### Version

0.126.2

VolkerK62, 2024-05-13:
due to IEC standard, 6A is the minimum.

jomuench, 2024-05-13:
not an issue but rather UI offering settings that are not possible due to technical restrictions.

closed 2024-05-13 by jomuench

andig, 2024-05-13:
/cc @naltatis