evcc-io/evcc #13899: "vehicle soc: timeout" obwohl API (veraltete) Daten liefert

An-Wi, 2024-05-14:
### Describe the bug

Das eigentliche Problem scheint zu sein, dass die NISSAN-API gern mal abends (ab ca. 18:00:00Z) keine aktuellen Daten mehr liefert.
Da mein evcc-Server 1x pro Woche (montags, 22:00 Uhr) per Taskplaner neu startet, entsteht dabei folgende Situation: 
evcc-UI zeigt danach keinen Ladestand ("---") und keine Reichweite mehr an.

Theoretisch soweit korrekt, allerdings etwas verwirrend, da ohne Neustart das "Problem" nicht sichtbar geworden wäre.
Momentan habe ich aber auch keine richtige Idee, wie evcc damit anders umgehen sollte, es schien nur so (bevor ich die Logs studiert hatte) als wäre die NISSAN-API gar nicht mehr erreichbar gewesen, was hin und wieder durch das mehrfach notwendige Login auch gelegentlich auftritt (eine Eigenart der API, die bereits im Code ausreichend gut behandelt ist).

Vielleicht sollte generell ein "veralteter" SOC-Wert visualisiert werden, wenn sein timestamp eine bestimmte Zeitgrenze überschreitet?

### Steps to reproduce

1. Logs checken, bis "timestamp" von "[nissan] TRACE" _einfriert_
2. evcc neu starten
3. UI zeigt "---" beim Ladestand an

### Configuration details

```yaml
# open evcc at http://evcc.local:7070

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 20s   #25s #30s # control cycle interval

[...]

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: ev1
    type: template
    #template: nissan 
    template: nissan-ariya
    title: Ariya 
    capacity: 87  # kWh
    cache: 5m   #default: 15m

[...]
```

### Log details

```text
[site  ] DEBUG 2024/05/12 06:49:44 ----
[lp-1  ] DEBUG 2024/05/12 06:49:44 charge power: 4048W
[lp-1  ] DEBUG 2024/05/12 06:49:44 charge currents: [5.72 5.7 5.73]A
[site  ] DEBUG 2024/05/12 06:49:44 pv power: 533W
[site  ] DEBUG 2024/05/12 06:49:44 grid meter: 3660W
[site  ] DEBUG 2024/05/12 06:49:44 grid currents: [5.21 5.11 5.29]A
[site  ] DEBUG 2024/05/12 06:49:44 site power: 3560W
[lp-1  ] DEBUG 2024/05/12 06:49:44 detected active phases: 3p
[lp-1  ] DEBUG 2024/05/12 06:49:44 charge total import: 4556.084kWh
[lp-1  ] DEBUG 2024/05/12 06:49:44 charger status: C
[nissan] TRACE 2024/05/12 06:49:44 GET https://alliance-platform-caradapter-prod.apps.eu2.kamereon.io/car-adapter/v2/cars/***/battery-status
[nissan] TRACE 2024/05/12 06:49:45 {"id":"***","timestamp":"2024-05-11T17:41:31Z","batteryLevel":31,"batteryAutonomy":164,"batteryCapacity":87,"batteryAvailableEnergy":28,"plugStatus":1,"chargingStatus":0.0,"chargingRemainingTime":2046}
[lp-1  ] WARN 2024/05/12 06:49:45 vehicle soc: timeout (ignored by estimator)
[lp-1  ] DEBUG 2024/05/12 06:49:45 soc estimated: 32.14% (vehicle: 31.00%)
[... 113 lines cut ...]
[lp-1  ] DEBUG 2024/05/14 07:34:46 charge total import: 4605.671kWh
[lp-1  ] DEBUG 2024/05/14 07:34:46 charger status: A
[lp-1  ] DEBUG 2024/05/14 07:34:46 next soc poll remaining time: 31m40s
[nissan] TRACE 2024/05/14 07:34:58 GET https://alliance-platform-caradapter-prod.apps.eu2.kamereon.io/car-adapter/v2/cars/***/battery-status
[nissan] TRACE 2024/05/14 07:34:58 {"id":"***","timestamp":"2024-05-14T05:34:49Z","batteryLevel":57,"batteryAutonomy":306,"batteryCapacity":87,"batteryAvailableEnergy":50,"plugStatus":0,"chargingStatus":0.0,"chargingRemainingTime":330}
[site  ] DEBUG 2024/05/14 07:35:05 ----
[lp-1  ] DEBUG 2024/05/14 07:35:05 charge power: 0W
[lp-1  ] DEBUG 2024/05/14 07:35:05 charge currents: [0 0 0]A
[site  ] DEBUG 2024/05/14 07:35:05 pv power: 854W
[site  ] DEBUG 2024/05/14 07:35:06 grid meter: -403W
[site  ] DEBUG 2024/05/14 07:35:06 grid currents: [1.04 0.669 1.05]A
[site  ] DEBUG 2024/05/14 07:35:06 site power: -503W
[lp-1  ] DEBUG 2024/05/14 07:35:06 charge total import: 4605.671kWh
[lp-1  ] DEBUG 2024/05/14 07:35:06 charger status: A
[lp-1  ] DEBUG 2024/05/14 07:35:06 next soc poll remaining time: 31m20s
[site  ] DEBUG 2024/05/14 07:35:25 ----
```

### What type of operating system are you running?

Windows

### Version

0.126.2

andig, 2024-05-15:
Wenns zur Laufzeit passiert sollte es zumindest einen Fehler im Log geben. Stattdessen „—„ zu setzen scheint auch keine gute Alternative zu sein. Daher wontfix?

closed 2024-05-15 by andig

An-Wi, 2024-05-17:
Ok, wollte es eigentlich nicht weiter ausdehnen, da deine Einschätzung für mich ok war, aber es hat sich gestern eine neue Situation gegeben.
Die Nissan-API verweigert seit gestern früh den Zugriff (ggf. mache ich dazu noch ein eigenes Issue auf, wenn ich es nicht nur temporär sein sollte).

Durch den fehlenden Zugriff auf die Vehicle-API ist der angezeigt Akkustand von 70% bei mir über 24h alt! 
Habe 1x stündliches Aktualisieren hinterlegt:
`loadpoints:`
`  soc:`
`    poll:`
`      mode: always`
`      interval: 60m`

Heute früh vor Ladebeginn waren es real 17%:
[image]
Beim Laden dann irgendwann etwa 25% (laut NissanConnect-App). evcc zeigte weiterhin konstant 70% an:
[image]
...mit vorausgesagter Fehlermeldung im Log (oben schön sichtbar beim Ausrufezeichen):
`[lp-1 ] ERROR 2024/05/17 07:49:35 vehicle soc: Post "https://alliance-platform-caradapter-prod.apps.eu2.kamereon.io/car-adapter/v1/cars/JN***88/actions/refresh-battery-status": oauth2: "invalid_client" "Client authentication failed"` 

Nach Abstecken steht nun aktuell beim Ladestand "--".
Den Zustand jetzt finde ich wieder ok, aber beim Laden und davor den sehr veralteten Akkustand anzuzeigen ohne dass man erkennt, dass er gar nicht mehr passen kann, finde ich unglücklich: 
"Ausgrauen", "kursiver Text", "Tooltip-Info", "Wert in Klammer setzen" oder "Disconnect-Symbol dahinter" könnten da hilfreich sein.

Das kein "estimate" beim Laden in dieser Situation erfolgt ist, finde ich gut - sollte so bleiben.

An-Wi, 2024-05-17:
"nuken" musste ich erstmal nachschlagen 🤣

Den SOC würde ich nicht sofort beim 1. Abfragefehler "nuken" - denke hier sollte ein vernünftiges timeout definiert werden (1h?). Solange wir noch in diesem Intervall bleiben, soll alles laufen wie bisher.
Danach würde ich SOC als "outdated" betrachten und dann auch nur "visuell" hinweisen.

minSOC ist vermutlich hier nochmal ein Thema für sich. Wenn der veraltete SOC-Wert sehr niedrig ist und nicht mehr aktualisiert wird, dann würde die Mindestladung permanent weiterlaufen, oder?
Denke mal, dass im Status "SOC outdated" auch die Mindestladung pausieren oder gar nicht erst anfangen sollte.