evcc-io/evcc #13900: Add dynamic tariff from beyond

mabasger, 2024-05-14:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
Emond is a new dynamic tariff for EV Charge Energy. Until now, they can only control the vehicles charging system via remote control (VW Connect) which is troublesome.
https://eyond.mobilityhouse.com/de_de

**Describe alternatives you've considered**
Changing the tariff :-)
**Additional context**
Add any other context or screenshots about the feature request here.

andig, 2024-05-15:
Do they offer an api?

mabasger, 2024-05-16:
Got the response from customer support: no API available. I suspect the support to be informed, as they answer individually (no text blocks) and seemed to understand what I asked for.

Therefore: issue can be closed; no support from provider