evcc-io/evcc #13901: Batterie-Ladestand auf Hauptseite

mabasger, 2024-05-14:
**Is your feature request related to a problem? Please describe.**
Als Nutzer muss ich in ein Untermenü, um den Ladestand des Hausakkus zu sehen.

**Describe the solution you'd like**
Als Nutzer will ich den Ladestand des Hausakkus transparent auf den ersten Blick sehen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Die Startseite zeigt den aktuellen Energiefluss und extrem prominent das Ladeszenario. Der aktuelle Ladestand des Hausakkus könnte zB sehr einfach hinter dem Batteriesymbol angezeigt werden.

[image]

StevieC121176, 2024-05-14:
wenn du einmal auf den Balken klickst, klappen doch alle Werte aus.
Wenn man deinem Beispiel folgt müssten ja alle Werte wie Hausverbrauch, Auto laden usw hinter den Symbolen angezeigt werden.

mabasger, 2024-05-14:
Das mit der Animation sehe ich gerade erst. Ich finde die Granularität aber sehr schwierig. Kann man die Symbole größer designen, so dass da mehr Ladebalken reinpassen? Und ggf. das Symbol so gestalten, dass man sieht, wie viele Ladebalken es noch sind?

Das Detailmenü ist nur mit Nachdenken klar verständlich. Es stellen sich mir mehrere Fragen:

- warum sind Werte angezeigt, die es zum aktuellen Zeitpunkt nicht gibt? "Batterie entladen", wenn aktuell gelandet wird. "Netzbezug", wenn aktuell eingespeist wird.
- dass die Farben oben eine Bedeutung haben, muss ich erst durch Erfahrung lernen
- IN/OUT - ich muss die Headline lesen und verstehen; sie kann nicht visuell erfasst werden. Intuitiv hätte ich einen Balken erwartet, wie auf der Hauptseite. 

Und bitte: no offence! Das ist hier ein tolles Projekt. Ich möchte nur konstruktives Feedback aus Sicht eines Erstnutzers beisteuern :-)

michiproep, 2024-05-15:
Ja, das wollte ich auch schon mal vorschlagen. Find ich gut!

naltatis, 2024-05-15:
Bislang hatte das Batterie-Icon fünf Stufen. Im PR (s.o.) hab ich den Balken etwas vergrößert und das Icon so umgestellt, dass es nun "stufenlos" Werte anzeigen kann. Gerade bei niedrigen SoCs sollte das hilfreich sein. Bislang wurde bspw. alles unter 20% als "emtpy" angezeigt. Das ist jetzt deutlicher zu unterscheiden.

Deine anderen Kommentare und Vorschläge nehme ich als Impulse mit, falls wir den Bereich noch mal strukturell umbauen. Sehe hier aber keinen konkreten Handlungsbedarf. Prozentanzahl direkt im Diagramm oben haben wir schon mehrfach diskutiert. Hier gewinnt bei uns oft die Strategie "clean and simple". Generell zeigen wir ja schon relativ viel Zahlen und Details an. Aber mit dem Fokus auf das Laden.

michiproep, 2024-05-15:
Ja, alles ok. Und ihr macht das auch super. Gibt selten so ne cleane UI!!

PR #13909 (2024-05-15): UI: higher precision battery icon
fixes #13901

- 🪫 changed battery icon from five-step (`empty`, `quarter`, `half`, `three-quarter`, `full`) to continuous levels based on `soc`.
- 🔎 slightly increased battery level area inside icon

**battery levels**

[battery-animation.webm](https://github.com/evcc-io/evcc/assets/152287/d5d74239-6934-4bfc-8f66-c48f73202ce5)

**increased height**

[image]
closed 2024-05-15 by andig