evcc-io/evcc #13902: "Vehicle limit <xx%> reached" inconsistent with charge percentage

GrafRudi, 2024-05-14:
### Describe the bug

"Vehicle limit <xx%> reached" inconsistent with charge percentage. See attached picture
[image]

IMHO a low prio issue as the functionality works correctly.

### Steps to reproduce

1.  Set the charge limit to 90%
2. Start charging
3. While charging increase the charging limit in the vehicle to 95%
4. Charging will stop at 95% but a wrong message is displayed.

Assume an internal variable is not updated if charging limit is adapted while charging.

### Configuration details

```yaml
If really needed I can attach one
```

### Log details

```text
n.a.
```

### What type of operating system are you running?

Linux

### Version

0.126.2

naltatis, 2024-05-15:
Can you provide a log file? The in-vehicle limit should be shown as `vehicle soc limit: xx%`

naltatis, 2024-05-17:
Ok, so looks like a pure UI and not a data problem. Thats good :D

naltatis, 2024-05-17:
@GrafRudi I can't reproduce this. In the log you've provided, the vehicle is charging. This does not correlate with the state in the screenshot where charging has finished.

But maybe a few details on how we update the vehicle soc value: By default, evcc only requests soc (and limit soc) infos from the vehicle when it gets connected or during charging. When charging has stopped, we don't continue requesting soc (and limit soc) data to allow the vehicle to enter sleep state. The soc usually doesn't change unless its the vehicle is driven (disconnect) or charged again. Self-discharge or vampire drain is ignored here. With the limit soc from the vehicle it's different. You can change this at any time in the app, but we only update it during charging. So it's quite possible to get an out of sync state when changing limit soc while the car is not charging.

Functionality wise, this should not be a problem. Increasing the vehicle limit soc afterwards will result in the car starting to charge until the new limit is reached because evcc keeps the charger enabled. Only decreasing an already reached limit will not reflect in the UI.

The described default behavior can also be configured via `poll` https://docs.evcc.io/docs/reference/configuration/loadpoints#poll

If this does not explain your observed behavior, I need more logs and detailed reproduction steps that go along.

naltatis, 2024-05-29:
I think here is nothing to do at the moment. @GrafRudi feel free to add comments here if your effect does not match the above explanation.

closed 2024-05-29 by naltatis

GrafRudi, 2024-05-29:
Sorry forgot to answer, closure is ok