evcc-io/evcc #13908: LRT Essential+ 11KW Support

t-lie, 2024-05-15:
Hallo zusammen,

hat schon jemand eine LRT Essential+ 11kw Wallbox mit evcc zum Laufen bekommen? Sie hat einen Modbus Anschluss, konnte in der Geräte Liste jedoch kein passendes Modell finden.

Würde mich über Hinweise freuen.

Lieben Dank!

premultiply, 2024-05-15:
Der erste Schritt wäre die vollständige Modbus-Dokumentation beim Hersteller aufzutreiben und hier bereitzustellen.

Ohne Doku können wir nichts tun.

RenatusRo, 2024-05-15:
https://lrt-emobility.de/wp-content/uploads/2021/09/20210831_Anschluss_an_PV_Anlage_EMS_Anderung_Ladeleistung_AC09C_A010C_DE-2.pdf

closed 2024-06-14 by premultiply

Xtensa2C, 2025-03-22:
Hi guys,
I saw in the documentation, that the FW-Version 2022 is required.
My Wallbox (ABB Rebrand of this model) was shipped with FW-2021.
Therefore I wanted to see what works via Modbus and what doesnt.

Unfortunately all LRT websides and documentation links I found are offline.
Does someone still have the linked document from above as a PDF somewhere?