evcc-io/evcc #13916: Innogy: Fehler "charger ...: dial tcp ...: connect: connection refused" nach Upgrade auf 0.126.2

niniob, 2024-05-15:
### Describe the bug

Seit dem Upgrade von 0.125.0 zu 0.126.2 bekomme ich ständig Fehler (rotes Warndreieck im UI) angezeigt. Die Fehlermeldungen sind immer die gleichen und wiederholen sich ständig:

> Garage: charger status: dial tcp 192.168.107.220:5555: connect: connection refused
> Garage: charge total import: dial tcp 192.168.107.220:5555: connect: connection refused
> Garage: charge currents: dial tcp 192.168.107.220:5555: connect: connection refused
> Garage: charge power: dial tcp 192.168.107.220:5555: connect: connection refused
> Garage: charger status: dial tcp 192.168.107.220:5555: connect: connection refused
> Garage: charge total import: EOF

Nach einem Downgrade zurück zu 0.125.0 ist wieder alles okay. Ich vermute es hängt mit folgender Änderung zusammen:
https://github.com/evcc-io/evcc/commit/da4af01d736503b77901648d8a03f45516a99591
Das dort eingeführte Register 307 (igyRegEnergy) gibt es laut Modbus Doku (https://www.eon.de/content/dam/eon/eon-de-zwei/documents/Privatkunden/emobility/ModbusTCP_eBox_manual_v1.pdf) für meine Wallbox (Innogy / e.on / Compleo eBox Professional) nicht. 

### Steps to reproduce

1. Upgrade von 0.125.0 zu 0.126.2

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# database configuration for persisting charge sessions and settings
database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db

# unique installation id
plant: 1b9b8f3f9d85af13394be13964816b05ec8f08f52b4d2674f5853bda089d6caf

interval: 15s # control cycle interval

sponsortoken: ...

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: solax 
  id: 2  
  host: 192.168.107.176  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solax 
  id: 2  
  host: 192.168.107.176  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: solax 
  id: 2  
  host: 192.168.107.176  
  port: 502  
[... 271 lines cut ...]
[modbus] TRACE 2024/05/15 15:59:09 modbus: recv 00 19 00 00 00 07 02 04 04 b2 af 00 00
[site  ] DEBUG 2024/05/15 15:59:09 pv power: 3830W
[modbus] TRACE 2024/05/15 15:59:09 modbus: send 00 1a 00 00 00 06 02 04 00 16 00 01
[modbus] TRACE 2024/05/15 15:59:09 modbus: recv 00 1a 00 00 00 05 02 04 02 00 00
[modbus] TRACE 2024/05/15 15:59:09 modbus: send 00 1b 00 00 00 06 02 04 00 1c 00 01
[modbus] TRACE 2024/05/15 15:59:09 modbus: recv 00 1b 00 00 00 05 02 04 02 00 64
[site  ] DEBUG 2024/05/15 15:59:09 battery soc: 100%
[site  ] DEBUG 2024/05/15 15:59:09 battery power: 0W
[modbus] TRACE 2024/05/15 15:59:09 modbus: send 00 1c 00 00 00 06 02 04 00 46 00 02
[modbus] TRACE 2024/05/15 15:59:09 modbus: recv 00 1c 00 00 00 07 02 04 04 03 9f 00 00
[site  ] DEBUG 2024/05/15 15:59:09 grid meter: -927W
[modbus] TRACE 2024/05/15 15:59:09 modbus: send 00 1d 00 00 00 06 02 04 00 4a 00 02
[modbus] TRACE 2024/05/15 15:59:09 modbus: recv 00 1d 00 00 00 07 02 04 04 28 bc 00 07
[site  ] DEBUG 2024/05/15 15:59:09 site power: -827W
[innogy] TRACE 2024/05/15 15:59:09 modbus: send 00 15 00 00 00 06 01 04 01 33 00 02
[innogy] TRACE 2024/05/15 15:59:09 modbus: close connection and retry, because of EOF
```

### What type of operating system are you running?

Linux

### Version

0.126.2

VolkerK62, 2024-05-15:
das ist nach dem Release 0.126.2. wieder zurückgenommen worden. Ist aktuell nur im Nightly verfügbar.

StefanSchoof, 2024-05-15:
Siehe #13771

niniob, 2024-05-15:
Ok, vielen Dank für den Hinweis. Sorry, hatten den anderen Issue wohl übersehen.

closed 2024-05-15 by niniob