evcc-io/evcc #13919: mqtt-Bug: parsing "null": invalid syntax - Log rattert los

Tombra1889, 2024-05-15:
### Describe the bug

Sporadisch kommt es vor, dass das Log überflutet wird mit Nachrichten "parsing "null": invalid syntax".
Es hilft dann nur evcc zu stoppen. Alles aus dem mqtt Broker zu löschen und dann evcc wieder zu starten. Wenn man dann Glück hat funktioniert es wieder eine Zeit lang.

### Steps to reproduce

1. mqtt Broker läuft, Daten zu evcc sind schon im Broker vorhanden
2. cvcc wird gestartet
3. Log rattert los

### Configuration details

```yaml
# mqtt message broker
mqtt:
  broker: 192.168.66.193:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  # user:
  # password:
```

### Log details

```text
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
[mqtt ] ERROR 2024/05/15 22:17:41 set evcc/loadpoints/2/limitSoc/set: strconv.Atoi: parsing "null": invalid syntax
```

### What type of operating system are you running?

Linux

### Version

v0.126.2

andig, 2024-05-16:
> Sporadisch kommt es vor

null ist kein numerischer Wert- den darfst du also nicht dort hin schreiben und auch nicht retsined!

closed 2024-05-16 by andig

Tombra1889, 2024-05-16:
Irgendetwas scheint nicht ganz rund zu sein:
[image]
Bei den jeweiligen "Set" erwartet er einen string, bei den jeweils übergelagerten Objekten dann eine number.
[image]

[image]