evcc-io/evcc #13924: SMA: Active battery control via SHM 2 plant control

humi208, 2024-05-16:
Since Sunny Home Manager FW update 2.14.10.R it is no more possible to controll battery charging of STP SE via Modbus

See:
https://github.com/evcc-io/evcc/discussions/13667#discussioncomment-9366894
https://www.photovoltaikforum.com/thread/142863-prognosebasierte-ladung-mittels-iobroker/?postID=3713881#post3713881
https://www.photovoltaikforum.com/thread/220338-shm2-0-software-version-2-14-10-r-ca-m%C3%A4rz-2024/?postID=3712264#post3712264

**Describe the solution you'd like**
For me I found a solution using the plant control registers of the Sunny Home Manager 2 itself. See:
https://files.sma.de/downloads/HM-20-Modbus-NSD-TI-de-12.pdf

**Describe alternatives you've considered**
Provide option to use Modbus battery control registers of SHM2 instead of hybrid inverter STP SE which does not work anymore with new SHM2 FW. Before being able to set those a grid gurad code must be set once. This does also allow power control of the plant if you use PV direct marketing and prices are below 0 Cent or -3 Cent.

premultiply, 2024-05-16:
It is a well known bug in the new SMA firmware.

Please contact SMA to fix it.
We can not do anything. 😔

premultiply, 2024-05-16:
Die Modbus-Schnittstelle des SHM benötigt leider sinnloserweise einen Gridguard-Code.
Das ist daher leider kein zielführender Weg.

humi208, 2024-05-16:
Ich habe es gestern kurz Mal getestet und in 60 Minuten zum Laufen gebracht. Jetzt kann ich die Batterie wieder steuern. War äußerst zielführend.

Grid Guard Code gibt es hier: 
https://my.sma-service.com/s/osc-30-grid-guard-code?language=en_US

premultiply, 2024-05-16:
Der typische Anlagenbetreiber hat bei SMA leider immer noch keine Chance einen GGC zu erhalten.

humi208, 2024-05-19:
Siehe auch: 
https://www.photovoltaikforum.com/thread/206718-sma-stp10-0-3se-40-welcher-modbus-register-zum-laden-der-batterie/?postID=3758778#post3758778

https://www.photovoltaikforum.com/thread/206718-sma-stp10-0-3se-40-welcher-modbus-register-zum-laden-der-batterie/?postID=3758900#post3758900

Und folgende...

Das ist ein Feature und wird nur über die Schnittstelle wie von mir oben vorgeschlagen zu lösen sein.
Wenn also nicht gewollt könnt ihr die Batteriesteuerung für SMA mit SHM im System streichen...

Hooch76, 2024-05-28:
> Since Sunny Home Manager FW update 2.14.10.R it is no more possible to controll battery charging of STP SE via Modbus

Ich bin ebenfalls betroffen. Allerdings habe ich keinen STP SE, sondern einen SBS, der aber identisches Verhalten seit dem (automatischen) Upgrade auf 2.14.13.R des SHM zeigt. 
Es gibt eine neue Firmware (4.04.11.R) für den SBS (https://www.sma.de/produkte/batterie-wechselrichter/sunny-boy-storage-37-50-60) in dessen Readme folgendes steht:

```
Fehlerbehebung:
•
Der Sunny Home Manager 2.0 (SHM 2.0) hat ein neues Update bekommen (FW 2.14.13.R).
•
Mit diesem Update kommt es zu Problemen beim Laden und Entladen der angeschlossenen Batterien.
•
Durch ein Update auf die FW-Version 04.04.11.R des SBS 3.7 – 6.0 wird dieses Problem behoben.
•
Das Update ist daher notwendig, wenn der SHM 2.0 die neue Firmware-Version 2.14.13.R erhalten hat, um den normalen Betrieb der angeschlossenen Batterie zu gewährleisten.
•
Betroffene Systeme sollten nach dem Update wieder normal laufen. Weitere Aktionen sind nicht erforderlich.
•
Automatisches Update beim Wechselrichter muss dafür eingeschaltet sein.
```

Werde ich demnächst mal ausprobieren. Evtl folgt auch ein Update für den STP SE.

humi208, 2024-05-28:
Auch das Update des SBS bringt die Kontrolle von außen nicht zurück. Es behebt nur einen generellen Fehler. Das Feature Batterieladung zu kontrollieren mittels EVCC wird weiterhin nicht funktionieren, sofern EVCC nicht auf das Plant Control Modbus IF des SHM umgestellt wird.

wittmannchris, 2024-05-28:
Im PV Forum habe ich soeben erfahren ,dass wohl, zumindest beim SBS 2.5, das Schreiben via Modbus noch funktionieren soll. 
Konnte ich noch nicht testen, aber wäre ein Hoffnungsschimmer. Alle 6 Register zum steuern müssen zyklisch nacheinander geschalten werden....

humi208, 2024-05-28:
40151und 40149 funktionieren auch dort nicht. Beim SBS kann man aber wenigstens die Wirkleistungs-Min/Max Werte setzen und das funktioniert, da der SHM nur die Wirkleistung direkt steuert über 40149.

40793 Minimale Batterieladeleistung, in W U32 FIX0 RW
40795 Maximale Batterieladeleistung, in W U32 FIX0 RW
40797 Minimale Batterieentladeleistung, in W U32 FIX0 RW
40799 Maximale Batterieentladeleistung, in W U32 FIX0 RW

diese dürfen allerdings nicht zyklisch geschrieben werden, da sie im speicher abgelegt werden

wittmannchris, 2024-05-28:
Du hattest ja im PV forum auch gerade kontakt mit birdy ;)

Aber was heisst dies nun konkret für EVCC (da bin ich nämlich zu viel Laie)?

Wird eine Steuerung über EVCC möglich sein? Vermutlich dennoch nicht oder?

premultiply, 2024-05-28:
Dazu müsste man mal ganz konkret erfahren was genau jetzt bei welchem BWR nicht mehr funktioniert.

wittmannchris, 2024-05-28:
> Dazu müsste man mal ganz konkret erfahren was genau jetzt bei welchem BWR nicht mehr funktioniert.

So wie es aussiieht ist jeder BWR betroffen, welcher über den SHM gesteuert wird.

Wie @humi208 bereits geschrieben hat ist es wohl einem JEDEM möglich einen GGC von SMA zu erhalten (habe ich gerade getestet und nun einne in der Hand). 
Dadurch wäre das "beschreiben" des SHM möglich wie in den Links von @humi208 beschrieben und man könnte die aktive Batteriesteuerung wieder aktivieren.

Wäre es nun nicht dennoch eine Option für ein Issue oder Pull request?

Danke, dass du dir das nochmals ansiehst.

Timer2000, 2024-05-28:
Sollte ich so einen GGC bekommen müsste doch in EVCC trotzdem was angepasst werden, damit der HM nun gesteuert wird und nicht mehr der WR.

Oder wie ist das?
Ich persönlich kenne mich mit Modbus nicht aus und kann das selbst nicht umsetzen.

Zudem fehlt mir sogar Admin Rechte beim WR somit weiß ich nicht ob und wie mir der GGC helfen kann.

Ist das beim Tripower Hybrid SE überhaupt möglich? Oder gilt das nur für die SBS?

wittmannchris, 2024-05-28:
> Sollte ich so einen GGC bekommen müsste doch in EVCC trotzdem was angepasst werden, damit der HM nun gesteuert wird und nicht mehr der WR.
>
> Oder wie ist das? Ich persönlich kenne mich mit Modbus [...]

So wie ich das aktuell lese ist das "Überschreiben" des Steuerbefehls im SHM für jeden BWR möglich. 

EVCC müsste natürlich komplett angepasst werden. :/

Timer2000, 2024-05-28:
[https://www.photovoltaikforum.com/thread/220338-shm2-0-software-version-2-14-10-r-ca-m%C3%A4rz-2024/?postID=3775117#post3775117](url)

Hier wird gefragt welche Parameter vom GGC befreit werden müssten. Da kann ich leider nicht helfen.
Mir ist wichtig das EVCC wieder die Kontrolle über die Speichernutzung zurück bekommt.
Was mir nicht klar ist, ich habe in meiner Konfig den SMA Home Manager konfiguriert und nicht den Wechselrichter. Zusätzlich habe ich den Speicher konfiguriert für die Akku Anzeige. Wo steuert EVCC jetzt meinen Tripower per Modbus? 
Oder lief das in meinem Fall schon immer über den SMA HM?

premultiply, 2024-05-28:
Ich hab mir das nochmal angeschaut.
Es ist mir einfach zu kompliziert.

Wenn man das machen wollte müsste man neben den Batteriespeicher-Meter jeweils zwei weitere völlig nutzlose Batterie-Meter mit der IP-Adresse des SHM anlegen. Der bekommt seine Adresse standardmäßig per DHCP...
Dann sorgt ein Meter auf Modbus ID 1 für die regelmäßige GGC-Anmeldung und der zweite macht über Modbus ID 2 (nur) die Batteriesteuerung. Beide Meter liefern jedoch keine sinnvollen Daten bzw. Nullwerte.
Zu allem Überfluss wird auch noch jeder automatisierte Login und Steuerzugriff laut Doku im Anlagenlogbuch einzeln eingetragen.

Das passt einfach vorne und hinten nicht ins Konzept, das versteht kein Mensch.
Zusätzlich bleibt die Sache mit dem Grid Guard Code für einfache Anwender eine erhebliche Hürde - auch wenn man inzwischen theoretisch und tatsächlich solche Codes auch als Normalanwender ohne Heiligenschein erhalten kann.

closed 2024-05-28 by premultiply

Timer2000, 2024-05-28:
Ich kann mich irren aber so wie ich das verstehe müssten zukünftig die Modbus Befehle für die Batteriesteuerung an den Home Manager adressiert werden und nicht wie bisher an den Wechselrichter.
Der Home Manager ist bereits meiner Konfiguration also kein neuer Teilnehmer. Derzeit ist Modbus dort aber gesperrt, was sich möglicherweise ändern könnte. Im übrigen hat der bei mir eine feste IP.

Das weiter zu verfolgen macht aber natürlich nur Sinn sofern SMA an der Firmware was ändern sollte.

premultiply, 2024-05-28:
Wir warten mal ab wie sich SMA dahingehend zukünftig aufstellen wird.

wittmannchris, 2024-05-29:
> Wir warten mal ab wie sich SMA dahingehend zukünftig aufstellen wird.

@premultiply sollte es SMA nicht hinbringen gäbe es ja diesen workaround via SHM-Modbus. Im Iobroker ja schon erfolgreich getestet.

Wäre es denn eine Möglichkeit die Batteriesperre via API in der EVCC instanz z.b. anzuzeigen? Dann hätte man die Möglichkeit sich über andere Skripte eine eigene Batteriesperre zu bauen. 

Dann kann auch jeder mit seinem eigenen GGC etc. rumpfuschen ;)

karlheupel, 2024-05-29:
@wittmannchris @premultiply benötigt man überhaupt zwingend den SHM?

Ist der mit evcc.io nicht einfach nur noch ein Energy meter? Sprich kann man den nicht gegen nen Meter tauschen?
evcc liest ja die produktion der PV WR selbst aus und steuert den Batteriewechselrichter.

Oder übersehe ich da etwas?

premultiply, 2024-05-29:
evcc selbst funktioniert mit dem SMA EM oder einem anderen Meter am Netzübergang genauso.
Völlig egal.

Ob du die integrierten Zusatzfunktionen des SHM zur Anlagensteuerung benötigst oder ggf. anderweitig selbst lokal realisierst musst du wissen.

Die SMA-WR laufen auch mit EM statt SHM.

karlheupel, 2024-05-29:
OK so verstehe ich das auch.. nur nachgefragt:

SHM
- SMA Sunnyboy PV WR
- SMA Tripower PV WR
- SMA SBS mit BYD als Speicher

Alles was ich will ist:
Überschuss erst in die Batterie und ins Auto, dann ins Netz.
Und dann eben die evcc Funktionen zur Priorisierung und ab Winter wenn mein Vertrag umgestellt ist Tibber optimiert laden.

Nichts davon benötigt den SHM oder das macht ja alles evcc für mich.

Oder übersehe ich da noch etwas?