evcc-io/evcc #13925: aktuelle Nightly: Ruckeln der Animationen in der oberen Balkengrafik (Klammern)

StevieC121176, 2024-05-16:
### Describe the bug

Seit der heutigen nightly `0.126.2 (940ce81a)` ruckeln die "Klammern" in der oberen Balkengrafik bei Änderung der Werte. sh. Video

https://github.com/evcc-io/evcc/assets/113791299/fefa413e-3a59-4f27-b854-841497ef67ad

### Steps to reproduce

0.126.2 (940ce81a) starten, Balkengrafik beobachten

### Configuration details

```yaml
nicht relevant
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

0.126.2 (940ce81a)

StefanSchoof, 2024-05-16:
Bei der oberen Klammer neben dem "In" sieht das auch nach ein Artefakt aus das nicht da sein sollte.

naltatis, 2024-05-16:
Mist, ich vermute, das hängt mit https://github.com/evcc-io/evcc/pull/13882 zusammen. Dann gabs doch nen Grund, warum ich damals `width` und nicht `flex-basis` verwendet hatter. Ich schau mal ob ich da ne gute Lösung finde.

naltatis, 2024-05-16:
Ok, war dann doch was Banales. Hatte die Änderung an der CSS Transition vergessen. Nun läufts wieder flüssig 😎

closed 2024-05-16 by naltatis