evcc-io/evcc #13935: UI: Ladeleistung beim Loadpoint wird falsch dargestellt?

mrbgfromv, 2024-05-16:
### Describe the bug

Hallo zusammen,

nutze evcc schon länger erfolgreich, habe jetzt die Version 0.126.2 compiliert und mit den bisherigen Einstellungen vermeintlich lauffähig auf einer Synology installiert. Leider startet der Ladevorgang im Modus PV nicht mehr zuverlässig.
Bei der Fehlersuche ist mir aufgefallen, dass in der UI nach dem Klick auf die Einstellungen beim Lademodus die Leistungen (in leicht grau dargestellt) nicht zu stimmen scheinen - siehe Bild.
Wenn ich von 3 - phasig auf 1 - phasig bei nicht angeschlossenem Auto wechsle, sieht man, wie kurz andere Leistungen eingeblendet und dann wieder überschrieben werden: 3p und 1p Leistungen werden gleich dargestellt. Heute war das Auto schon mal angeschlossen, da blieben die Einstellungen mWn bestehen. 
Zum Starten der Ladung musste ich erst auf "MIN+PV" schalten, konnte dann aber wieder auf "PV" zurück wechseln.
Die Wallbox ist eine Heidelberg WB, 3 - phasig am HAK. Das Auto lädt nur langsam 1-phasig (Smart EQ).
Was da ursächlich mit wem zusammenhängt oder nur zufällig ist, weiß ich noch nicht.
Die Sache mit den Leistungen macht mich jedenfalls stutzig.
Habe im Moment das Gefühl, dass die Ladung nicht beginnt, weil irrtümlich 4 kW Überschuss gefordert werden (die gab es nicht), obwohl 2 gereicht hätten (die gab es).
Beste Grüße!

### Steps to reproduce

1. Auf Einstellungen hinter Lademodus klicken
2. 1p Leistung beobachten
3. zwischen 3p und 1p wechseln
...

[image]

### Configuration details

```yaml
Vermeintlich relevanter Auszug:
loadpoints:
- title: Carport
  charger: Heidelberg
  mode: pv
  #resetOnDisconnect: true # set defaults when vehicle disconnects --> Jetzt in GUI
  #phases: 1 # ev phases (default 3) --> Jetzt in GUI
  enable: # pv mode enable behavior
    delay: 2m # threshold must be exceeded for this long
  disable: # pv mode disable behavior
    delay: 10m # threshold must be exceeded for this long
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  #minCurrent: 6 # minimum charge current (default 6A) --> Jetzt in GUI
  #maxCurrent: 16 # maximum charge current (default 16A) --> Jetzt in GUI

chargers:
- name: Heidelberg
  type: Heidelberg
  uri: 192.168.1.86
  id: 1
  rtu: false

meters:
- name: Zaehlerkasten
  type: custom
  power:
   source: mqtt
   topic: energie/summenbezug
```

### Log details

```text
Auto ist am WE leider unterwegs

evcc charger
[main  ] INFO 2024/05/16 18:23:55 evcc
[main  ] INFO 2024/05/16 18:23:55 using config file: /etc/evcc.yaml
[db    ] INFO 2024/05/16 18:23:55 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/05/16 18:23:55 connecting evcc-1107584169 at tcp://192.168.1.2:1883
Power:          0W
Energy:         3237.9kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 235V 235V 236V
Charge status:  A
Enabled:        false

evcc meter
[main  ] INFO 2024/05/16 18:26:06 evcc
[main  ] INFO 2024/05/16 18:26:06 using config file: /etc/evcc.yaml
[db    ] INFO 2024/05/16 18:26:06 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/05/16 18:26:06 connecting evcc-1772540841 at tcp://192.168.1.2:1883
Zaehlerkasten
-------------
Power: 431W
```

### What type of operating system are you running?

Docker container

### Version

unknown shorthand flag: 'v' in -v

VolkerK62, 2024-05-16:
Deine Wallbox übermittelt die Phasenspannungen, dadurch wird automatisch die Anschlussart (1p oder 3p) erkannt.
Wenn das Auto nur 1p kann, dann beim vehicle `phases: 1` eintragen. Dann brauchst du in den Settings gar nichts anpassen.

ps: `guardduration` gibt es nicht mehr

mrbgfromv, 2024-05-16:
OK, danke für den Hinweis.
Trotzdem sollte doch die minimale Ladeleistung kleiner als 4kW sein, oder?

mrbgfromv, 2024-05-16:
Direkt mal alles ausprobiert und die Konfiguration umgeschrieben:
Der Smart wird jetzt als Standard "erkannt" und auch die oben angesprochene fehlerhafte Ladeleistung wird richtig angezeigt.
Es ist ja doch einiges an mir vorbei gegangen, auch wenn ich zwischendurch immer mal wieder ein Update eingespielt habe. Deshalb umsomehr vielen Dank für die extrem schnelle und professionelle Hilfestellung.

Damit dürfte das Ticket geschlossen werden können.

StefanSchoof, 2024-05-16:
Das schließen kannst du selbst machen (und dann muss sich das evcc Team auch nicht mehr hier rum kümmern)

mrbgfromv, 2024-05-16:
Geschlossen.

closed 2024-05-16 by mrbgfromv