evcc-io/evcc #13941: evcc Nissan Leaf Login failed: code not found

gorman553, 2024-05-17:
### Describe the bug

cannot query my Nissan Leaf.

### Steps to reproduce

1. evcc -l vehicle
2.
3.
...

### Configuration details

```yaml
vehicles:
- name: leaf
  type: nissan
  title: Leaf # display name for UI
  capacity: 40 # kWh
  user: email@addr.com
  password: "my-secret-words"
  cache: 15m
  phases: 1
```

### Log details

```text
pi@raspberrypi:~ $ /usr/bin/evcc -l vehicle
[main  ] INFO 2024/05/17 08:16:21 evcc 0.126.2 (940ce81a)
[main  ] INFO 2024/05/17 08:16:21 using config file: /home/pi/evcc.yaml
[main  ] ERROR 2024/05/17 08:16:23 creating vehicle leaf failed: cannot create vehicle type 'nissan': login failed: code not found
[main  ] FATAL 2024/05/17 08:16:23 failed configuring site: missing either grid or pv meter
[main  ] FATAL 2024/05/17 08:16:23 will attempt restart in: 5m0s
[server] FATAL 2024/05/17 08:16:23 unlinkat /tmp/evcc: operation not permitted
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.126.2 (940ce81a)

gorman553, 2024-05-17:
Hi Andi,

since yesterday (16.05.2024) I can not query my nissan leaf. the days before it worked fine. The Android Nissan Tool works, so I think Nissan changed something in their setup :(.

thanks for having a look on it.

gorman

docolli, 2024-05-17:
Yes, I can confirm that the carwings API might have changed, since I cannot query my Leaf either. I can see the problem with the carwings implementation of FHEM also. Android APP works, so no server problem on Nissan side.

Someone needs to run apktool and find out the changes, hopefully they only have changed some URLs, like 4 years ago: https://gitlab.com/tobiaswkjeldsen/dartcarwings/-/issues/15

andig, 2024-05-17:
Gibts eine Lösung mit der das noch funktioniert? Welche?

docolli, 2024-05-17:
Sniffen geht nicht, die Aufrufe werden über Port 443 geroutet. Da komm ich nicht ran. Zumindest sehe ich, dass die Aufrufe noch an `gdcportalgw.its-mo.com` gesendet werden, leider keinen weiteren Teil der URL.

Auch beim Versuch die NissanConnected APK aus einem APK-Store zu laden und zu dekompilieren finde ich keine URL. Habe das aber zum ersten Mal versucht. Kann also an mir liegen.

docolli, 2024-05-17:
@andig 
Nissan hat nur die Base URL geändert -> https://gdcportalgw.its-mo.com/api_v230317_NE
Ich glaube wir sind damit die Ersten mit dieser Info 😎

geert-claes, 2024-05-17:
same problem (with an Ariya)
```
[main  ] INFO 2024/05/17 20:11:28 evcc 0.124.10
[main  ] INFO 2024/05/17 20:11:28 using config file: /etc/evcc.yaml
[db    ] INFO 2024/05/17 20:11:29 using sqlite database: /root/.evcc/evcc.db
[main  ] ERROR 2024/05/17 20:11:31 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'nissan': login failed: code not found
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'nissan': login failed: code not found
Capacity: 63.0kWh
Features: [Offline]
```

docolli, 2024-05-17:
@andig 
Fix in carwings go library ist raus: https://github.com/joeshaw/carwings/commit/cf29a185820cba1e10d83a2de859ea8012243fd3

Ich habe dennoch einen fork von carwings gemacht und habe versucht einen PR zu machen, der auf mein Repository verweist. Bin aber zu unerfahren, das bekomme ich nicht hin. 😟

docolli, 2024-05-18:
Test mit neuer Bibliothek von joeshaw (github.com/joeshaw/carwings v0.0.0-20240517194654-cf29a185820c) unter Windows erfolgreich:
[image]

andig, 2024-05-18:
Carwings ist im master aktualisiert. Ich sehe lokal keinen Unterschied, kann aber am Testfahrzeug liegen.

docolli, 2024-05-18:
>Nightly baut.
d.h. ich kann die binary nachher bei cloudsmith runterladen und testen?

docolli, 2024-05-18:
Ok, Paket ist fertig und erfolgreich getestet:
evcc 0.126.3 (562c702c)

[image]

geert-claes, 2024-05-18:
```
[main  ] INFO 2024/05/18 11:26:46 evcc 0.126.3 (562c702c)
[main  ] INFO 2024/05/18 11:26:46 using config file: /etc/evcc.yaml
[main  ] INFO 2024/05/18 11:26:46 starting ui and api at :7070
[db    ] INFO 2024/05/18 11:26:46 using sqlite database: /root/.evcc/evcc.db
[main  ] ERROR 2024/05/18 11:26:46 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'nissan': login failed: Unauthorized: Authentication Failed
```

andig, 2024-05-18:
Tja...

geert-claes, 2024-05-18:
Not working for me.

gorman553, 2024-05-18:
For me also not working, sorry...

pi@raspberrypi:~ $ evcc -l debug vehicle
[main  ] INFO 2024/05/18 11:56:49 evcc 0.126.3 (562c702c)
[main  ] INFO 2024/05/18 11:56:49 using config file: /home/pi/evcc.yaml
[db    ] INFO 2024/05/18 11:56:49 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] ERROR 2024/05/18 11:56:50 creating vehicle leaf failed: cannot create vehicle type 'nissan': login failed: code not found
.Soc:     vehicle not available: cannot create vehicle type 'nissan': login failed: code not found
Capacity: 40.0kWh
Features: [Offline Retryable]

docolli, 2024-05-18:
Strange.... 🤔
```sh
Mai 18 11:56:31 fhem-heizung evcc[19475]: [main  ] INFO 2024/05/18 11:56:31 evcc 0.126.3 (562c702c)
Mai 18 11:56:31 fhem-heizung evcc[19475]: [main  ] INFO 2024/05/18 11:56:31 using config file: /etc/evcc.yaml
Mai 18 11:56:31 fhem-heizung evcc[19475]: [main  ] INFO 2024/05/18 11:56:31 starting ui and api at :7070
Mai 18 11:56:32 fhem-heizung evcc[19475]: [db    ] INFO 2024/05/18 11:56:32 using sqlite database: /var/lib/evcc/evcc.db
Mai 18 11:56:41 fhem-heizung evcc[19475]: [site  ] INFO 2024/05/18 11:56:41 site config:
Mai 18 11:56:41 fhem-heizung evcc[19475]: [site  ] INFO 2024/05/18 11:56:41   meters:      grid ✓ pv ✓ battery ✓
Mai 18 11:56:41 fhem-heizung evcc[19475]: [site  ] INFO 2024/05/18 11:56:41     grid:      power ✓ energy ✗ currents ✗
Mai 18 11:56:41 fhem-heizung evcc[19475]: [site  ] INFO 2024/05/18 11:56:41     pv 1:      power ✓ energy ✗ currents ✗
Mai 18 11:56:41 fhem-heizung evcc[19475]: [site  ] INFO 2024/05/18 11:56:41     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
Mai 18 11:56:41 fhem-heizung evcc[19475]: [site  ] INFO 2024/05/18 11:56:41   vehicles:
Mai 18 11:56:41 fhem-heizung evcc[19475]: [site  ] INFO 2024/05/18 11:56:41     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Mai 18 11:56:41 fhem-heizung evcc[19475]: [site  ] INFO 2024/05/18 11:56:41     vehicle 2: range ✓ finish ✗ status ✓ climate ✓ wakeup ✗
Mai 18 11:56:41 fhem-heizung evcc[19475]: [site  ] INFO 2024/05/18 11:56:41     vehicle 3: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-1  ] INFO 2024/05/18 11:56:41 loadpoint 1:
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-1  ] INFO 2024/05/18 11:56:41   mode:        pv
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-1  ] INFO 2024/05/18 11:56:41   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-1  ] INFO 2024/05/18 11:56:41   meters:      charge ✓
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-1  ] INFO 2024/05/18 11:56:41     charge:    power ✓ energy ✓ currents ✓
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-2  ] INFO 2024/05/18 11:56:41 loadpoint 2:
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-2  ] INFO 2024/05/18 11:56:41   mode:        off
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-2  ] INFO 2024/05/18 11:56:41   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-2  ] INFO 2024/05/18 11:56:41   meters:      charge ✓
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-2  ] INFO 2024/05/18 11:56:41     charge:    power ✓ energy ✓ currents ✗
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-3  ] INFO 2024/05/18 11:56:41 loadpoint 3:
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-3  ] INFO 2024/05/18 11:56:41   mode:        off
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-3  ] INFO 2024/05/18 11:56:41   charger:     power ✓ energy ✗ currents ✗ phases ✗ wakeup ✗
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-3  ] INFO 2024/05/18 11:56:41   meters:      charge ✓
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-3  ] INFO 2024/05/18 11:56:41     charge:    power ✓ energy ✗ currents ✗
Mai 18 11:56:41 fhem-heizung evcc[19475]: [site  ] DEBUG 2024/05/18 11:56:41 set buffer soc: 70
Mai 18 11:56:41 fhem-heizung evcc[19475]: [site  ] DEBUG 2024/05/18 11:56:41 set buffer start soc: 75
Mai 18 11:56:41 fhem-heizung evcc[19475]: [site  ] DEBUG 2024/05/18 11:56:41 set priority soc: 40
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-1  ] DEBUG 2024/05/18 11:56:41 phase timer inactive
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-1  ] DEBUG 2024/05/18 11:56:41 pv timer inactive
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-1  ] INFO 2024/05/18 11:56:41 vehicle updated: unknown -> Nissan Leaf
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-1  ] DEBUG 2024/05/18 11:56:41 max charge current: 6A
Mai 18 11:56:41 fhem-heizung evcc[19475]: [site  ] WARN 2024/05/18 11:56:41 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
Mai 18 11:56:41 fhem-heizung evcc[19475]: [site  ] DEBUG 2024/05/18 11:56:41 ----
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-1  ] DEBUG 2024/05/18 11:56:41 charge power: 4888W
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-1  ] DEBUG 2024/05/18 11:56:41 charge currents: [23 0 0]A
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-1  ] DEBUG 2024/05/18 11:56:41 charge voltages: [220 237 235]V
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-1  ] DEBUG 2024/05/18 11:56:41 detected connected phases: 3p
Mai 18 11:56:41 fhem-heizung evcc[19475]: [lp-1  ] DEBUG 2024/05/18 11:56:41 charge total import: 3708.847kWh
Mai 18 11:56:42 fhem-heizung evcc[19475]: [lp-1  ] DEBUG 2024/05/18 11:56:42 charger status: C
Mai 18 11:56:42 fhem-heizung evcc[19475]: [lp-1  ] DEBUG 2024/05/18 11:56:42 charge start energy: 3.71e+03kWh
Mai 18 11:56:42 fhem-heizung evcc[19475]: [lp-1  ] INFO 2024/05/18 11:56:42 car connected
Mai 18 11:56:42 fhem-heizung evcc[19475]: [lp-1  ] DEBUG 2024/05/18 11:56:42 pv timer elapse
Mai 18 11:56:42 fhem-heizung evcc[19475]: [lp-1  ] DEBUG 2024/05/18 11:56:42 pv timer inactive
Mai 18 11:56:42 fhem-heizung evcc[19475]: [lp-1  ] DEBUG 2024/05/18 11:56:42 charge total import: 3708.847kWh
Mai 18 11:56:42 fhem-heizung evcc[19475]: [lp-1  ] DEBUG 2024/05/18 11:56:42 charge start energy: 3.71e+03kWh
Mai 18 11:56:42 fhem-heizung evcc[19475]: [lp-1  ] INFO 2024/05/18 11:56:42 start charging ->
[... cut ...] "78"}},"PluginState":"CONNECTED","CruisingRangeAcOn":"175000","CruisingRangeAcOff":"188000","TimeRequiredToFull200_6kW":{"HourRequiredToFull":"3","MinutesRequiredToFull":"0"},"NotificationDateAndTime":"2024\/05\/18 09:56","TargetDate":"2024\/05\/18 09:56"}}
Mai 18 12:01:29 fhem-heizung evcc[19498]: [carwings] TRACE 2024/05/18 12:01:29 POST https://gdcportalgw.its-mo.com/api_v230317_NE/gdc/BatteryStatusCheckRequest.php
```

Bei mir läuft "linux" unter "arm" Architektur (Raspi 4).
Welche habt ihr?

andig, 2024-05-18:
Erfreulicherweise fehlt ja jegliche relevante Information 😰

docolli, 2024-05-18:
Please do a log with debug infos and search for this line:
`[carwings] TRACE 2024/05/18 12:01:29 POST https://gdcportalgw.its-mo.com/api_v230317_NE/gdc/BatteryStatusCheckRequest.php`

I want to verfiy which URL your evcc carwings is using.

gorman553, 2024-05-18:
ich bin auch unter raspberry Pi 4 arm (Raspbian/buster)

ein strings /usr/bin/evcc | grep api_v230317_NE   matched.. also die neue API ist enthalten

was mich aber etwas wundert.

wenn ich ein strace -s 2048 -f -o /tmp/strace evcc -l debug vehicle laufen lasse... dann kann ich im strace den Aufruf auf die URL nicht finden .... strange ... ich hätte erwartet dass ich da im strace irgendwo den Aufruf finde

@andig was genau brauchst Du an debug Info, dann kann ich versuchen das bereit zu stellen?

Danke

gorman553, 2024-05-18:
Was mir eben noch auffällt.

ich nutze template: nissan
und nicht template: carwings

ist die URL nur für carwings evtl. geändert?

docolli, 2024-05-18:
Ich bin bei `template: carwings`. Guter Hinweis!

andig, 2024-05-18:
Wir machen hier mal zu. Bitte neues Issue mit relevanten Informationen.

andig, 2024-05-18:
Also: behoben wurde Carwings. Bei Nissan hat sich nichts geändert.

andig, 2024-05-18:
https://github.com/Tobiaswk/dartnissanconnect/commit/e0301993e872a63c059c3afaa258b413d435fe0e

andig, 2024-05-18:
Nightly baut

closed 2024-05-18 by andig

gorman553, 2024-05-18:
so type: nissan funktioniert wieder.

Vielen lieben Dank @andig

geert-claes, 2024-05-18:
Confirmation : working now.

tm-107, 2024-05-18:
Auch hier läuft es mit der unstable wieder ... Danke!

docolli, 2025-01-12:
I have also problems and I have also observed that this weekend even the Nissan App could not connect. Maybe there are problems/work in progress on Nissan side...

DaFcz, 2025-01-13:
The connection by “Nissan EV” app is working, but sometimes it ends with message:
"The service cannot be provided. Please try again or contact NISSAN".
Next attempt is usually successful.

docolli, 2025-01-13:
On my side the "Nissan EV" app is not connecting. Even after next attempt. So I guess Nissan has a problem and we should be patient and try again day by day...

docolli, 2025-01-14:
Today the Nissan App and evcc is working again on my side!

DaFcz, 2025-01-17:
Thank you for info. It is working now also on my side.

docolli, 2025-01-28:
@DaFcz I still had problems, but there's a fix for carwings library out (https://github.com/evcc-io/evcc/commit/ad9b6ad3afc384d70cf26187c58f010643d57f99) that is already integrated into the nightly build. Now it works again smoothly.

https://github.com/joeshaw/carwings/issues/53
https://cloudsmith.io/~evcc/repos/unstable/packages/

DaFcz, 2025-02-21:
@docolli Thank you, I have check it with version 0.200.1 it is stable now.

antonka2000, 2025-07-12:
mit 0.204.5 kann ich mich leider aktuell nicht mit nissan verbinden. 
wenn ich mit BN und PW direkt bei Nissan Connect anmelde, funktioniert der login.
bei EVCC leider nicht:  ```
cannot create vehicle type 'nissan': login failed: Unauthorized: Authentication Failed

testweise habe ich carwings installiert. Unter carwings kann ich mich anmelden und auch die Batteriewerte auslesen!

Bei carwings musste ich noch die Region eingeben. Kann es sein, dass diese bei EVCC auch irgendwo hinterlegt werden muss??

vehicles:
  - name: my_car
    type: nissan
    title: Leaf
    user: xxx@yyy.zz
    password: Xyyy123456
    capacity: 39
    cache: 15m