evcc-io/evcc #13942: falsche Anzeige Ladevorgänge

haalvi, 2024-05-17:
### Describe the bug

ich habe falsche Angaben der Lademenge in kwh. Habe die csv für aktuellen Monat heruntergeladen und sehe falsche Subtraktionen: Energieangabe ist ungleich Differenz Zählerstände von go-e. Im go-e Protokoll stehen weitere Ladungen nach Unterbrechungen mangels PV-Strom, die in evcc nicht auftauchen bzw. aufaddiert sind.
[session-2024-05.csv](https://github.com/evcc-io/evcc/files/15344482/session-2024-05.csv)
[089818.csv](https://github.com/evcc-io/evcc/files/15344499/089818.csv)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Anlage Log
```

### Log details

```text
[session-2024-05.csv](https://github.com/evcc-io/evcc/files/15344482/session-2024-05.csv)
```

### What type of operating system are you running?

Windows

### Version

v0124.10

StefanSchoof, 2024-05-17:
Ist das #13726?

haalvi, 2024-05-17:
Sieht so aus. Ich habe die logfiles von evcc und go-e verglichen, die ich auch angehängt habe.

haalvi, 2024-05-17:
...aber ich habe keine identische Einträge, sondern nur fehlende

StefanSchoof, 2024-05-17:
Ich denke das hier, aber auch eine ähnliche Log wie dort gefragt benötigt wird.

andig, 2024-05-17:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-05-17 by andig

phrozen77, 2024-05-17:
Das mit den falschen Angaben geladene Menge vs. Anzeige der Energiemenge der Wallbox zieht sich schon eine Weile.

Passiert das bei dir (hauptsächlich) nach einem oder mehreren Wechseln der Phasen (also von 1p -> 3p und hin und her)?

#8503 #12980 #10589 #10452 sind so Kandidaten die ich auf die schnelle gefunden habe :(

haalvi, 2024-05-17:
ich habe keine Phasenwechsel - es passiert wenn mangels PV Strom die Ladung unterbrochen wird und dann weitergeführt wird.