evcc-io/evcc #13947: Problem zu KEBA evcc.yaml: cannot create keba charger: listen udp :7090: bind: address already in use

die4M, 2024-05-17:
### Describe the bug

Hallo Zusammen, ich bekomme, wie manche user zuvor auch die Fehlermeldung bei meiner KEBA, das "bind: address already in use". evcc läuft ansonsten einwandfrei, die KEBA wallbox läd auch wunderbar. Woher kommt dieser Fehler, und wie bekommt man ihn weg? Ich habe auch geprüft, evcc ist nur einmal installiert.
[image]

aktive Verbindungen:
[image]

Danke euch!

### Steps to reproduce

Siehe oben

### Configuration details

```yaml
chargers:

- name: wallbox
  type: template
  template: keba
  host: 192.168.2.53

- name: wallbox2
  type: template
  template: easee
  user: xxx
  password: xxx
  charger: xxx

loadpoints:

- title: Hof KEBA
  charger: wallbox
  vehicle: my_car
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 32
  meter: my_charger

- title: Garage
  charger: wallbox2
  mode: pv
  phases: 0
```

### Log details

```text
[main  ] FATAL 2024/05/17 11:14:02 could not open port- check that evcc is not already running (cannot create charger 'wallbox': cannot create charger type 'template': cannot create charger type 'keba-udp': listen udp :7090: bind: address already in use)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.126.3

StefanSchoof, 2024-05-17:
In deiner netstat Ausgabe ist doch schon eine laufende evcc Instanz (PID 29450). Evcc kann nur einmal zur Zeit ausgeführt werden. Also erst die andere Instanz stoppen.

die4M, 2024-05-17:
@StefanSchoof : danke für die Rückmeldung. Ich bin Laie auf dem Gebiet und Linux kenne ich mich auch nicht aus. Mein Verständnis war, dass PID 29450 mein aktuell laufendes evcc Programm ist, es aber nicht schafft auf die KEBA zuzugreifen da sie schon anderweitig genutzt wird, richtig? Wenn nicht, wie stoppe ich die PID 29450? "sudo systemctl stop evcc" stoppt ja das ganze Programm?! Ich benutze auch kein HA von Keba was ich in anderen Beträgen gelesen habe.
Dankeschön

die4M, 2024-05-17:
@andig : wieso completed? Das Problem besteht weiterhin

StefanSchoof, 2024-05-17:
Ich denke das fällt unter der Note aus dem bug Report den du ausgefüllt hast:

> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.

Am besten machst du einmal in den Diskussion ein Thema auf. Bitte auch ein bisschen genauer Beschrieben, was du eigentlich gerade erreichen willst. Siehe dazu https://docs.evcc.io/docs/faq#etwas-funktioniert-nicht-was-nun