evcc-io/evcc #13949: Ford login

juessi, 2024-05-17:
### Describe the bug

Hallo, ich habe gerade auf die aktuelle Version geupdatet, habe aber immer noch das problem mit dem Ford login. 
Wolke durchgestrichen beim Fahrzeug
"Fahrzeug war nicht ereicbar. Versuche EVCC neu zu starten"
Habe EVCC schon mehrfach neu gestartet, genau wie ein neustart von Homeassistant.
EVCC läuft als Addon bei Homeassistant
[main ] INFO 2024/05/17 14:16:54 evcc 0.126.3
[main ] INFO 2024/05/17 14:16:54 using config file: /config/evcc.yaml
[main ] INFO 2024/05/17 14:16:54 starting ui and api at :7070
[main ] ERROR 2024/05/17 14:17:01 creating vehicle Ford failed: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed.
Könne sie mir da weiterhelfen bei der Fehlmeldung?
meine evcc.yaml:
# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: Ford
    type: template
    template: ford
    title: Ford Kuga Hybrid
    user: user@xxxxxx.de
    password: xxxxxxxxxxxx
    vin: WF0Fxxxxxxxxxxxxxx
    capacity: 11 # Akkukapazität in kWh # Optional 

Danke für die Hilfe

### Steps to reproduce

1.
2.
3.
...
update auf v0.1263
Homeassistant 2024.05.1
neustart EVCC und Homeassistant

### Configuration details

```yaml
# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: Ford
    type: template
    template: ford
    title: Ford Kuga Hybrid
    user: user@xxxxxx.de
    password: xxxxxxxxxxxx
    vin: WFxxxxxxxxxxxxxxx
    capacity: 11 # Akkukapazität in kWh # Optional
```

### Log details

```text
Error log
[main ] ERROR 2024/05/17 14:17:01 creating vehicle Ford failed: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed 
Info log:
[main ] INFO 2024/05/17 14:16:54 evcc 0.126.3
[main ] INFO 2024/05/17 14:16:54 using config file: /config/evcc.yaml
[main ] INFO 2024/05/17 14:16:54 starting ui and api at :7070
[db ] INFO 2024/05/17 14:16:54 using sqlite database: /data/evcc.db
[mqtt ] INFO 2024/05/17 14:16:54 connecting evcc-1530637014 at tcp://192.168.178.10:1883
[main ] ERROR 2024/05/17 14:17:01 creating vehicle Ford failed: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed
[lp-1 ] WARN 2024/05/17 14:17:01 deprecated: mincurrent setting is ignored, please remove
[lp-1 ] WARN 2024/05/17 14:17:01 deprecated: maxcurrent setting is ignored, please remove
[lp-1 ] WARN 2024/05/17 14:17:01 deprecated: phases setting is ignored, please remove
[site ] INFO 2024/05/17 14:17:02 site config:
[site ] INFO 2024/05/17 14:17:02 meters: grid ✓ pv ✓ battery ✗
[site ] INFO 2024/05/17 14:17:02 grid: power ✓ energy ✓ currents ✗
[site ] INFO 2024/05/17 14:17:02 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/05/17 14:17:02 vehicles:
[site ] INFO 2024/05/17 14:17:02 vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1 ] INFO 2024/05/17 14:17:02 loadpoint 1:
[lp-1 ] INFO 2024/05/17 14:17:02 mode: pv
[lp-1 ] INFO 2024/05/17 14:17:02 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2024/05/17 14:17:02 meters: charge ✓
[lp-1 ] INFO 2024/05/17 14:17:02 charge: power ✓ energy ✓ currents ✓
[lp-1 ] INFO 2024/05/17 14:17:02 vehicle updated: unknown -> Ford Kuga Hybrid
[site ] WARN 2024/05/17 14:17:02 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[lp-1 ] INFO 2024/05/17 14:17:03 car connected 
debug log:
[site ] DEBUG 2024/05/17 14:44:12 ----
[lp-1 ] DEBUG 2024/05/17 14:44:12 charge power: 0W
[lp-1 ] DEBUG 2024/05/17 14:44:12 charge currents: [0 0 0]A
[site ] DEBUG 2024/05/17 14:44:13 pv power: 498W
[site ] DEBUG 2024/05/17 14:44:13 grid meter: 295W
[site ] DEBUG 2024/05/17 14:44:13 site power: 295W
[lp-1 ] DEBUG 2024/05/17 14:44:13 charge total import: 1085.436kWh
[lp-1 ] DEBUG 2024/05/17 14:44:13 charger status: B
[lp-1 ] DEBUG 2024/05/17 14:44:13 pv charge current: 0A = 0A + -1.28A (295W @ 1p)
[site ] DEBUG 2024/05/17 14:44:22 ----
[lp-1 ] DEBUG 2024/05/17 14:44:22 charge power: 0W
[lp-1 ] DEBUG 2024/05/17 14:44:22 charge currents: [0 0 0]A
[site ] DEBUG 2024/05/17 14:44:23 pv power: 496W
[site ] DEBUG 2024/05/17 14:44:23 grid meter: 295W
[site ] DEBUG 2024/05/17 14:44:23 site power: 295W
[lp-1 ] DEBUG 2024/05/17 14:44:23 charge total import: 1085.436kWh
[lp-1 ] DEBUG 2024/05/17 14:44:23 charger status: B
[lp-1 ] DEBUG 2024/05/17 14:44:23 pv charge current: 0A = 0A + -1.28A (295W @ 1p)
```

### What type of operating system are you running?

Linux

### Version

v0.1263

StefanSchoof, 2024-05-17:
ist im nigthly gefixt, siehe #13845

juessi, 2024-05-17:
Hallo Stefan, ich kann in Homeassistant nur die aktulle version updaten leider nicht das nightly.

StefanSchoof, 2024-05-17:
Gerade gemerkt, der Fix sollte in https://github.com/evcc-io/evcc/releases/tag/0.126.3 sein.

Klappt es wenn du ein `domain: de` einfügst?

juessi, 2024-05-17:
Habe gerade domain: de eingefügt und alles neu gestartet, leider ist die Wolke noch durch gestrichen und im Error log noch die gleiche Meldung

andig, 2024-05-17:
Und com?

juessi, 2024-05-17:
leider auch nicht anbei ausschnit aus der yaml.
info log:
[main ] INFO 2024/05/17 15:43:02 evcc 0.126.3
[main ] INFO 2024/05/17 15:43:02 using config file: /config/evcc.yaml
[main ] INFO 2024/05/17 15:43:02 starting ui and api at :7070
[db ] INFO 2024/05/17 15:43:02 using sqlite database: /data/evcc.db
[mqtt ] INFO 2024/05/17 15:43:02 connecting evcc-165387620 at tcp://192.168.178.10:1883
[main ] ERROR 2024/05/17 15:43:10 creating vehicle Ford failed: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed
[site ] INFO 2024/05/17 15:43:11 site config:
[site ] INFO 2024/05/17 15:43:11 meters: grid ✓ pv ✓ battery ✗
[site ] INFO 2024/05/17 15:43:11 grid: power ✓ energy ✓ currents ✗
[site ] INFO 2024/05/17 15:43:11 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/05/17 15:43:11 vehicles:
[site ] INFO 2024/05/17 15:43:11 vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1 ] INFO 2024/05/17 15:43:11 loadpoint 1:
[lp-1 ] INFO 2024/05/17 15:43:11 mode: pv
[lp-1 ] INFO 2024/05/17 15:43:11 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2024/05/17 15:43:11 meters: charge ✓
[lp-1 ] INFO 2024/05/17 15:43:11 charge: power ✓ energy ✓ currents ✓
[lp-1 ] INFO 2024/05/17 15:43:11 vehicle updated: unknown -> Ford Kuga Hybrid
[lp-1 ] INFO 2024/05/17 15:43:12 car connected 
vehicles:
  - name: Ford
    type: template
    template: ford
    title: Ford Kuga Hybrid
    user: user@xxxxxxx.de
    password: xxxxxxxxxxxx
    vin: WFxxxxxxxxxxxxxx
    domain: com
    capacity: 11 # Akkukapazität in kWh # Optional

GrimmiMeloni, 2024-05-17:
Ich habe mir das gerade schonmal zwei Stunden angeschaut.
Ich hab einen MITM Trace von meiner Ford App. Leider bringt mich das nicht weiter.
Der Self-Assert Schritt geht schief, mir ist aber nicht klar warum.

andig, 2024-05-18:
@juessi probier mal bitte ioBroker. Wenn das nicht geht bitte dort ein Issue.

juessi, 2024-05-18:
Hallo Andig, ich habe kein iobroker, ich nutze Homeassistant.

andig, 2024-05-18:
Und gibts da eine funktionierende Lösung?

juessi, 2024-05-18:
Nur mit der FordPass integration werden die Daten vom Ford in Homeassistant angezeigt.
Das ist die Integration: 
FordPass
https://github.com/itchannel/fordpass-ha

Velines, 2024-05-19:
Bei der FordPass-Integration gibt es aber auch Probleme, und die anvisierte Lösung klingt leider nicht wirklich gut: https://github.com/itchannel/fordpass-ha/issues/488#issuecomment-2119075884

weindler, 2024-05-19:
Also bei mir funktioniert die ganze Sache mit der Ford Schnittstelle auch nicht mehr, habe nun die Endung de und com ausprobiert, leider beides keinen Erfolg, obwohl daß die Endung com im iobroker funktioniert.

andig, 2024-05-19:
Das war ja meine Frage ganz oben. Also ioBroker geht?

weindler, 2024-05-19:
Jepp, hier wurde von @Ta2K (https://github.com/TA2k/ioBroker.ford/blob/master/README.md) der Adapter geändert, so daß man selbst in der Konfiguration die Auswahl hat die Endung einzustellen. Hier habe ich auf com gestellt und seitdem läuft es, war vor 3 Tagen. Aber wie gesagt hier in der yaml habe ich es auch versucht mit der domain: com, dies hat nicht funktioniert

andig, 2024-05-19:
Da wurde nur reverted. Damit müsste `com` noch laufen. Oder eben eine alte evcc Version. Kann ich mangels Credentials nicht prüfen.

weindler, 2024-05-19:
Habe ich schon probiert, bis zurück zur 0.126.1, mit dieser ging es leider auch nicht.

GrimmiMeloni, 2024-05-19:
Das hat nichts mit evcc zu tun. Doengrade ist hier sinnlos. 
Ford hat den Akamai Bot Manager auf seinen Schnittstellen aktiviert. Da wir keinen Headless Browser verwenden, können wir die Java Script Challenges nicht generieren, und somit scheitern wir nun.

weindler, 2024-05-19:
Aaaah, ok, da habe ich bei Ta2k auch schon was darüber gelesen, ist sozusagen ein Sicherheits Bot der die Abfragen auf die Api verhindert, sehe ich das so richtig? Das ist aber wirklich blöd, und das heißt nun für die Ford Benutzer das wars mit evcc, ich hoffe nicht.

GrimmiMeloni, 2024-05-19:
Ford hat eine neue API veröffentlicht. Diese muss  angebunden werden.
Ich habe da gestern mal draufgeschaut, bin aber nicht weit gekommen da ich schon bei der Registrierung mangels Ford EV scheitere. 

Ein PR dazu wäre super.

GrimmiMeloni, 2024-05-19:
Kurzer Zusatz: das Problem ist nicht die API sondern der Login Flow.
@andig würde es ggf. Sinn machen die Ford Integration auf ein Token in der Config umzustellen? Ggf. könnten sich Ford User ein Token über Ihren Browser besorgen.

weindler, 2024-05-19:
Auf jeden Fall danke euch für die unermüdliche Arbeit auch an Pfingsten, irgendwie wird es hoffentlich eine Lösung geben, wenn ich euch irgendwie helfen kann, dann her damit. Habe eine Kuga PHEV mit der Ford Pass Connect App, bin auch im Developer Portal von Ford angemeldet, sollte euch das was helfen.

Chris591, 2024-05-19:
es sollte dann aber der refresh token sein - der gild länger aber soweit ich das sehe müssten wir auch eine client_secret und client_id nutzen
oder den code dynamisch per Konfiguration vom Auto abfragen um einen Token zu bekommen.
also dann so : 
https://fordconnect.cv.ford.com/common/login/?make=F&application_id=AFDC085B-377A-4351-B23E-5E1D35FB3700&client_id=<clientID>&response_type=code&state=123&redirect_uri=<evccURL2getTheCode>
Aber ich finde nichts wie das mit der ClientId und ClientSecret gedacht ist. Persönlich oder pro App - aber wie schützt man diese das sie nicht von anderen Apps genutzt werden und dann die Limitierung von den requests uns wieder ein Problem macht.

andig, 2024-05-19:
Ford hat ein offizielles API, allerdings noch nicht für die EU: https://developer.ford.com/apis/fordconnect#apiReferenceContent. Anscheinend kann man sich selbst registrieren. Sobald das verfügbar würde könnten wir das supporten.

weindler, 2024-05-19:
Da kannst dich auch jederzeit registrieren, auf sign in klicken und dann "i don´t have an account" klicken und dann das formular ausfüllen. und dann hat man zugriff auf die api dokumentation.

andig, 2024-05-19:
Und was mache ich dann damit wenn sie nicht verwendet werden kann???

weindler, 2024-05-19:
warum sollte sie nicht verwendet werden können? irgendwie muß ja die handy app auch auf die api zugreifen können obwohl ich in der eu bin, oder verstehe ich da was falsch?

GrimmiMeloni, 2024-05-19:
Die offizielle Aussage von Ford ist, dass die neue API vorerst nur in Nord Amerika funktioniert. 

Die Ford App nutzt daher sehr wahrscheinlich noch die alte bestehende API.

weindler, 2024-05-19:
Ok, verstehe, und in der alten api ist das problem wie oben beschrieben mit dem  Akamai Bot Manager?

Chris591, 2024-05-19:
Ich kann sie Nutzen auch wenn da nur Nord Amerika steht - aber wie gesagt - ClientID und ClientSecret - sind Persönlich oder für die App die man bauen will??
Aber sie ist auch limitiert das man nicht zuviele zugriffe machen kann - aber ist das jetzt auf die Source IP oder die ClientID bezogen???
Sind noch viele fragen offen.

weindler, 2024-05-19:
Wie gesagt in iobroker sind auch alle Daten da, da kamen die letzten Wochen immer mehr Datenpunkte dazu. Auch hier ist die Api nutzbar auch in Verwendung mit der Google Map, daß man sieht wo das Auto gerade steht. Auch beim heutigen laden mit dem go-e Charger wurden die Punkte upgedatet. Also funktioniert so wie wenn sich nichts geändert hätte, daß mit dem Client ID usw. da kann ich euch nicht helfen daß ist mir zu hoch weiß nicht mal worauf sich das bezieht.:(

Velines, 2024-05-19:
@weindler die Client-ID ist vermutlich für alle die gleiche. Ich hab zumindest die selbe im Ford Developer Account wie hier in der URL aus der API-Referenz: 

> https://fordconnect.cv.ford.com/common/login/?make=F&application_id=AFDC085B-377A-4351-B23E-5E1D35FB3700&client_id=30990062-9618-40e1-a27b-7c6bcb23658a&response_type=code&state=123&redirect_uri=https%3A%2F%2Flocalhost%3A3000&scope=access

Das Secret ist dann vermutlich anders? Meins beginnt mit RHO8Q und endet mit 6AAcZ-.

Ansonsten kann ich bestätigen, dass ich derzeit auf die API zugreifen kann, auch wenn sie nur für Nordamerika gedacht sein sollte. Aufruf von https://api.mps.ford.com/api/fordconnect/v3/vehicles/{{vehicleId}} liefert auf jeden Fall die gewünschten Daten, u.a. Batteriestand und Reichweite.

andig, 2024-05-19:
In der Ford Doku steht nix über Zugangsdaten. Woher bekommt man die? Individuell fürs eigene Fahrzeug?

Velines, 2024-05-19:
In der "Getting Started with FordConnect.docx" ist Schritt 4 das Account Linking, ich vermute ich hab da jetzt meinen Nutzer-Account mit meinem Entwickler-Account gelinkt. Wenn ich die oben erwähnte URL eingebe muss ich mit meinem Nutzer-Account einloggen und dann das Auto auswählen.

Die Tokens hole ich mir dann über `https://dah2vb2cprod.b2clogin.com/914d88b1-3523-4bf6-9be4-1b96b4f6f919/oauth2/v2.0/token?p=B2C_1A_signup_signin_common` , da ist die client_id und das client_secret dabei (vom Dev-Account).

andig, 2024-05-19:
Ich sehe im dev account keine client credentials.

StefanSchoof, 2024-05-19:
`b2clogin.com` ist meines Wissens nach Azure Active Directory B2C. https://learn.microsoft.com/en-us/azure/active-directory-b2c/overview

Velines, 2024-05-19:
@andig bin mir unsicher, was Du meinst. Ich hab Zugangsdaten für den Dev-Account und für meinen Nutzer-Account (Fordpass-App).
Auf developer.ford.com seh ich dann folgendes:
[image]

Das ist alles was ich habe, und das scheint ja auch zu reichen.

andig, 2024-05-19:
Yep. Bei mir gibts kein Dshboard. Frag mich, wie Ford sich das für Anwendungsentwickler vorstellt. Finde auch keine Doku zum Oauth setup, nur zu den Api endpoints. Das reicht leider nicht.

andig, 2024-05-19:
Erwartung an Ford: sinnvolle und vollständige Doku. Einfach mal bei Tesla schauen. Ist kein Hexenwerk…

Velines, 2024-05-19:
So richtig fertig wirkt das auf der Seite von Ford noch nicht :(

Chris591, 2024-05-19:
ja da bin ich auch hängengeblieben.
aber das heißt ja wir müssten nur die URL aufrufen beim einrichten vom Ford und die URL von EVCC angeben damit der code am ende an EVCC weitergeben wird. Mit dem code kann man dann einen Token wie schon implementiert bekommen und einen Refresh Token der dann 90 Tage gültig ist. mit diesem kann man dann einen neuen Token bekommen und so weiter.
Aber für diese Token Geschichte braucht man auch einen ClientSecret - aber da gibt es auch keine Infos drüber.

weindler, 2024-05-20:
So wie alles bei Ford ist, sorry wenn es auch nicht hier hingehört, aber die app wird immer schlechter von monat zu monat, was hat man in der app am anfang alles gesehen, nur mehr ein bruchteil dessen, das mit dem laden über die app ist auch nur grottig, sorry aber das ist alles von vor 20 jahren so kommt mir das vor. Genau so daß mit den ganzen Rückrufen, da bekommst eine Mitteilung in der App daß was nicht stimmt und dann rufst die Autofirma (Original Ford) die wissen das ja, aber sie haben seit Monaten nicht die passende Software, das ist alles der Wahnsinn.

weindler, 2024-05-21:
@andig hier ist die dokumentation, was ich finden haben können.
[Ford Developer 3 PROD Collection.zip](https://github.com/evcc-io/evcc/files/15391924/Ford.Developer.3.PROD.Collection.zip)
[Getting Started with FordConnect.docx](https://github.com/evcc-io/evcc/files/15391926/Getting.Started.with.FordConnect.docx)

andig, 2024-05-21:
> Das ClientSecret habe ich bei der Registrierung bzw. der ersten Anmeldung auf dem Ford Developer Portal erhalten.

Ich nicht :( Bevor wir hier Aufwand rein stecken möchte ich erstmal das Ziel verstehen. Z.B. ein Account für evcc (analog Tesla) oder ein Account pro User (analog Mercedes)?

Chris591, 2024-05-21:
Hab mal mit Ford geschrieben und die Möglichkeit das wir ein ClientID und Secret nehmen für alle gibt es noch nicht - kommt vielleicht aber versuch ich noch rauszufinden.
Aber im Augenblick gild jeder User braucht einen eigenen Satz.
Auch die art und weise wie man sich anmeldet ist wohl nur für Developer - also keine User die nicht wissen was sie tun können das im Augenblick nutzen.

weindler, 2024-05-21:
> > Das ClientSecret habe ich bei der Registrierung bzw. der ersten Anmeldung auf dem Ford Developer Portal erhalten.
> 
> Ich nicht :( Bevor wir hier Aufwand rein stecken möchte ich erstmal das Ziel verstehen. Z.B. ein Account für evcc (analog Tesla) oder ein Account pro User (analog Mercedes)?

das kannst du dir im developer portal holen die id und die secret da gibt es einen link, und dann kann man es nachträglich erstellen lassen. Aber ich gebe euch recht das sollte schon hand und Fuss haben, ist ja auch mit Arbeit verbunden.

andig, 2024-05-21:
Bei mir gibts keinen Link…

Max112233445566, 2024-05-22:
> > Das Secret ist dann vermutlich anders? Meins beginnt mit RHO8Q und endet mit 6AAcZ-.
ClientId und Secret sind bei mir identisch.

Max112233445566, 2024-05-22:
> Bei mir gibts keinen Link…

developer.ford.com -> APIs -> View API Reference  -> Request Access
Dann erhälst du ClientID und Secret und hast unter My Developer Account dann auch die Ansicht My Dashboard

andig, 2024-05-22:
[image]

Max112233445566, 2024-05-22:
Kein Login zur Zeit möglich unter developer.ford.com

[image]

https://api.mps.ford.com/dmapiresource/v1/login
502 Server Error

Max112233445566, 2024-05-24:
Die Fordpass Integration für Home Assistant läuft wieder mit der alten API. Evtl. kann man sich da etwas abschauen.
https://github.com/itchannel/fordpass-ha/blob/1.69/custom_components/fordpass/fordpass_new.py

weindler, 2024-05-25:
Auch die iobroker Integration läuft wieder siehe github projekt. Hier ist es glaube ich die Public Api, aber da könnte mit Sicherheit @TA2k mehr dazu sagen.

https://github.com/TA2k/ioBroker.ford/tree/v1.0.1

andig, 2024-05-25:
Dann bräucht ich mal komplette credentials für info@evcc.io

weindler, 2024-05-25:
@andig sind raus

andig, 2024-05-26:
Seid ihr in der Lage den "account linking" Prozess soweit durchzuführen, dass ihr Token und RefreshToken erhaltet? 

@weindler falls ja- könntest Du auch die beiden bitte teilen? Der Rest sollte dann einfach sein.

weindler, 2024-05-26:
> Seid ihr in der Lage den "account linking" Prozess soweit durchzuführen, dass ihr Token und RefreshToken erhaltet?
> 
> @weindler falls ja- könntest Du auch die beiden bitte teilen? Der Rest sollte dann einfach sein.

Ich weiß jetzt leider nicht genau was du damit meinst, ich habe dir mal eine mail geschickt wo ich denke daß es daß vielleicht sein kann.

andig, 2024-05-26:
Siehe Mail- der Code muss noch in Tokens umgewandelt werden.

andig, 2024-05-26:
I'm able to retrieve the list of vehicles, but not the VIN:

    Www-Authenticate: Bearer error="insufficient_scope", error_description="The request requires higher privileges than provided by the access token.", error_uri="https://tools.ietf.org/html/rfc6750#section-3.1"

VIN is needed to compare against config. Is anybody able to get the VIN?

weindler, 2024-05-26:
Ja, das ist mir auch schon aufgefallen im iobroker daß die Vin abgeht, vielleicht kommt sie ja noch mit rein, man könnte eventuell nach dem Nicknamen das einrichten, vielleicht hinterlegen daß beim Nicknamen = Vin *********************, denn der Nickname wird mit ausgelesen.

[16 more replies]

closed 2024-05-26 by andig