evcc-io/evcc #13953: KEBA failsafe timeout using evcc configuration with modbus

phL9r, 2024-05-17:
### Describe the bug

My Dad has a KEBA P30 and the charging worked good. Now starting with the version 0.126.2, I cannot start the EVCC Server after configuring the KEBA in the setup menue. Looking into the generatet yaml-file, I saw that evcc generated a keba-modbus template which gives out the following error after starting. Once turning back to template "KEBA" and deleting the other generated stuff like port etc., the server started again and the wallbox works.

### Steps to reproduce

1. evcc configure
2. choose KEBA P30 as a wallbox
3. input all information
4.  start the server
5. get the error message
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxx

interval: 30s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: e3dc-rscp
  usage: grid
  host: 192.168.178.39
  port: 5033
  user: xxx
  password: xxx
  key: xxx
  name: grid1
- type: template
  template: e3dc-rscp
  usage: pv
  host: 192.168.178.39
  port: 5033
  user: xxx
  password: xxx
  key: xxx
  name: pv2
- type: template
  template: e3dc-rscp
  usage: battery
  host: 192.168.178.39
  port: 5033
  user: xxx
  password: xxx
  key: xxx
  name: battery3

chargers:
- type: template
  template: keba
  host: 192.168.178.176
[... 763 lines cut ...]
[lp-1 ] DEBUG 2024/05/17 19:42:03 charge power: 0W
time="2024-05-17T19:42:03+02:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
[site ] DEBUG 2024/05/17 19:42:03 pv power: 194W
time="2024-05-17T19:42:03+02:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-05-17T19:42:03+02:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
[site ] DEBUG 2024/05/17 19:42:03 battery soc: 0%
[site ] DEBUG 2024/05/17 19:42:03 battery power: 333W
time="2024-05-17T19:42:03+02:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
[site ] DEBUG 2024/05/17 19:42:03 grid meter: 115W
[site ] DEBUG 2024/05/17 19:42:03 site power: 448W
[lp-1 ] DEBUG 2024/05/17 19:42:03 charger status: B
[lp-1 ] DEBUG 2024/05/17 19:42:03 pv charge current: 0A = 0A + -0.649A (448W @ 3p)
time="2024-05-17T19:42:24+02:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-05-17T19:42:24+02:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-05-17T19:42:24+02:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-05-17T19:42:24+02:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
```

### What type of operating system are you running?

Linux

### Version

0.126.3

StefanSchoof, 2024-05-17:
Hat die Wallbox die nötige Firmware Version (https://docs.evcc.io/docs/devices/chargers#keba-kecontact-p20-p30-cx-series)?

Warum hast du ein neues configure ausgeführt, wenn es vorher funktioniert hat?

phL9r, 2024-05-17:
Es gab Probleme mit der Steuerung der Batterie in E3DC mit dem template. Da erschien immer ein Schloss-Symbol vor der Batterie und unterbrach Entladung oder Ladung. Mit der neuen Version funktioniert es seitdem, daher die neue config. Jetzt habe ich das Problem, dass die Wallbox trotz Einstellung auf 1A Ladeleistung 3phasig mit 6A lädt und es sich nicht ändert. Die Firmware lautet P30 v 3.10.27 (210105-174852). Wie man ein Update zieht muss ich mir dann wohl mal anschauen, auch wie ich das Register 3 setze, denn die KEBA hat keine Möglichkeiten im Webinterface irgendetwas umzustellen.