evcc-io/evcc #13957: JLR not working due to API update

ignazhabibi, 2024-05-18:
### Describe the bug

Same as last two times. JLR changed their API again to break third party app connections

### Steps to reproduce

-

### Configuration details

```yaml
I-Pace added via UI
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

andig, 2024-05-18:
Might be https://github.com/ardevd/jlrpy/pull/117?

andig, 2024-05-18:
I'd need credentials for testing, please mail to info@evcc.io

ignazhabibi, 2024-05-19:
Credentials still needed?

andig, 2024-05-19:
I don't know. Merge since I can't test.

ignazhabibi, 2024-05-19:
> Might be [ardevd/jlrpy#117](https://github.com/ardevd/jlrpy/pull/117)?

Think this is the newest one and it seems it is hard to work around this time. 
https://github.com/ardevd/jlrpy/issues/131

derkroesink, 2024-05-26:
i only have live credentials, but can help with debug.
please let me know if you need it.

andig, 2024-06-09:
Closed in https://github.com/evcc-io/evcc/commit/a85d0dc58d65b217f5fa1144e8c76425e06a4e34.

The API is dead, no workarounds known.

closed 2024-06-09 by andig