evcc-io/evcc #13963: Pracht Alpha XT Doppellader Fragen

alexraven1970, 2024-05-18:
Hi,
hoffentlich wird der Thread mit meiner Frage reaktiviert ...

Ich habe gerade eine Pracht Alpha XT mit zwei Ladepunkten installiert bekommen. 
Dazu den passenden PCC als Steuereinheit und den PDCC80 Stromzähler. 
Den PNI, das Netzwork interface haben wir nicht bestellt, es hiess das PV Überschussladen funktioniere nur ohne.
Der Hintergedanke war, alles aus einer Hand, sollte ja ohne Schwierigkeiten laufen.... denkste. 
Er läd PV nur 3phasig mit 6A, keinerlei Automatik .. mein Elektriker klärt das gerade mal wieder mit Pracht. 

Ich habe aber inzwischen so meine Zweifel, daß nurPV mit der Pracht Hard & Software funktioniert. 

Falls nicht, möchte ich aber eine Alternativlösung in der Hinterhand haben.
EVCC kenn ich schon - wir hatten vorher eine KEBA. In der Doku hab ich die Pracht gefunden. 
"Pracht Alpha XT, XT+, Mono XT, Mono XT+, PNI" 

Die Frage, die ich mir stelle, benötige ich den PNI von Pracht, nacgdem das Teil im Namen genannt ist? 
Oder wäre ein alternativer Modbusadapter besser?
Wenn ich die posts von @eismarkt in diesem Thread lese scheint mit auch das PNI eine Fehlerquelle zu sein.
Momentan habe ich keine Kommunikationsmöglichkeit mit der WB.

Ein Schütz zum Phasenabwurf ist verbaut, der funktioniert auch wenn ich den Phasenabwurf manuell im PCC auslöse. Kann evcc den Impuls auch softwareseitig auslösen? 
Weiterhin ist im Stromkasten ein Shelly 3EMPro Energiezähler. 

Ich habe ein technisches Grundverständniss, yaml hab ich schon angepasst, mit dem Terminal kann ich arbeiten, wenn ich die Befehle kenne. 
Elektrik/Hardware ... einen Rechner zusammenbauen und installieren, kein Thema; Hauselektrik - das fass ich nicht an! 
Dafür kommt der Elektriker. Seit dem ich als Kind mal eine Stromschlag bekommen habe, hab ich davor größten Respekt.

Welche PE11 (1 pro Ladepunkt vermute ich) soll ich besorgen?
Bekommen wir evcc zum Laufen? Wer hat mit der WB die meiste Erfahrung und würde mich bei der Installation unterstützen?

Danke für Eure Rückmeldung
Alexandra

_Originally posted by @alexraven1970 in https://github.com/evcc-io/evcc/discussions/9872#discussioncomment-9477228_

closed 2024-05-18 by andig

alexraven1970, 2024-05-19:
Hi @andig 
was hab ich falsch gemacht, daß Du den Thread geschlossen hast? 
Gibt es ein unterforum je WB, wo ich das Thema aufnehmen kann? 
Grüße
Alexandra

andig, 2024-05-19:
Du hattest ja schon eine Diskusssion dazu. Einmal reicht.