evcc-io/evcc #13967: Fordpass connection stopped working in evcc

kr3at0r, 2024-05-18:
### Describe the bug

After a few weeks of no issues evcc can no longer login to fordpass.

my ford account is not locked and is working fine in both fordpass and home assistant.

User and pass are correct

please help

### Steps to reproduce

Nothing special… i just start evcc addon in homeassistant

### Configuration details

```yaml
Standard config:
name: Kuga
type: template
template: ford
title: Kuga
user: ***@gmail.com
pass: ****
capacity: 12
```

### Log details

```text
[main ] ERROR 2024/05/18 17:36:15 creating vehicle Kuga failed: cannot create vehicle type 'template': cannot create vehicle type 'ford': login failed: self-assert failed
```

### What type of operating system are you running?

HomeAssistant Add-on

juessi, 2024-05-18:
Mit der Core Version 2024.4.1 Supervisior 2024.05.1 geht es bei mir alle Daten vom Ford werden angezeigt

juessi, 2024-05-18:
Also die FordPass integration nicht die EVCC

juessi, 2024-05-18:
okay, i have not updated to version 2024.05.4 i had problems with automatic integration. The integration with FordPass still works with this  Core Version 2024.4.1

TobiasHuber1980, 2024-05-18:
siehe #13949

kr3at0r, 2024-05-18:
Already tried the domain part but i had no luck

closed 2024-05-18 by andig