evcc-io/evcc #13971: Mercedes Integration - Token error

Nele02, 2024-05-18:
### Describe the bug

Ich habe meinen Mercedes in das evcc eingebunden. Leider wird dieser jedoch als nicht erreichbar angezeigt.

Die folgende Fehlermeldung wird geloggt:
 
```
[mercedes] DEBUG 2024/05/18 21:12:38 identity.NewIdentity - token not valid - Add expiry
[mercedes] DEBUG 2024/05/18 21:12:38 identity.NewIdentity - token not valid - database token check started
[mercedes] TRACE 2024/05/18 21:12:38 GET https://bff.emea-prod.mobilesdk.mercedes-benz.com/v2/vehicles
[main  ] ERROR 2024/05/18 21:12:38 creating vehicle ev2 failed: cannot create vehicle type 'template': cannot create vehicle type 'mercedes': cannot get vehicles: unexpected status: 400 (Bad Request)
```

Die Tokens wurden mit `evcc token ev2` generiert und in die evcc.yaml eingefügt. Auffällig ist, dass bei dem Aufruf von `evcc settings get` die Region beim Eintrag des Tokens fehlt, diese ist jedoch in der Config angegeben:

`mercedes.xxx@t-online.de-: xxx`

Ich nutze die Version: evcc version 0.126.3 (cee3cf79)

### Steps to reproduce

1. Fahrzeug zur config hinzufügen.
2. Tokens mit `evcc token ev2` generieren und in die Config einfügen.
3. evcc neustarten

### Configuration details

```yaml
- type: template
  template: mercedes
  user: xxx@t-online.de
  title: EQA
  icon: car
  region: EMEA
  accessToken: xxx
  refreshToken: xxx
  vin: W...
  capacity: 66.5
  phases: 3
  mode: minpv
  minCurrent: 6
  maxCurrent: 16
  name: ev2
```

### Log details

```text
[mercedes] DEBUG 2024/05/18 21:12:38 identity.NewIdentity - token not valid - Add expiry
[mercedes] DEBUG 2024/05/18 21:12:38 identity.NewIdentity - token not valid - database token check started
[mercedes] TRACE 2024/05/18 21:12:38 GET https://bff.emea-prod.mobilesdk.mercedes-benz.com/v2/vehicles
[main  ] ERROR 2024/05/18 21:12:38 creating vehicle ev2 failed: cannot create vehicle type 'template': cannot create vehicle type 'mercedes': cannot get vehicles: unexpected status: 400 (Bad Request)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.126.3 (cee3cf79)

andig, 2024-05-18:
Bitte mal komplette Zugangsdaten, Config und evcc.db an info@evcc.io

aklickermann, 2024-05-23:
> Bitte mal komplette Zugangsdaten, Config und evcc.db an [info@evcc.io](mailto:info@evcc.io)

Ist dabei etwas herausgekommen? Ich habe ein identisches Problem, darf ich meine Daten ebenfalls schicken?

matthiazzz, 2024-05-29:
I get a different error  when I try to update my mercedes token on the latest nightly and EMEA region:
`unexpected status: 418 (I'm a teapot)`

[image]

Only the NORAM region provides a token, but that is not where my car is located and thus does not work...

andig, 2024-05-29:
Can you try https://github.com/evcc-io/evcc/pull/14093?

cracyfloyd, 2024-05-29:
> I get a different error when I try to update my mercedes token on the latest nightly and EMEA region:
> `unexpected status: 418 (I'm a teapot)`

Same here !!!!!

matthiazzz, 2024-05-29:
> Can you try #14093?

Don't have a build env setup in my LXC. I need a compiled version to check.

closed 2024-05-29 by andig

deadrabbit87, 2024-06-01:
Also with https://github.com/evcc-io/evcc/pull/14093 not working: 
```
./evcc -c evcc.yaml token eqe -l trace
[main  ] INFO 2024/06/01 14:02:24 evcc 0.126.5 (aa4e26e3)
[main  ] INFO 2024/06/01 14:02:24 using config file: evcc.yaml
? Please enter your Mercedes ME user-account (e-mail or mobile) ************@gmx.de
? Choose your MB region: EMEA
[main  ] TRACE 2024/06/01 14:02:36 GET https://bff.emea-prod.mobilesdk.mercedes-benz.com/v1/config
[main  ] TRACE 2024/06/01 14:02:36 {"forceUpdate":{"status":"OK","storeUrl":"https://apps.apple.com/app/mercedes-me-2020/id1487652920"},"vehicleImagesBaseUrl":"https://eceprod2vehimages.blob.core.windows.net/fin-images"}
[main  ] TRACE 2024/06/01 14:02:36 POST https://bff.emea-prod.mobilesdk.mercedes-benz.com/v1/login
[main  ] TRACE 2024/06/01 14:02:37 {"emailOrPhoneNumber":"************@gmx.de","countryCode":"EN","nonce":"******-*****-****-****-********"}
--
{"isEmail":true,"username":"************@gmx.de"}
[main  ] FATAL 2024/06/01 14:02:37 unknown PinResponse - 200, result empty
```
evcc.yaml:
```
vehicles:
  - name: my_car
    type: template
    template: mercedes
    title: EQE
    user: *************@gmx.de
    region: EMEA
    accessToken:
    refreshToken:
    capacity: 50
```

christophhhh, 2024-06-04:
Same here:
`[main ] INFO 2024/06/04 23:00:14 evcc 0.126.6
[main ] INFO 2024/06/04 23:00:14 using config file: /evcccfg/evcc.yaml
[main ] INFO 2024/06/04 23:00:14 starting ui and api at :7070
[db ] INFO 2024/06/04 23:00:15 using sqlite database: /root/.evcc/evcc.db
[mercedes] DEBUG 2024/06/04 23:00:16 identity.NewIdentity - token not valid - Add expiry
[mercedes] DEBUG 2024/06/04 23:00:16 identity.NewIdentity - token not valid - database token check started
[main ] ERROR 2024/06/04 23:00:16 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'mercedes': cannot get vehicles: unexpected status: 400 (Bad Request) `

I checked the tokens twice, but still get that error

HiroP77, 2024-06-17:
same here:
[main ] INFO 2024/06/17 14:32:17 evcc 0.127.2
[main ] INFO 2024/06/17 14:32:17 using config file: /etc/evcc.yaml
[db ] INFO 2024/06/17 14:32:17 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2024/06/17 14:32:17 listening at :7070
[mercedes] DEBUG 2024/06/17 14:32:18 identity.NewIdentity - token not valid - Add expiry
[mercedes] DEBUG 2024/06/17 14:32:18 identity.NewIdentity - token not valid - database token check started
[main ] ERROR 2024/06/17 14:32:18 creating vehicle GLE failed: cannot create vehicle type 'template': cannot create vehicle type 'mercedes': cannot get vehicles: unexpected status: 400 (Bad Request)

andig, 2024-06-17:
@ReneNulschDE hast Du eine Idee?

ReneNulschDE, 2024-06-17:
@HiroP77 : May I ask you to create a new discussion and add me to it with `@ReneNulschDE` so that I get informed, then we can analyze it. Please share your config, short environment info and the log.

HiroP77, 2024-06-17:
done, its [14438](https://github.com/evcc-io/evcc/issues/14438)

Nele02, 2024-06-22:
Nachdem es bei mir nicht richtig funktioniert hat, habe ich ein Workaround mit dem ioBroker gebaut und den Mercedes als custom vehicle hinterlegt. Bisher funktioniert das sehr zuverlässig. Ich habe hier mal eine Anleitung dazu geschrieben: [evcc_mercedes_integration](https://github.com/Nele02/evcc_mercedes_integration)