evcc-io/evcc #13974: Nissan Ariya Template kein Connect

adobe-hills, 2024-05-19:
### Describe the bug

Hallo EVCC Team,

Bei dem Starten von EVCC wird das Fahrzeug Nissan Ariya nicht eingebunden. Das ganze hat letzte Woche/vorletzte Woche funktioniert. Dies war jedoch noch unter Windows zum Testen. Nach dem Wechsel auf einen PI hat er bei "evcc configure" noch die VIN übertragen, nun klappt wohl auch keine Anmeldung mehr. Mit welchem String wird die Autorisierung bei Nissan geprüft? Die LOG Meldung würde ja bedeuten, dass auf der Gegenseite gar nichts lauscht. 

cannot create vehicle type 'nissan': login failed: code not found

Ich habe auch das normale Template mit dem Leaf probiert, hier habe ich aber die gleiche Meldung. Damit hatte es unter Windows zumindest funktioniert. Das Kennwort bei Nissan Connect habe ich gewechselt, aber keine Änderung. Die Credentials habe ich in der evcc.yaml maskiert, die kann ich aber zum Testen zur Verfügung stellen, ich möchte diese nur nicht öffentlich hier zeigen.
Ich sehe gerade dass mit dem Fix https://github.com/evcc-io/evcc/issues/13941 das Leaf template geändert wurde. Habe es nun mit diesem Template probiert, klappt aber auch nicht.

Vielen Dank für eure Hilfe. 

### Steps to reproduce

1. 
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 9fa18bcf7fb5725bd74edaf79982fad8520866317f3bd0cc43165d71f73faeb6

interval: 20s # control cycle interval

meters:
- type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.2.107  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.2.107  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery2
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.2.105  
  name: grid3

chargers:
- type: template
  template: pulsarplus 
  timeout: 2m  
  name: wallbox5

vehicles:
- type: template
  template: nissan-ariya 
  title: Ariya
  user: *********** 
  password: *********  
  vin: ************* 
  capacity: 87  
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: ev4
  mode: pv

site:
  title: Klingel.7
  meters:
    grid: grid3
    pv:
    - pv1
    battery:
    - battery2

hems:
  type: sma
```

### Log details

```text
[main ] INFO 2024/05/19 09:15:34 evcc 0.126.3
[main ] INFO 2024/05/19 09:15:34 using config file: /etc/evcc.yaml
[main ] INFO 2024/05/19 09:15:34 starting ui and api at :7070
[db ] INFO 2024/05/19 09:15:34 using sqlite database: /var/lib/evcc/evcc.db
[ocpp-1] DEBUG 2024/05/19 09:15:36 waiting for chargepoint: 5m0s
[ocpp ] INFO 2024/05/19 09:16:38 charge point connected, registering: pulsarplus
[main ] ERROR 2024/05/19 09:16:44 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'nissan': login failed: code not found
[site ] WARN 2024/05/19 09:16:44 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[semp ] WARN 2024/05/19 09:16:44 SEMP_BASE_URL unspecified, using http://192.168.2.96:7070 instead
[site ] INFO 2024/05/19 09:16:44 site config:
[site ] INFO 2024/05/19 09:16:44 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/05/19 09:16:44 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2024/05/19 09:16:44 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/05/19 09:16:44 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site ] INFO 2024/05/19 09:16:44 vehicles:
[site ] INFO 2024/05/19 09:16:44 vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1 ] INFO 2024/05/19 09:16:44 loadpoint 1:
[lp-1 ] INFO 2024/05/19 09:16:44 mode: pv
[lp-1 ] INFO 2024/05/19 09:16:44 charger: power ✗ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/05/19 09:16:44 meters: charge ✗
[site ] DEBUG 2024/05/19 09:16:44 set battery discharge control: true
[lp-1 ] DEBUG 2024/05/19 09:16:44 phase timer inactive
[lp-1 ] DEBUG 2024/05/19 09:16:44 pv timer inactive
[lp-1 ] INFO 2024/05/19 09:16:44 vehicle updated: unknown -> Ariya
[site ] WARN 2024/05/19 09:16:44 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site ] DEBUG 2024/05/19 09:16:44 ----
[lp-1 ] DEBUG 2024/05/19 09:16:44 charge power: 0W
[site ] DEBUG 2024/05/19 09:16:45 pv power: 2403W
[site ] DEBUG 2024/05/19 09:16:45 battery soc: 98%
[site ] DEBUG 2024/05/19 09:16:45 battery power: -2101W
[site ] DEBUG 2024/05/19 09:16:45 grid meter: -4W
[site ] DEBUG 2024/05/19 09:16:45 grid powers: [-46 92 -51]W
[site ] DEBUG 2024/05/19 09:16:45 grid currents: [-0.532 0.942 -0.515]A
[site ] DEBUG 2024/05/19 09:16:45 site power: -2105W
[lp-1 ] DEBUG 2024/05/19 09:16:45 charger status: A
[lp-1 ] INFO 2024/05/19 09:16:45 car disconnected
[lp-1 ] DEBUG 2024/05/19 09:16:45 set session soc limit: 0
[lp-1 ] DEBUG 2024/05/19 09:16:45 set session energy limit: 0
[site ] DEBUG 2024/05/19 09:16:45 set battery mode: normal
[site ] DEBUG 2024/05/19 09:17:04 ----
[lp-1 ] DEBUG 2024/05/19 09:17:04 charge power: 0W
[site ] DEBUG 2024/05/19 09:17:04 pv power: 2403W
[site ] DEBUG 2024/05/19 09:17:05 battery soc: 99%
[site ] DEBUG 2024/05/19 09:17:05 battery power: -2120W
[site ] DEBUG 2024/05/19 09:17:05 grid meter: 8W
[site ] DEBUG 2024/05/19 09:17:05 grid powers: [-44 98 -47]W
[site ] DEBUG 2024/05/19 09:17:05 grid currents: [-0.53 0.961 -0.513]A
[site ] DEBUG 2024/05/19 09:17:05 site power: -2112W
[lp-1 ] DEBUG 2024/05/19 09:17:05 charger status: A
```

### What type of operating system are you running?

Linux

### Version

0.126.3  aber auch in der Vorgängerversion

andig, 2024-05-19:
Im nightly behoben.

closed 2024-05-19 by andig

adobe-hills, 2024-05-19:
Super vielen Dank, werde es nach dem Mittag gleich testen. Schöne Pfingsten 😀Grüße KarlVon meinem iPhone gesendet

adobe-hills, 2024-05-19:
eben probiert, funktioniert wunderbar.
DANKE !

An-Wi, 2024-05-19:
Ihr seid stark 💪
Auch bei mir ist seit Donnerstag früh (7 Uhr ging es noch, 10 Uhr nicht mehr) das Problem beim `template: nissan-ariya` aufgetreten.
Freitag Nacht hatte ich dank des Updates der "My Leaf"-App die Änderung von `client_id=a-ncb-nc-android-prod` herausgefunden, kam aber durch familiäre Verpflichtungen nicht dazu, es zu testen (wollte es nicht nur als Vermutung hier reinschreiben)...
...und nun sehe ich beim kurzen Blick in die issues, dass ihr das alles schon gelöst habt - Respekt & vielen Dank allen Mitwirkenden! 👍

Leider habe ich als "Windows"-User keine Möglichkeit, das nightly zu testen oder ich habe sie noch nicht gefunden?