evcc-io/evcc #13992: Token für Peugot nicht generierbar

Herbi54, 2024-05-20:
### Describe the bug

Da mir der PSA Car Controller zu umständlich ist, möchte ich wieder direkt auf Peugot "umsteigen". Ich möchte dass Auto über die GUI einbinden und habe daher "vehicles" aus der evcc.yaml entfernt. Bei dem Befehl "evcc token [name]" gibt es aber die unten aufgeführten Fehlermeldungen. Ich habe es mit mehreren Namen schon probiert.

### Steps to reproduce

1. Befehl eingeben  "evcc token [name ]
2.
3.
...

### Configuration details

```yaml
#vehicles:
#- name: my_car
#  type: template 
#  template: peugeot 
#  title: Peugeot e-208
#  user: xxx@xxx.de  
#  password: xxx
#  accessToken: [unbekannt]
#  refreshToken: [unbekannt]
#  vin: xxx  
#  capacity: 50  
#  phases: 1  
#  icon: car  
#  mode: pv  
#  minCurrent: 6  
#  maxCurrent: 16  
#  priority: 1
```

### Log details

```text
pi@raspberrypi:~ $ sudo evcc token my_car
[main  ] INFO 2024/05/20 17:26:51 evcc 0.126.3 (cee3cf79)
[main  ] INFO 2024/05/20 17:26:51 using config file: /etc/evcc.yaml
[main  ] FATAL 2024/05/20 17:26:51 vehicle not found, have []
pi@raspberrypi:~ $ sudo evcc token Peugeot
[main  ] INFO 2024/05/20 17:27:48 evcc 0.126.3 (cee3cf79)
[main  ] INFO 2024/05/20 17:27:48 using config file: /etc/evcc.yaml
[main  ] FATAL 2024/05/20 17:27:48 vehicle not found, have []
```

### What type of operating system are you running?

Linux

### Version

0.126.3 (cee3cf79)

0x3d13f, 2024-05-20:
Versuch mal `db:1` als Name für das Fahrzeug zu verwenden.

Herbi54, 2024-05-20:
Danke, aber das funktioniert leider auch nicht 😢

horstvogel, 2024-05-20:
Zum Erzeugen der Token muss das anscheinend in der yaml eingefügt sein, bei Dir ist das "aus kommentiert" Dann klappt bei mir der Aufruf. 
Habe allerdings keinen Peugeot

Herbi54, 2024-05-20:
Wenn ich es mit der yaml mache, kommt diese Meldung, obwohl ich evcc mit "sudo systemctl start evcc" starte:
pi@raspberrypi:~ $ evcc token my_car
[main  ] INFO 2024/05/20 19:52:11 evcc 0.126.3 (90285a62)
[main  ] INFO 2024/05/20 19:52:11 using config file: /etc/evcc.yaml
[main  ] FATAL 2024/05/20 19:52:11 

Found systemd service database at "/var/lib/evcc/evcc.db", evcc has been invoked with database "~/.evcc/evcc.db".
Running evcc with vehicles configured in evcc.yaml may lead to expiring the yaml configuration's vehicle tokens.
This is due to the fact, that the token refresh will be saved to the local instead of the service's database.
If you have vehicles with touchy tokens like PSA or Tesla, make sure to remove vehicle configuration from the yaml file.

If you know what you're doing, you can run evcc ignoring the service database with the --ignore-db flag.