evcc-io/evcc #14005: Template `tasmota`, `usage: pv` ohne Funktion

VolkerK62, 2024-05-21:
### Describe the bug

Beim Template `tasmota` ist der Parameter `usage: pv` ohne Funktion. Es werden nur 0-Werte geliefert.

### Steps to reproduce

.

### Configuration details

```yaml
- name: tasmota
  type: template
  template: tasmota
  host: 192.168.178.27
  usage: pv
```

### Log details

```text
wsl@DESKTOP-7RU7VSU:~$ evcc meter tasmota --log trace
[main  ] INFO 2024/05/21 18:05:47 evcc 0.126.3 (5855882e)
[main  ] INFO 2024/05/21 18:05:47 using config file: /home/wsl/evcc.yaml
[db    ] INFO 2024/05/21 18:05:47 using sqlite database: /home/wsl/.evcc/evcc.db
[db    ] TRACE 2024/05/21 18:05:47 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT * FROM `settings` 26 <nil>
[mqtt  ] INFO 2024/05/21 18:05:47 connecting evcc-1694281835 at tcp://raspberrypi:1883
[mqtt  ] DEBUG 2024/05/21 18:05:47 tcp://raspberrypi:1883 connected
[db    ] TRACE 2024/05/21 18:05:47 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%   device_id       %") -1 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT * FROM `config_details` WHERE `config_details`.`config_id` = 1 4 <nil>
[db    ] TRACE 2024/05/21 18:05:47 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 1 <nil>
[tasmota] TRACE 2024/05/21 18:05:47 GET http://192.168.178.27/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2024/05/21 18:05:47 {"StatusSNS":{"Time":"2024-05-21T17:05:49","SML":{"Power_curr":505,"Total_in":10782,"Total_out":19075}}}
Power:  0W
Energy: 0.0kWh
```

### What type of operating system are you running?

Linux

### Version

evcc 0.126.3 (5855882e)

thierolm, 2024-05-21:
Hast du einen separaten Netz-Zähler mit einem Hichi Tasmota IR Lesekopf für deine Solareinspeisung?
Die Tasmota `pv` Option ist für eine Tasmota-basierte PM Steckdose gedacht ...

thierolm, 2024-05-21:
Wenn du einen Netzzähler ausliest, dann musst du auch `grid` usage mit `tasmota-sml` Template nutzen.

VolkerK62, 2024-05-21:
ah, ok.   Dann ist alles klar.
Vor ein paar tagen kam das in einer Diskussion auf (Erzeugungszähler mit IR Lesekopf) und deshalb habe ich das mit meinem IR Lesekopf ausprobiert.
Es kommen nur bei `usage: grid` Werte, sonst immer `0`.

Da wäre dann ein entsprechender Verwendungshinweis in der Doku hilfreich.
Dann verstehe ich jetzt auch die Intention mit dem neuen `SML` Template.

closed 2024-05-21 by VolkerK62