evcc-io/evcc #14016: Gui: Restzeit per default als Standardanzeige setzen

manny4566, 2024-05-22:
Hi,

wenn ich das richtig sehe, wird bei jedem Ladevorgang die "Ladedauer" als Standard angezeigt in der GUI. Gibt es eine Möglichkeit in der Konfiguration festzuschreiben, welche der Optionen als Standard angezeigt werden soll bei jedem starten des Ladevorgangs? Möchte das gerne auf die "Restzeit" fest einstellen.
Falls nicht, würde ich mir das Feature gerne wünschen.
Oder gibt es die Möglichkeit bereits ?
Danke

manny4566, 2024-05-22:
> @naltatis per Default den vorherig ausgewählten Wert wieder selektieren?

Das ein Werter unterhalb dieses Menüs der default ist, wenn ein Ladevorgang startet:

[image]

So wie ich das immer gesehen habe, wird die Ladedauer aber per default angezeigt, wenn ein Ladevorgang startet.
Ich wähle dann immer Restzeit aus, was als weiterer Punkt dazu kommt, sobald ein Ladevorgang startet. Diesen würde ich gerne als default direkt aktiv haben.

naltatis, 2024-05-22:
Aktuell speichert die UI bereits im Browser bei welchem Ladepunkt du welche Metrik aktiv ausgewählt hast. Ist diese Metrik nicht verfügbar (Auto nicht verbunden, anderes Fahrzeug, ...) gibt es einen Fallback auf "Ladedauer". Sobald diese Metrik aber wieder da ist sollte diese automatisch wieder angezeigt werden. Wenn das bei dir nicht funktioniert, müsstest du einmal Details zum Reproduzieren zusammenstellen. Also welchen Browser und die Schritte, die du gemacht hast. Gerne auch einen Screenshot der Local Storage Ansicht aus den Browser Dev-Tools (siehe Beispiel).

[image]

manny4566, 2024-05-22:
ok Danke, ich gucke es mir an. Wird aber erst am Wochende klappen.

manny4566, 2024-05-25:
Hi,
also so sieht es aus, wenn ich den Ladevorgang starte und danach die evcc Seite Aufrufe im Browser.

[image]

Nachdem ich dann auf Restzeit wechsel:

[image]

Allerdings merkt er sich die Einstellung nicht bei bei mir. Beim nächsten mal wird wieder die Ladedauer per default angezeigt. 

Update:
Ich habe jetzt im Anschluss mal die Cookies und alles zu der Webseite gelöscht.
Jetzt bleibt die "Restzeit" stehen als Default, auch nach Browser Neustart.
Muss es mal beobachten, ob es jetzt vielleicht klappt in Zukunft.

naltatis, 2024-05-29:
Ok, LocalStorage sollte eigentlich, in der Browser Standardeinstellungen, relativ persistent sein. Aber gut, dass das jetzt schon mal klappt. Meld dich gerne hier noch mal, wenn es nicht wie erwartet funktioniert und wir das reproduzieren können. Bis dahin mach ich hier erstmal zu.

closed 2024-05-29 by naltatis