evcc-io/evcc #14023: Scheduled / One Time: "Discharge home battery to Car"

MarcelMertens, 2024-05-22:
**Is your feature request related to a problem? Please describe.**
If you want to discharge you home battery to your car you have to manual play around with max discharge settings and "schnell" mode

**Describe the solution you'd like**
Loading Mode "Discharge Home Battery" will drain home battery (max discharge power and low SoC configurable) to the car. 
Example: This morning i had 30% (~3kWh) left in my Huawei home battery storage and i want to charge my car with it before heading to work. During work hours my home batterie will charge up to 100% again. 
To drain the home batterie to my car i manual have to set the max amp to 7 (max discharge of home battery is 5kW) and select "Schnell".
At 3% SoC of the home battery i had to manually stop the charging process.
I guess their is something more automation possible.

**Additional context**
In combination with PV yield prediction (e.g. https://forecast.solar/) this can be used for a new charging plan option.
https://forecast.solar/ is also used by HomeAssistant PV yield prediction and at least for my home is some kind reliable.

theoretically scenario:
At some point (evening or early morning) i have xx % SoC (configurable) in my home battery. The PV yield prediction provides information that on the next/same day the pv yield is higher than yy kWh (configurable). If this is true, that discharge home battery to car.
Example for me: At 5am: If Home Battery is >30% and PV yield prediction is > 25kWh then discharge home battery to car until home battery is <10% SoC 

Personal Question: GitHub Issues in English or German?

StefanSchoof, 2024-05-22:
Sound similar to #12190

MarcelMertens, 2024-05-22:
Yes, general idea is identical. Merge if possible.

andig, 2024-05-23:
Then let's move the discussion to https://github.com/evcc-io/evcc/discussions/12190

closed 2024-05-23 by andig