evcc-io/evcc #14028: Fehlermeldung Seat Mii

pablopalec, 2024-05-22:
### Describe the bug

Online Abfrage PKW gestört. Konfiguration lief bereits über mehrere Wochen ohne Probleme. Habe die Installation heute von einem Raspberry 4 auf Raspberry 5 übertragen (neues System, evcc.yaml und evcc.db kopiert, Sponsortoken erneuert). Zunächst lief der Login von evcc über vw/seat, nun kommt eine Fehlermeldung.

### Steps to reproduce

1. Fehlermeldung im Log
2. Test per "evcc vehicle"
3. Test einer neuen evcc.yaml per "evcc configure"
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 3axxx
interval: 10s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.254.86  
  name: grid1
- type: template
  template: sma-hybrid 
  id: 4  
  host: 192.168.254.81  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sma-inverter-speedwire 
  usage: pv  
  host: 192.168.254.79  
  password: S0nnend1v1de  
  name: pv3
- type: template
  template: sma-hybrid 
  id: 4  
  host: 192.168.254.81  
  port: 502  
  usage: battery  
  modbus: tcpip  
  capacity: 6.5  
  watchdog: 30s  
  name: battery4

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.254.125  
  name: wallbox6
- type: template
  template: go-e-v3 
  host: 192.168.254.137  
  name: wallbox7

vehicles:
- type: template
  template: seat 
  title: Mii-electric  
  icon: car  
  user: xx@px.de  
  password: xxx  
  vin: xxx  
  capacity: 32  
  phases: 2  
  cache: 15m  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16  
  identifiers:
  - xxx 
  name: ev5

loadpoints:
- title: Hof
  charger: wallbox6
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
- title: Garage
  charger: wallbox7
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16

site:
  title: Fxxx
  meters:
    grid: grid1
    pv:
    - pv2
    - pv3
    battery:
    - battery4

hems:
  type: sma
  AllowControl: false
```

### Log details

```text
[main ] INFO 2024/05/22 16:00:47 evcc 0.126.3
[main ] INFO 2024/05/22 16:00:47 using config file: /etc/evcc.yml
[main ] INFO 2024/05/22 16:00:47 starting ui and api at :7070
[db ] INFO 2024/05/22 16:00:48 using sqlite database: /var/lib/evcc/evcc.db
[main ] ERROR 2024/05/22 16:00:49 creating vehicle ev5 failed: cannot create vehicle type 'template': cannot create vehicle type 'seat': invalid character 'i' looking for beginning of object key string 

 Title seatmii
 Username xx@pxx.de
 Password [? for help] ************
 Vehicle Identification Number xx
 Battery capacity in kWh 32

Testing the configuration of Seat CupraConnect Gen3 (Ateca, Leon, Formentor) ...
   Error: cannot create vehicle type 'seat': invalid character 'i' looking for beginning of object key string

 Testing Seat CupraConnect Gen3 (Ateca, Leon, Formentor) failed. Do you want to add it anyway? (y/N) 
The configuration was cancelled.
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.126.3

andig, 2024-05-22:
Terms & Conditions?

stefanjakobi, 2024-05-22:
Habe die gleiche Fehlermeldung nach dem Update von 0.126.0 auf 0.126.3, an der evcc.yaml habe ich nichts geändert, vor dem Update lief es mit unserem Cupra (template: seat) einwandfrei.

@andig "Terms & Conditions" bedeutet genau was?

TobiasHuber1980, 2024-05-22:
Geh mal auf die Seite von VW/Seat und schau ob du irgendwelchen (neuen) AGBs zustimmen kannst/musst....

stefanjakobi, 2024-05-22:
@TobiasHuber1980 Nichts zu finden, scheint alles okay zu sein. Cupra-App funktioniert auch. Trotzdem danke!

micw, 2024-05-22:
Ich hatte bei Skoda den Terms zuzustimmen. Hatte ich aber gemacht, bevor ich https://github.com/evcc-io/evcc/issues/14031 aufgemacht hatte - hat nix gebracht.

micw, 2024-05-22:
Woher kommt eigentlich "invalid character 'i' looking for beginning of object key string"? Kann man da mehr debugging einschalten? Wenn was "falsches" vom Server kommt, könnte man ja eine gescheite Fehlermeldung draus bauen.

MackKaufmann, 2024-05-22:
Same here mit Audi:

_creating vehicle ev2 failed: cannot create vehicle type 'template': cannot create vehicle type 'etron': invalid character 'i' looking for beginning of object key string_

andig, 2024-05-22:
Does ioBroker work?

andig, 2024-05-22:
Fixed in master

closed 2024-05-22 by andig

Joeknx, 2024-05-23:
Now my e-Golf is not working anymore. With 0.126.3 everything worked. 

I only have one error in the logs:
[main ] ERROR 2024/05/23 21:06:09 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': missing state

vistalba, 2024-06-13:
I running into the same issue today with pur Born. Log shows this:
```
[main  ] ERROR 2024/06/13 19:19:43 creating vehicle Born failed: cannot create vehicle type 'template': cannot create vehicle type 'cupra': cannot get vehicles: unexpected status: 404 (Not Found)
[site  ] WARN 2024/06/13 19:19:44 vehicle 'Born' does not support automatic detection
```

Cupra App and also the Homeassistant Integration are working well. So there should be no cloud ir account issue.