evcc-io/evcc #14031: Batteriestand Enyaq wird nicht mehr angezeigt

micw, 2024-05-22:
### Describe the bug

Der Batteriestand vom Enyaq wird seit ein paar Tagen nicht mehr angezeigt

### Steps to reproduce

1. Vehicle vom Typ "template: enyaq" anlegen
2. evcc starten
3. log anschauen
...

### Configuration details

```yaml
vehicles:
          - name: Enyaq
            title: Enyaq
            type: template
            template: enyaq
            user: xy@xy
            password: xyxy
            capacity: 80
            phases: 3
```

### Log details

```text
evcc [main  ] ERROR 2024/05/22 19:05:51 creating vehicle Enyaq failed: cannot create vehicle type 'template': cannot create vehicle type 'enyaq': invalid character 'i' looking for beginning of object key string
```

### What type of operating system are you running?

Linux

### Version

0.126.3 (7b8aeeff)

TobiasHuber1980, 2024-05-22:
Gleicher Fehler wie in #14028

TobiasHuber1980, 2024-05-22:
Beides... gleicher Text, dann wohl auch gleiche Ursache

closed 2024-05-22 by andig

micw, 2024-05-22:
Haben Seat und Skoda das gleiche System im Backend? Wenn nicht, würde ich eher annehmen unterschiedliche Probleme mit gleicher Fehlermeldung.

MackKaufmann, 2024-05-22:
Gleiche Fehlermeldung bei Audi: https://github.com/evcc-io/evcc/issues/14028#issuecomment-2125691415