evcc-io/evcc #14040: Wallbox Pulsar Plus & MIN+PV funktioniert nicht mehr

Nightbreezer, 2024-05-23:
### Describe the bug

Die Wallbox Pulsar Plus funktionierte immer einwandfrei über OCPP mit EVCC.

Seit dem Update der Wallbox auf 6.4.19 funktioniert der Min+PV Modus leider nicht mehr.

Es treten 2 verschiedene Probleme auf:

1. Der Ladevorgang startet, läuft auch eine Weile wie er soll, fängt dann aber irgendwann an mit konstant 11 kW zu laden, obwohl gar nicht so viel Sonne da ist und der Speicher auch leer ist. Also voll aus dem Netz. Er regelt auch nicht mehr runter.

z.B. heute: Um 10:25 Uhr laden gestartet. Ab 10:45 dann nur noch mit Fullspeed.

Interessanterweise zeigt EVCC dann bei mir die minimale Ladeleistung mit 4,1 kW an. Er lädt aber definitiv mit 11 kW, das sehe ich in der Wallbox App und das sieht man auch bei OUT Gesamt. Der Rest der Ladeleistung geht dann in den Verbrauch.

2. Der Ladevorgang startet gar nicht und es kommt ständig die Fehlermeldung: Garage: charger enable: ocpp message (3477658652): ProtocolError - validation of RemoteStartTransaction failed
Das Problem tritt sporadischer auf. Ein mehrmaliges Ein- und Ausstecken kann dieses Problem lösen.

Der Schnellmodus läuft einwandfrei. Ich kann auch die maximale Ladegeschwindigkeit im Schnellmodus über das EVCC regeln. Die Wallbox und EVCC sind auch ständig im Netz erreichbar.

[image]

### Steps to reproduce

Problem 1 tritt bei jedem Laden auf. Jedoch zu völlig unterschiedlichen Zeitpunkten, mal lädt er nach 1 Minute bereits mit 11kW, mal nach 30 mins.

Problem 2 tritt ab und zu auf. Ich kann keine Regelmäßigkeit feststellen.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  occp: trace

# unique installation id
plant: fd5a9e359ffe0d30ad2fcf6024b49d89b002e9f85c569932d7412a33cbfd6aad

interval: 30s # control cycle interval                              

meters:
- type: template
  template: solarmax-maxstorage
  id: 1
  host: 192.168.2.223
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: solarmax-maxstorage
  id: 1
  host: 192.168.2.223
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: solarmax-maxstorage
  id: 1
  host: 192.168.2.223
  port: 502
  usage: battery
  modbus: tcpip
[... 604 lines cut ...]
May 23 10:49:03 myroof evcc[1408914]: [lp-1  ] DEBUG 2024/05/23 10:49:03 vehicle soc: 69%
May 23 10:49:03 myroof evcc[1408914]: [lp-1  ] DEBUG 2024/05/23 10:49:03 vehicle range: 282km
May 23 10:49:03 myroof evcc[1408914]: [lp-1  ] DEBUG 2024/05/23 10:49:03 pv charge current: 0A = 6A + -12.6A (8678W @ 3p)
May 23 10:49:33 myroof evcc[1408914]: [site  ] DEBUG 2024/05/23 10:49:33 ----
May 23 10:49:33 myroof evcc[1408914]: [lp-1  ] DEBUG 2024/05/23 10:49:33 charge power: 4140W
May 23 10:49:33 myroof evcc[1408914]: [site  ] DEBUG 2024/05/23 10:49:33 pv power: 3656W
May 23 10:49:33 myroof evcc[1408914]: [site  ] DEBUG 2024/05/23 10:49:33 battery soc: 67%
May 23 10:49:33 myroof evcc[1408914]: [site  ] DEBUG 2024/05/23 10:49:33 battery power: 4800W
May 23 10:49:33 myroof evcc[1408914]: [site  ] DEBUG 2024/05/23 10:49:33 grid meter: 2956W
May 23 10:49:33 myroof evcc[1408914]: [site  ] DEBUG 2024/05/23 10:49:33 site power: 7856W
May 23 10:49:33 myroof evcc[1408914]: [lp-1  ] DEBUG 2024/05/23 10:49:33 !! session: chargeRater.chargedEnergy=48.6 - chargedAtStartup=0.0
May 23 10:49:33 myroof evcc[1408914]: [lp-1  ] DEBUG 2024/05/23 10:49:33 charger status: C
May 23 10:49:33 myroof evcc[1408914]: [lp-1  ] DEBUG 2024/05/23 10:49:33 soc estimated: 69.11% (vehicle: 68.00%)
May 23 10:49:33 myroof evcc[1408914]: [lp-1  ] DEBUG 2024/05/23 10:49:33 vehicle soc: 69%
May 23 10:49:33 myroof evcc[1408914]: [lp-1  ] DEBUG 2024/05/23 10:49:33 vehicle range: 282km
May 23 10:49:33 myroof evcc[1408914]: [lp-1  ] DEBUG 2024/05/23 10:49:33 pv charge current: 0A = 6A + -11.4A (7856W @ 3p)
```

### What type of operating system are you running?

Linux

### Version

0.126.3

andig, 2024-05-23:
Aussagekräftiges Logfile fehlt

closed 2024-05-23 by andig

Nightbreezer, 2024-05-23:
So ganz verstehe ich es leider nicht, ich habe das Logfile mit: 
sudo journalctl -u evcc -q -S "2024-05-23 10:25:00" -U "2024-05-23 10:50:00" erstellt. 

Es war aber zu zu groß, deswegen habe ich ab 10:29 Uhr eingefügt.

Genau in dem Zeitraum ist das Problem aufgetreten.

Was hätte ich denn besser machen können?

andig, 2024-05-23:
Du schreibst von Fehlermeldungen. Im Log ist davon nichts zu sehen.

andig, 2024-05-26:
[lp-1 ] ERROR 2024/05/09 08:59:05 charger enable: ocpp message (2537314205): ProtocolError - validation of RemoteStartTransaction failed

Das sieht sehr aus wie https://github.com/evcc-io/evcc/issues/14056, daher bitte https://github.com/evcc-io/evcc/issues/14056#issuecomment-2131116728

mrhund, 2024-07-04:
Ich habe exakt das gleich Problem, wie @Nightbreezer als Punkt 1 beschreibt. Spannender Weise sind keine Fehlermeldungen im OCPP Log, alles wird Accepted. Meine Vermutung ist also ein Problem in der Pulsar Firmware. Ich kommentiere trotzdem, weil ich das Verhalten weiter analysieren werde.

mrhund, 2024-07-05:
Ich hab das Thema nun heute wieder gehabt, vielleicht kann @andig kurz einen Blick darauf werfen und ihm fällt was auf:

Ich habe die Min + PV Ladesession um 10:20 Uhr gestartet und alles war "normal" - Stromstärke immer wieder automatisch durch EVCC angepasst, je nach PV-Ertrag - bis 12:27 Uhr. Dann feherhafte Umstellung der Pular von 6A auf 16A. Keine Fehler im Log, Anzeige in EVCC weiterhin 4,3KW (6A). Auffällig ist, dass kurz vorher (12:26 Uhr) die Stromstärke via EVCC 2 Mal auf 6A gesetzt wurde, jedes Mal mit "Accept" quittiert. Beim zweiten Mal wurde aber auf 16A . Einträge, die mir seltsam vorkommen habe ich fett markiert. 

Von unten nach oben zu lesen:

2024/07/05 12:28:15,stdout,"[ocpp  ] TRACE 2024/07/05 12:28:15 sent g001: [3,\"917449575\",{\"currentTime\":\"2024-07-05T10:28:15Z\"}]

2024/07/05 12:28:15,stdout,"[ocpp  ] TRACE 2024/07/05 12:28:15 recv g001: [2,\"917449575\",\"Heartbeat\",{}]

2024/07/05 12:28:14,stdout,[lp-1  ] DEBUG 2024/07/05 12:28:14 charge power: 4140W

**2024/07/05 12:27:50,stdout,"[lp-1  ] DEBUG 2024/07/05 12:27:50 pv charge current: min 6A > 0A (4370W @ 3p, battery: false)**

2024/07/05 12:27:50,stdout,[lp-1  ] DEBUG 2024/07/05 12:27:50 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)

2024/07/05 12:27:50,stdout,[lp-1  ] DEBUG 2024/07/05 12:27:50 vehicle range: 202km

2024/07/05 12:27:50,stdout,[lp-1  ] DEBUG 2024/07/05 12:27:50 vehicle soc: 58%

2024/07/05 12:27:50,stdout,[lp-1  ] DEBUG 2024/07/05 12:27:50 soc estimated: 58.09% (vehicle: 58.00%)

2024/07/05 12:27:50,stdout,[lp-1  ] DEBUG 2024/07/05 12:27:50 charger status: C

2024/07/05 12:27:50,stdout,[lp-1  ] DEBUG 2024/07/05 12:27:50 !! session: chargeRater.chargedEnergy=5.5 - chargedAtStartup=0.0

2024/07/05 12:27:50,stdout,[lp-1  ] DEBUG 2024/07/05 12:27:50 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)

2024/07/05 12:27:44,stdout,[lp-1  ] DEBUG 2024/07/05 12:27:44 charge power: 4140W

2024/07/05 12:27:14,stdout,"[ocpp  ] TRACE 2024/07/05 12:27:14 sent g001: [3,\"718647047\",{\"currentTime\":\"2024-07-05T10:27:14Z\"}]

2024/07/05 12:27:14,stdout,"[ocpp  ] TRACE 2024/07/05 12:27:14 recv g001: [2,\"718647047\",\"Heartbeat\",{}]

2024/07/05 12:27:14,stdout,[lp-1  ] DEBUG 2024/07/05 12:27:14 charge power: 4140W

2024/07/05 12:26:56,stdout,[lp-1  ] DEBUG 2024/07/05 12:26:56 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)

**2024/07/05 12:26:56,stdout,[lp-1  ] DEBUG 2024/07/05 12:26:56 max charge current: 6A

2024/07/05 12:26:56,stdout,"[ocpp  ] TRACE 2024/07/05 12:26:56 recv g001: [3,\"2126241791\",{\"status\": \"Accepted\"}]**

**2024/07/05 12:26:54,stdout,"[ocpp  ] TRACE 2024/07/05 12:26:54 sent g001: [2,\"2126241791\",\"SetChargingProfile\",{\"connectorId\":1,\"csChargingProfiles\":{\"chargingProfileId\":1,\"transactionId\":3,\"stackLevel\":0,\"chargingProfilePurpose\":\"TxProfile\",\"chargingProfileKind\":\"Relative\",\"chargingSchedule\":{\"chargingRateUnit\":\"A\",\"chargingSchedulePeriod\":[{\"startPeriod\":0,\"limit\":6}]}}}]**

**2024/07/05 12:26:54,stdout,"[lp-1  ] DEBUG 2024/07/05 12:26:54 pv charge current: min 6A > 0A (4489W @ 3p, battery: false)**

2024/07/05 12:26:54,stdout,[lp-1  ] DEBUG 2024/07/05 12:26:54 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)

2024/07/05 12:26:54,stdout,[lp-1  ] DEBUG 2024/07/05 12:26:54 vehicle range: 202km

2024/07/05 12:26:54,stdout,[lp-1  ] DEBUG 2024/07/05 12:26:54 vehicle soc: 58%

2024/07/05 12:26:54,stdout,[lp-1  ] DEBUG 2024/07/05 12:26:54 vehicle status: C

2024/07/05 12:26:54,stdout,[lp-1  ] DEBUG 2024/07/05 12:26:54 charger status: C

2024/07/05 12:26:54,stdout,[lp-1  ] DEBUG 2024/07/05 12:26:54 !! session: chargeRater.chargedEnergy=5.4 - chargedAtStartup=0.0

2024/07/05 12:26:54,stdout,[lp-1  ] DEBUG 2024/07/05 12:26:54 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)

2024/07/05 12:26:44,stdout,[lp-1  ] DEBUG 2024/07/05 12:26:44 charge power: 4199W

2024/07/05 12:26:14,stdout,"[ocpp  ] TRACE 2024/07/05 12:26:14 sent g001: [3,\"1047295258\",{\"currentTime\":\"2024-07-05T10:26:14Z\"}]

2024/07/05 12:26:14,stdout,"[ocpp  ] TRACE 2024/07/05 12:26:14 recv g001: [2,\"1047295258\",\"Heartbeat\",{}]

2024/07/05 12:26:14,stdout,[lp-1  ] DEBUG 2024/07/05 12:26:14 charge power: 4199W

2024/07/05 12:25:52,stdout,[lp-1  ] DEBUG 2024/07/05 12:25:52 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)

[... 16 lines cut ...]
2024/07/05 12:25:51,stdout,[lp-1  ] DEBUG 2024/07/05 12:25:51 charger status: C

2024/07/05 12:25:51,stdout,[lp-1  ] DEBUG 2024/07/05 12:25:51 !! session: chargeRater.chargedEnergy=5.3 - chargedAtStartup=0.0

2024/07/05 12:25:51,stdout,[lp-1  ] DEBUG 2024/07/05 12:25:51 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)

2024/07/05 12:25:44,stdout,[lp-1  ] DEBUG 2024/07/05 12:25:44 charge power: 4140W

2024/07/05 12:25:14,stdout,"[ocpp  ] TRACE 2024/07/05 12:25:14 sent g001: [3,\"242069770\",{\"currentTime\":\"2024-07-05T10:25:14Z\"}]

2024/07/05 12:25:14,stdout,"[ocpp  ] TRACE 2024/07/05 12:25:14 recv g001: [2,\"242069770\",\"Heartbeat\",{}]

2024/07/05 12:25:14,stdout,[lp-1  ] DEBUG 2024/07/05 12:25:14 charge power: 4140W

2024/07/05 12:24:52,stdout,"[ocpp  ] TRACE 2024/07/05 12:24:52 sent g001: [3,\"837838074\",{}]

[... cut ...] ",\"value\": \"183692.4\"} ]} ],\"connectorId\": 1,\"transactionId\": 3}]

2024/07/05 12:24:50,stdout,"[lp-1  ] DEBUG 2024/07/05 12:24:50 pv charge current: min 6A > 5.82A (124W @ 3p, battery: false)

2024/07/05 12:24:50,stdout,[lp-1  ] DEBUG 2024/07/05 12:24:50 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)

2024/07/05 12:24:50,stdout,[lp-1  ] DEBUG 2024/07/05 12:24:50 vehicle range: 197km

2024/07/05 12:24:50,stdout,[lp-1  ] DEBUG 2024/07/05 12:24:50 vehicle soc: 58%

andig, 2024-07-06:
> Dann feherhafte Umstellung der Pular von 6A auf 16A. Keine Fehler im Log, Anzeige in EVCC weiterhin 4,3KW (6A). Auffällig ist, dass kurz vorher (12:26 Uhr) die Stromstärke via EVCC 2 Mal auf 6A gesetzt wurde, jedes Mal mit "Accept" quittiert. Beim zweiten Mal wurde von der Pulsar aber auf 16A gestellt. 

Wo siehst Du das? Er lädt laut Log mit 4kW, das sind 6A bei 3p. Wenn das nicht so ist bitte Pulsar kontaktieren- evcc tut ja offensichtlich das Richtige.

mrhund, 2024-07-06:
In deren App. Ich hab natürlich auch ein Ticket bei Wallbox/Pulsar aufgemacht ;)
Hätte ja sein können, dass eine Kombination der Befehle (der selbe Befehl kurz hintereinander, verursacht durch einen EVCC Fehler) das Problem ist.

andig, 2024-07-06:
Maybe, ist aber zulässig. Dann muss Pulsar das Problem erklären...

mrhund, 2024-07-14:
@Nightbreezer Ich hab als Workaround ein Programm geschrieben und in einen Container gepackt, das jede Minute prüft ob mit "PV" oder "Min+PV" geladen wird. Wenn ja, wird über die Wallbox.com API geschaut, ob 16A eingestellt sind, wenn ja wird auf die Amps zurückgestellt, die in EVCC eingestellt sind. Nicht 100% geil, aber macht den Job.

Bei Interesse am Container, gerne melden.

andig, 2024-07-14:
Wir können den Strom auch in evcc nachlesen und das tun- auf welches Reading müsste man da schauen?

andig, 2024-07-14:
Bisher hab ich das nicht verfolgt weil https://github.com/evcc-io/evcc/issues/14040#issuecomment-2211703274 eigentlich sagt, dass ich im Api den wahren Wert gar nicht bekomme. Da frag ich mich, wieso du das in HA nachlesen kannst, wir sehen das aber nicht? Anderes Reading?

mrhund, 2024-07-14:
Leider nicht lokal über die Box, ich muss den Umweg über die Cloud API des Herstellers (Wallbox.com) gehen. Diese API ist hier gut beschrieben: https://github.com/SKB-CGN/wallbox.
Nachdem OPCC den Wert "maxChargingCurrent" setzt, kann man diesen über die API wieder bekommen.

Nightbreezer, 2024-07-15:
> @Nightbreezer Ich hab als Workaround ein Programm geschrieben und in einen Container gepackt, das jede Minute prüft ob mit "PV" oder "Min+PV" geladen wird. Wenn ja, wird über die Wallbox.com API [...]

Hi @mrhund 
Lieben Dank! Toll, dass du eine Lösung gebastelt hast. Habe Interesse. Kannst du mir den Container zukommen lassen?
Viele Grüße

premultiply, 2024-07-30:
@andig Fall für das CurrentGetter-API? 🙂

andig, 2024-07-30:
Keine Ahnung. Was ist denn der Fix?

mrhund, 2024-07-31:
> Leider nicht lokal über die Box, ich muss den Umweg über die Cloud API des Herstellers (Wallbox.com) gehen. Diese API ist hier gut beschrieben: https://github.com/SKB-CGN/wallbox. Nachdem OCPP den Wert "maxChargingCurrent" setzt und die Box den Wert sofort an die Cloud übermittels, kann man diesen über die API "near realtime" wieder bekommen.

@andig Das ist der Fix.

andig, 2024-07-31:
Ich verstehe nur Bahnhof. Schritt 1 wäre, bitte mal die 0.129 zu probieren. Logfile. Danach https://github.com/evcc-io/evcc/pull/15028. Logfile. Dann können wir weiter schauen. Ein Mischmasch aus Hersteller-Api und Ocpp können/wollen wir nicht unterstützen. Eine gute Ocpp Lösung wollen wir aber!

andig, 2024-07-31:
@mrhund https://github.com/evcc-io/evcc/pull/14888 ist in 129 drin 👍🏻

mrhund, 2024-08-02:
>Ein Mischmasch aus Hersteller-Api und Ocpp können/wollen wir nicht unterstützen. Eine gute Ocpp Lösung wollen wir aber!

Das verstehe ich, ihr könnt nicht unzähliche Device spezifische Implementierungen machen. 
129 hat das Problem mal nicht gelöst. Ich bin mit dem Workaround eigentlich zufrieden und wenn andere das Problem haben, können sie ihn ja nutzen.

andig, 2024-08-02:
> Schritt 1 wäre, bitte mal die 0.129 zu probieren. Logfile. Danach https://github.com/evcc-io/evcc/pull/15028. Logfile.

Gibts ein Logfile für ocpp?

Nightbreezer, 2024-08-12:
> @andig @Nightbreezer Rückmeldung von Wallbox.com Support:
> 
> > Hello *,
> > Thank you for your reply.
> > If the error occurs only when the OCPP is enabled, please contact your OCPP provider about this as they take charge of the charger functions when its connected.
> > Best regards,
> > Hind
> > Wallbox Service Support
> 
> Hier ist also keine Hilfe zu erwarten. Ich habe meinen Fix gepusht: https://github.com/mrhund/Wallbox_Pulsar-OCPP-Charge_Current-Fix. Meine Programmierkenntnisse reichen nicht aus, um einen Pull Request für EVCC zu erstellen.

@mrhund Lieben Dank! Gibt es vielleicht ein Mini Tutorial, wie ich dies installiere? Ich bin leider noch totaler Anfänger.
Habe evcc so auf einem Raspberry installiert: https://docs.evcc.io/docs/installation/linux

@andig Ich hatte leider OCPP Trace nicht im Logfile aktiviert. Habe ich nun gemacht und werde es posten sobald der Fehler wieder auftritt.

andig, 2024-08-12:
Schön wäre auch der Vergleich dazu wenn es bei dir klappt.

Nightbreezer, 2024-08-15:
@andig Hier nun anbei das Log mit EVCC Version 0.129.0
Start von Min+PV um 10:57 Uhr
Normales Ladeverhalten dann bis ca. 11:07 Uhr, ab da dann laden mit den vollen 11kwH
Stopp meinerseits bei 11:13 Uhr

224855 ist der Name meiner Wallbox Pulsar

**Aug 15 10:57:00 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:00 set charge mode: minpv**
Aug 15 10:57:00 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:00 pv timer reset
Aug 15 10:57:00 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:00 pv timer inactive
Aug 15 10:57:00 myroof evcc[507665]: [site  ] DEBUG 2024/08/15 10:57:00 ----
Aug 15 10:57:00 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:00 charge power: 0W
Aug 15 10:57:00 myroof evcc[507665]: [site  ] DEBUG 2024/08/15 10:57:00 pv power: 6406W
Aug 15 10:57:00 myroof evcc[507665]: [site  ] DEBUG 2024/08/15 10:57:00 battery soc: 0%
Aug 15 10:57:00 myroof evcc[507665]: [site  ] DEBUG 2024/08/15 10:57:00 battery power: 0W
Aug 15 10:57:00 myroof evcc[507665]: [site  ] DEBUG 2024/08/15 10:57:00 grid meter: -5195W
Aug 15 10:57:00 myroof evcc[507665]: [site  ] DEBUG 2024/08/15 10:57:00 site power: -5095W
Aug 15 10:57:00 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:00 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
Aug 15 10:57:00 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:00 charger status: B
Aug 15 10:57:00 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:00 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
Aug 15 10:57:00 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:00 pv charge current: 7.38A = 0A + 7.38A (-5095W @ 3p)
Aug 15 10:57:00 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:00 max charge current: 7.38A
Aug 15 10:57:00 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:00 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
Aug 15 10:57:00 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 10:57:00 send 224855: [2,"867114990","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"sta>
Aug 15 10:57:00 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 10:57:00 recv 224855: [4,"867114990","ProtocolError","validation of RemoteStartTransaction failed",{}]
Aug 15 10:57:00 myroof evcc[507665]: [lp-1  ] ERROR 2024/08/15 10:57:00 charger enable: ocpp message (867114990): ProtocolError - validation of RemoteStartTransaction failed
Aug 15 10:57:00 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 10:57:00 recv 224855: [2,"108424928","Heartbeat",{}]
Aug 15 10:57:00 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 10:57:00 send 224855: [3,"108424928",{"currentTime":"2024-08-15T08:57:00Z"}]
Aug 15 10:57:15 myroof evcc[507665]: [site  ] DEBUG 2024/08/15 10:57:15 ----
Aug 15 10:57:15 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:15 charge power: 0W
Aug 15 10:57:15 myroof evcc[507665]: [site  ] DEBUG 2024/08/15 10:57:15 pv power: 6532W
Aug 15 10:57:15 myroof evcc[507665]: [site  ] DEBUG 2024/08/15 10:57:15 battery soc: 0%
Aug 15 10:57:15 myroof evcc[507665]: [site  ] DEBUG 2024/08/15 10:57:15 battery power: 0W
Aug 15 10:57:15 myroof evcc[507665]: [site  ] DEBUG 2024/08/15 10:57:15 grid meter: -5850W
Aug 15 10:57:15 myroof evcc[507665]: [site  ] DEBUG 2024/08/15 10:57:15 site power: -5750W
Aug 15 10:57:15 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:15 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
Aug 15 10:57:15 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:15 charger status: B
Aug 15 10:57:15 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:15 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
Aug 15 10:57:15 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:15 pv charge current: 8.33A = 0A + 8.33A (-5750W @ 3p)
Aug 15 10:57:15 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:15 max charge current: 8A
Aug 15 10:57:15 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:15 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
Aug 15 10:57:15 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 10:57:15 send 224855: [2,"772923199","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"sta>
Aug 15 10:57:16 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 10:57:16 recv 224855: [3,"772923199",{"status": "Accepted"}]
Aug 15 10:57:16 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:16 charger enable
Aug 15 10:57:16 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:16 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
Aug 15 10:57:16 myroof evcc[507665]: [lp-1  ] DEBUG 2024/08/15 10:57:16 wake-up timer: start
Aug 15 10:57:17 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 10:57:17 recv 224855: [2,"818138092","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"err>
Aug 15 10:57:17 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 10:57:17 send 224855: [3,"818138092",{}]
Aug 15 10:57:17 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 10:57:17 recv 224855: [2,"1995772889","StartTransaction",{"connectorId": 1,"meterStart": 1120666,"idTag": "evcc","timestamp": "2024-08-15T0>
Aug 15 10:57:17 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 10:57:17 send 224855: [3,"1995772889",{"idTagInfo":{"status":"Accepted"},"transactionId":1}]
[... cut ...]
Aug 15 11:13:01 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 11:13:01 recv 224855: [2,"1089727290","StopTransaction",{"meterStop": 1122621,"timestamp": "2024-08-15T09:13:01Z","idTag": "","transactionD>
Aug 15 11:13:01 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 11:13:01 send 224855: [3,"1089727290",{"idTagInfo":{"status":"Accepted"}}]
Aug 15 11:13:03 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 11:13:03 recv 224855: [2,"715785083","Heartbeat",{}]
Aug 15 11:13:03 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 11:13:03 send 224855: [3,"715785083",{"currentTime":"2024-08-15T09:13:03Z"}]
Aug 15 11:13:15 myroof evcc[507665]: [site  ] DEBUG 2024/08/15 11:13:15

andig, 2024-08-15:
Dein Logfile zeigt überhaupt nicht, dass da irgendwas gesteuert würde, und zwar ab Beginn. Dafür ist da aber ein Fehler:

```
Aug 15 10:57:00 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 10:57:00 send 224855: [2,"867114990","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"sta>
Aug 15 10:57:00 myroof evcc[507665]: [ocpp  ] TRACE 2024/08/15 10:57:00 recv 224855: [4,"867114990","ProtocolError","validation of RemoteStartTransaction failed",{}]
```

Ich kann mich nur wiederholen:

- aktuelle Software testen (am Besten Nightly)
- vollständiges Logfile

Da das Thema hier jetzt schon 3 Monate alt ist mache ich mal zu- gerne neuer Versuch auf dieser Basis