evcc-io/evcc #14042: Anzahl der gespeicherten Ladevorgänge geringer als über messaging gemeldete Ladevorgänge

hrpv, 2024-05-23:
### Describe the bug

Die Anzahl der  im csv gepeicherten Ladevorgänge ist geringer als die über das Messaging in Telegram gemeldeten Ladevorgänge.
Ausserdem scheinen die Variablen chargedEnergy  und chargeDuration manchmal zwischen den Ladevorgängen nicht zurückgesetzt zu werden sondern werden immer  weiter inkrementiert. Die **Anzahl** der Ladevorgänge scheint im Messenger richtig zu sein, aber die chargedEnergy  und chargeDuration stimmt nur nach Neustart von evcc wieder

`yaml
msg: Finished "${title}" charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
`

Beispiel aus Telegram, viel zu hoher Wert für chargeDuration und  chargedEnergy
Wallbox-Ule, [23.05.2024 **12:52**]
Started "Wärmepumpe" charging in "pv" mode

Wallbox-Ule, [23.05.2024 **13:10**]
Finished "Wärmepumpe" charging **8.3kWh** in **18h21m31s**.

### Steps to reproduce

Inhalt csv Ladevorgänge vergleichen mit Anzahl der Ladevorgänge über Messenger Telegram
[session-2024-05 (2).csv](https://github.com/evcc-io/evcc/files/15421606/session-2024-05.2.csv)
[telegram.txt](https://github.com/evcc-io/evcc/files/15421615/telegram.txt)

### Configuration details

```yaml
#restrict to localhost (raspberry)
#uri: 0.0.0.0:7070 # uri for ui
#uri: 127.0.0.1:7070 # uri for ui
#uri: 192.168.168.48:7070 # uri for ui
network:
  schema: http
  host: 192.168.168.48
  port: 7070
  
interval: 10s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
# sponsortoken:

# log settings
# set log: error fuer serie

# log: debug
# levels:
  # core: debug
  # lp-1: debug
  # lp-2: debug

log: error
levels:
  core: debug
  lp-1: info
  lp-2: info

#hems Im Moment nur SMA Home Manager
#deviceid und vendorid: fest gepinnt, damit aenderungen von /etc/machine-id keine wirkung haben
# https://github.com/evcc-io/evcc/discussions/2394
# es wird nur die niedrigste device id eingetragen, die ladepunkte werden fortlaufend nummeriert
hems:
  type: sma
  allowcontrol: false # set true to allow SHM controlling charger in PV modes
  vendorid: 28081973 # Hartkodierte konstante
  deviceid: xxx # abgleitet aus /etc/machine-id 

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
meters:

- type: template
[... 305 lines cut ...]
### Log details

```text
no errors seen
[site  ] DEBUG 2024/05/23 21:22:04 ----
[lp-1  ] DEBUG 2024/05/23 21:22:04 charge power: 5W
[lp-2  ] DEBUG 2024/05/23 21:22:04 charge power: 1009W
[site  ] DEBUG 2024/05/23 21:22:04 pv power: 0W
[site  ] DEBUG 2024/05/23 21:22:04 battery soc: 72%
[site  ] DEBUG 2024/05/23 21:22:04 battery power: 694W
[site  ] DEBUG 2024/05/23 21:22:04 grid meter: -14W
[site  ] DEBUG 2024/05/23 21:22:04 grid powers: [300 232 -546]W
[site  ] DEBUG 2024/05/23 21:22:04 grid currents: [1.64 1.38 -2.41]A
[site  ] DEBUG 2024/05/23 21:22:04 site power: 680W
[lp-1  ] DEBUG 2024/05/23 21:22:04 charge total import: 737.120kWh
[lp-1  ] DEBUG 2024/05/23 21:22:04 charger status: A
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.126.2 (940ce81a)

VolkerK62, 2024-05-24:
Im Ladelog werden Sessions gespeichert, keine einzelnen Ladevorgänge.
Eine Session beginnt mit dem anstecken und endet mit dem abstecken.
Bei schaltbaren Steckdosen endet die Session entweder durch einen Neustart oder durch schalten auf "aus".

closed 2024-05-24 by andig

hrpv, 2024-05-24:
Ich teste mal die schaltbare Steckdose kurz vor Mitternaccht per con job auszuschalten und so einen Eintrag im Ladelog zu erzeugen:  **mosquitto_pub -t evcc/loadpoints/2/mode/set -m "off"**