evcc-io/evcc #14045: Configuration password

MenosChris, 2024-05-24:
### Describe the bug

evcc runs as an add on in home assistant.  
I do not know a password for configuration (right top corner menu).
I never set one up,  not changed,  or find an opportunity to setup, or reset a password.
I tried one or the other password,...all incorrect für sure. 

I need help with this, please. 

### Steps to reproduce

1.
2.
3.
...
1) go to top right Menü
2) click on configuration
3) password is asked
4) I do not know this password,  since I have never set one.....

### Configuration details

```yaml
Not needed, not changed anything
```

### Log details

```text
No log
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.126.4

andig, 2024-05-24:
> I never set one up, not changed, or find an opportunity to setup, or reset a password.

Then you should be asked to create one. Once its set, it is stored in the database. Workaround: remove it from the database using 

    evcc settings

For HA (which we don't support as the issue template makes clear!) that would probably imply copying the DB file somewhere else before doing so.

amtssp, 2024-05-24:
Sorry to hijack this - but when trying the online demo version - it also needs a for me unknown password. 
Which one could we use?
(I tried demo and admin) but without success.

naltatis, 2024-05-24:
The password for the online demo is not public. Maybe we'll introduce a dedicated read-only demo mode for the settings area, but currently we haven't.

BionicThomas, 2024-05-27:
tried both commands and it does not recognize a password being set on command line nor does a reset trigger the dialogue to set a password via GUI. I'm on 0.126.4 latest nightly

BionicThomas, 2024-05-27:
> @BionicThomas you'll need to make sure that you're using the same database as the service when it's running!

I one have one instance of evcc, this is the command line feedback.

```
pi@pi400:~ $ sudo evcc password set
[main  ] INFO 2024/05/27 16:09:05 evcc 0.126.4 (5a4220b8)
[main  ] INFO 2024/05/27 16:09:05 using config file: /etc/evcc.yaml
[db    ] INFO 2024/05/27 16:09:05 using sqlite database: /root/.evcc/evcc.db
? Password [? for help] ********
```

Strange that I have 3 times the evcc.db
Should the command not use the correct one? How can I see which one I'm actually using and can I force the password command to use it?

```
pi@pi400:/ $ sudo find -name evcc.db
./home/pi/.evcc/evcc.db
./var/lib/evcc/evcc.db
./root/.evcc/evcc.db
```

andig, 2024-05-27:
You‘ll need to use the same one as the evcc service.

BionicThomas, 2024-05-27:
> You‘ll need to use the same one as the evcc service.

I'm sorry I', confused. Do I have accidentally two evcc instances that are overlapping each other? 
why is the command `sudo evcc password set` using one db and the evcc service another one?
Any advice on how to resolve this?

Sillium007, 2024-05-29:
You can set the database to use e.g. by using an environment variable before calling evcc:
```
export EVCC_DATABASE_DSN=/var/lib/evcc/evcc.db
sudo -E evcc password set
```

closed 2024-05-29 by MenosChris

JHPWBN, 2024-05-29:
Not sure if this the right place, but regularly I am being asked for a password (even when I am not trying to change anything), with the message „not connected to server“. When I enter the password, I am told it is incorrect (I am 100% sure it is the right one).
When I leave the evcc browser tab and then come back it is gone, I am no longer asked for a PW.