evcc-io/evcc #14046: VAG login broken? (e-Golf) 0.126.4

Joeknx, 2024-05-24:
### Describe the bug

Nach dem Update von 0.126.3 zu 0.126.4 funktioniert die Verbindung du VAG (e-Golf nicht mehr).

Ein Neustart hilf auch nicht. 

Fehlermeldung nach dem Neustart:
[main ] ERROR 2024/05/23 21:06:09 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': missing state

### Steps to reproduce

Neustart hilf nicht.
VW-App funktioniert. 

### Configuration details

```yaml
...
```

### Log details

```text
[main ] ERROR 2024/05/23 21:06:09 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': missing state
```

### What type of operating system are you running?

Linux

andig, 2024-05-24:
Terms & Conditions? Ansonsten fehlt hier ein trace log.

closed 2024-05-24 by andig

TobiasHuber1980, 2024-05-24:
1) mach mal downgrade auf 0.126.3 (dann könnte es mit https://github.com/evcc-io/evcc/commit/73ece1ba6b66c4c808c45ec90e71f3c83bc303d9 zusammen hängen). trace-log wird wohl benötigt, sonst wird man nicht rausfinden, wo der fehler liegt.

2) teste mal das "id" template bei 0.126.4

Joeknx, 2024-05-24:
"id" template funktioniert!

Vielen Dank!

Soll ich trotzdem mal wieder umstellen und einen  trace-log machen?

TobiasHuber1980, 2024-05-24:
ja schadet auf jeden fall nicht ein tracelog mit "vw" zu machen, dann kann @andig da mal bei gucken.

Joeknx, 2024-05-24:
jetzt habe ich zurück auf "vw" gewechselt und es geht immer noch. Keine Ahnung was das dann gestern war. Nach dem Update auf 0.126.4 ging es nicht mehr und mehrmaliger Neustart hat auch nichts gebracht. 
Nächstes Mal mache ich einen trace-log.