evcc-io/evcc #14049: Planner: Conflict with load management and suggestion for Solcast and consumption

djfanatix, 2024-05-24:
**Is your feature request related to a problem? Please describe.**
When the Load management is activated, the forecast of the planner is not correct anymore.
Example:
- Max charge setting: 8kw
- Max grid import setting: 5kw

--> EVCC will calculate with 8kw, which is possible when there is 3kw Solar + 5kw Grid, but not during the night.

**Describe the solution you'd like**
1. Quick fix: when load management is activated, calculate planning with max grid import setting (5kw example), if we have solar, this is a nice extra.
2. Future: Implement Solcast to make a more accurate planning forecast
3. Very far Future: Add, for example, the possiblity to add own house consumption. (experimental)
  a. complicated: make a model to calculate house consumption based on past data
  b. easier: add the possibility to upload, fetch, mqtt,... a json with a load profile of your house for the next 24 hours. (can be achieved with home assistant or alternative)

**Describe alternatives you've considered**
Anyway, many thanks for all the work done for EVCC!

andig, 2024-05-24:
Load management does currently not work hand in hand with planner and priorities. Will be documented in https://github.com/evcc-io/docs/issues/561

andig, 2024-05-24:
> Add, for example, the possiblity to add own house consumption. 

Interesting and certainly doable. What would be the actual use case?

andig, 2024-05-24:
That would only be relevant together with load management then. Without LM, the planner will always charge at full speed.

andig, 2024-06-16:
Seems theres nothing left to do here that's not already in the backlog, hence closing. Thank you!

closed 2024-06-16 by andig

djfanatix, 2024-06-20:
But is something changed?

Could below be implemented?

_Quick fix: when load management is activated, calculate planning with max grid import setting (5kw example), if we have solar, this is a nice extra_

andig, 2024-06-20:
Yes, see https://github.com/evcc-io/evcc/issues/14261