evcc-io/evcc #14056: Wallbox Pulsar Plus rejects commands after Firmware Update

ChrisDaHub, 2024-05-25:
### Describe the bug

I updated my Wallbox and now it rejects commands.

### Steps to reproduce

1. Update the Wallbox
2. Try to connect evcc to your wallbox
3. Jave a look in the logfile
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 10s # control cycle interval

meters:
- type: template
  template: cg-emt3xx 
  id: 1  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: grid  
  modbus: rs485serial  
  name: grid1
- type: template
  template: solaredge-inverter 
  id: 18  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: pv  
  modbus: rs485serial  
  name: pv2
- type: template
  template: cg-emt3xx 
  id: 2  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: charge  
  modbus: rs485serial  
  name: charge7

chargers:
- type: template
  template: pulsarplus 
  name: wallbox6

vehicles:
- type: template
  template: tesla
  title: Model 3  
  accessToken: xxx
  refreshToken: xxx
  vin: LRW3E7EL0NCxxxxxx  
[... 979 lines cut ...]
[lp-1  ] DEBUG 2024/05/25 02:52:33 charger status: A
[lp-1  ] ERROR 2024/05/25 02:52:33 charger disable: Rejected
[lp-1  ] DEBUG 2024/05/25 02:52:41 charge power: 3W
[lp-1  ] DEBUG 2024/05/25 02:52:41 charge currents: [0.016 0 0]A
[lp-1  ] DEBUG 2024/05/25 02:52:42 charge voltages: [242 242 243]V
[lp-1  ] DEBUG 2024/05/25 02:52:42 detected connected phases: 3p
[lp-1  ] DEBUG 2024/05/25 02:52:42 charge total import: 3131.000kWh
[lp-1  ] DEBUG 2024/05/25 02:52:42 charger status: A
[lp-1  ] ERROR 2024/05/25 02:52:43 charger disable: Rejected
[lp-1  ] DEBUG 2024/05/25 02:52:51 charge power: 3W
[lp-1  ] DEBUG 2024/05/25 02:52:51 charge currents: [0.016 0 0]A
[lp-1  ] DEBUG 2024/05/25 02:52:52 charge voltages: [242 242 243]V
[lp-1  ] DEBUG 2024/05/25 02:52:52 detected connected phases: 3p
[lp-1  ] DEBUG 2024/05/25 02:52:52 charge total import: 3131.000kWh
[lp-1  ] DEBUG 2024/05/25 02:52:52 charger status: A
[lp-1  ] ERROR 2024/05/25 02:52:53 charger disable: Rejected
```

### What type of operating system are you running?

Linux

### Version

0.126.2

ChrisDaHub, 2024-05-25:
Wallbox Firmware v.6.4.14

andig, 2024-05-25:
Needs trace log.

closed 2024-05-25 by andig

ChrisDaHub, 2024-05-30:
Angeblich wurde gar nichts geändert.  Ich kann in der App keine Experimentellen Funktionen mehr finden.  Hier ein Trace Log. Ich versuche die Ladung auf "schnell"  zu starten. Angeblich bereit, aber es tut sich nichts.  Dann schalte ich die Ladung aus und es erscheint sofort der Fehler "rejected"

Wird mehr benötigt? Bitte einfach bescheid geben! Danke

[ocpp ] TRACE 2024/05/30 23:02:22 received JSON message from 307528: [2,"125536412","Heartbeat",{}]
[ocpp ] TRACE 2024/05/30 23:02:22 sent JSON message to 307528: [3,"125536412",{"currentTime":"2024-05-30T21:02:22Z"}]
[ocpp ] TRACE 2024/05/30 23:03:22 received JSON message from 307528: [2,"397179109","Heartbeat",{}]
[ocpp ] TRACE 2024/05/30 23:03:22 sent JSON message to 307528: [3,"397179109",{"currentTime":"2024-05-30T21:03:22Z"}]
[ocpp ] TRACE 2024/05/30 23:04:22 received JSON message from 307528: [2,"48367785","Heartbeat",{}]
[ocpp ] TRACE 2024/05/30 23:04:22 sent JSON message to 307528: [3,"48367785",{"currentTime":"2024-05-30T21:04:22Z"}]
[ocpp ] TRACE 2024/05/30 23:04:23 sent JSON message to 307528: [2,"965967664","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:04:23 received JSON message from 307528: [3,"965967664",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:04:26 sent JSON message to 307528: [2,"2081105694","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:04:26 received JSON message from 307528: [3,"2081105694",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:04:36 sent JSON message to 307528: [2,"1988829315","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:04:36 received JSON message from 307528: [3,"1988829315",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:04:46 sent JSON message to 307528: [2,"4208034594","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:04:46 received JSON message from 307528: [3,"4208034594",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:04:56 sent JSON message to 307528: [2,"398231206","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:04:56 received JSON message from 307528: [3,"398231206",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:05:06 sent JSON message to 307528: [2,"3859055028","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:05:06 received JSON message from 307528: [3,"3859055028",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:05:16 sent JSON message to 307528: [2,"229049334","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:05:16 received JSON message from 307528: [3,"229049334",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:05:23 received JSON message from 307528: [2,"125781515","Heartbeat",{}]
[ocpp ] TRACE 2024/05/30 23:05:23 sent JSON message to 307528: [3,"125781515",{"currentTime":"2024-05-30T21:05:23Z"}]
[ocpp ] TRACE 2024/05/30 23:05:26 sent JSON message to 307528: [2,"1341724828","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:05:26 received JSON message from 307528: [3,"1341724828",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:05:36 sent JSON message to 307528: [2,"1271458864","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:05:36 received JSON message from 307528: [3,"1271458864",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:05:46 sent JSON message to 307528: [2,"1392889409","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:05:46 received JSON message from 307528: [3,"1392889409",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:05:56 sent JSON message to 307528: [2,"1477933898","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:05:56 received JSON message from 307528: [3,"1477933898",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:06:06 sent JSON message to 307528: [2,"1824753916","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:06:06 received JSON message from 307528: [3,"1824753916",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:06:16 sent JSON message to 307528: [2,"1090027171","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:06:16 received JSON message from 307528: [3,"1090027171",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:06:23 received JSON message from 307528: [2,"1892453822","Heartbeat",{}]
[ocpp ] TRACE 2024/05/30 23:06:23 sent JSON message to 307528: [3,"1892453822",{"currentTime":"2024-05-30T21:06:23Z"}]
[ocpp ] TRACE 2024/05/30 23:06:26 sent JSON message to 307528: [2,"4166396262","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:06:26 received JSON message from 307528: [3,"4166396262",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:06:36 sent JSON message to 307528: [2,"2721759782","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:06:36 received JSON message from 307528: [3,"2721759782",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:06:46 sent JSON message to 307528: [2,"3609864036","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:06:46 received JSON message from 307528: [3,"3609864036",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:06:56 sent JSON message to 307528: [2,"182345783","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:06:56 received JSON message from 307528: [3,"182345783",{"status": "Rejected"}]
[... cut ...]
[ocpp ] TRACE 2024/05/30 23:08:23 sent JSON message to 307528: [3,"77549659",{"currentTime":"2024-05-30T21:08:23Z"}]
[ocpp ] TRACE 2024/05/30 23:08:26 sent JSON message to 307528: [2,"3398448532","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:08:26 received JSON message from 307528: [3,"3398448532",{"status": "Rejected"}]
[ocpp ] TRACE 2024/05/30 23:08:36 sent JSON message to 307528: [2,"1573325745","RemoteStopTransaction",{"transactionId":11}]
[ocpp ] TRACE 2024/05/30 23:08:36 received JSON message from 307528: [3,"1573325745",{"status": "Rejected"}]

ChrisDaHub, 2024-05-30:
Angeblich wurde gar nichts geändert. Ich kann in der App keine Experimentellen Funktionen mehr finden.  Ich versuche die Ladung auf "schnell" zu starten. Angeblich bereit, aber es tut sich nichts. Dann schalte ich die Ladung aus und es erscheint sofort der Fehler "rejected"

Ich habe jetzt noch ein Tracelog angehängt. 

Wird mehr benötigt? Bitte einfach bescheid geben! Danke und viele Grüße

andig, 2024-05-31:
Pulsar will need to tell you why they would reject these messages.

forum6691, 2024-06-14:
I 've a problem which is maybe not very far this discussion, so I write in this post.
I've also a pulsar plus wallbox in monophased system.
Firmware is 6.4.14 (the last normally)
The max current and power is set by hardware inside in the wallbox to 16A , 3,7 Kw.

Evcc version is [0.126.6].

In evcc, I set Max current to 12 A (2,8 kw) and min current to 6A ( 1,4 kw).
[image]

The max of my PV is 2,9 kw, this is why I set these values.

When the solar power appears in the morning, EVVC pilots the wallbox, at the beginning power is 1,4 kw, then the power raises to 2,8 kw, but after a short  moment, the power order by evcc raise to 3,7 kw and I don't want this because, the difference power between 3,7 and 2,9 kw comes from the grid .

Parts of my configuration evcc.yaml
chargers:
- type: template
  template: pulsarplus
  timeout: 45s
  name: wallbox3

loadpoints:
- title: Garage
  charger: wallbox3
  vehicle: ev2
  mode: pv

site:
  title: Maison
  meters:
    grid: grid1
    pv: pv1

Have you an idea to force evcc to doesn't raise 12A (2,8 kw) during the charge ? Tks by advance.