evcc-io/evcc #14057: GUI Shows "Laadplan start at ...."

Lightmanster, 2024-05-25:
### Describe the bug

When you activate( Min+ ) PV mode with a charge plan in future the status show " Laadplan begint op maandag 04:22"
You dont see status update from the current moment (Charging, not enough pv, swithcing phases  etc etc)
Maybe move charge plan message next to plan.

[image]

### Steps to reproduce

1. Make charge plan for tomorrow.
2. activated PV or Min + PV
3.
...

### Configuration details

```yaml
It is gui thing
```

### Log details

```text
It is gui thing
```

### What type of operating system are you running?

Linux

### Version

0.126.4

andig, 2024-05-25:
Please share /api/state

Lightmanster, 2024-05-25:
{
result: {
auth: {
vehicles: { }
},
battery: [ ],
batteryDischargeControl: false,
batteryMode: "unknown",
bufferSoc: 0,
bufferStartSoc: 0,
circuits: {
main: {
power: -77,
current: 5,
maxPower: 15870,
maxCurrent: 23
}
},
currency: "EUR",
fatal: null,
greenShareHome: 1,
greenShareLoadpoints: 0.6886153846153846,
gridConfigured: true,
gridCurrents: [
5,
-3,
-2
],
gridPower: -77,
homePower: 0,
interval: 20,
loadpoints: [
{
chargeCurrent: 7,
chargeCurrents: [
7.012000560760498,
0,
0
],
chargeDuration: 379000000000,
chargePower: 1625,
chargeRemainingDuration: 64225000000000,
chargeRemainingEnergy: 28990.555555555587,
chargeTotalImport: 4663.159,
chargedEnergy: 164.99999999996362,
chargerFeatureHeating: false,
chargerFeatureIntegratedDevice: false,
chargerIcon: null,
chargerPhases1p3p: true,
chargerPhysicalPhases: null,
charging: true,
connected: true,
connectedDuration: 0,
disableThreshold: 200,
effectiveLimitSoc: 80,
effectiveMaxCurrent: 16,
effectiveMinCurrent: 6,
effectivePlanSoc: 80,
effectivePlanTime: "2024-05-27T05:00:00Z",
effectivePriority: 0,
enableThreshold: 0,
enabled: true,
limitEnergy: 0,
limitSoc: 0,
maxCurrent: 16,
minCurrent: 6,
mode: "minpv",
phaseAction: "inactive",
phaseRemaining: 0,
phasesActive: 1,
phasesConfigured: 0,
phasesEnabled: 1,
planEnergy: 0,
planOverrun: 0,
planProjectedStart: "2024-05-27T04:22:27+02:00",
planTime: "0001-01-01T00:00:00Z",
priority: 0,
pvAction: "inactive",
pvRemaining: 0,
sessionCo2PerKWh: null,
sessionEnergy: 164.99999999996362,
sessionPrice: 0.02908748174438244,
sessionPricePerKWh: 0.17628776814781122,
sessionSolarPercentage: 29.765829423182783,
smartCostActive: false,
smartCostLimit: 0.2,
title: "Oprit",
vehicleClimaterActive: false,
vehicleDetectionActive: false,
vehicleLimitSoc: 0,
vehicleName: "Tesla",
vehicleOdometer: 0,
vehicleRange: 248,
vehicleSoc: 48.18109756097557
}
],
prioritySoc: 0,
pv: [
{
power: 1119
}
],
pvEnergy: 0,
pvPower: 1119,
residualPower: 0,
siteTitle: "Thuis",
smartCostType: "priceforecast",
sponsor: "xxxx",
sponsorTokenExpires: 0,
statistics: {
30d: {
avgCo2: 0,
avgPrice: 0.14214661391394842,
chargedKWh: 57.90686463441976,
solarPercentage: 47.511930578926304
},
365d: {
avgCo2: 0,
avgPrice: 0.14214661391394842,
chargedKWh: 57.90686463441976,
solarPercentage: 47.511930578926304
},
total: {
avgCo2: 0,
avgPrice: 0.14214661391394842,
chargedKWh: 57.90686463441976,
solarPercentage: 47.511930578926304
}
},
tariffGrid: 0.251,
tariffPriceHome: 0,
tariffPriceLoadpoints: 0.07815753846153846,
vehicles: {
Tesla: {
title: "Tesla",
capacity: 82,
limitSoc: 80,
features: [
"CoarseCurrent"
],
plans: [
{
soc: 80,
time: "2024-05-27T05:00:00Z"
}
]
}
},
version: "0.126.4"
}
}

naltatis, 2024-06-22:
Moving this info to the bottom does not work on mobile, but I'll think of a good way to combine both information in the status text.

closed 2024-07-07 by andig