evcc-io/evcc #14071: Improve ford-connect

andig, 2024-05-26:
Der erste Ansatz für `ford-connect` ist in wenigen Minuten im Nightly. Was noch fehlt:

- [ ] Persistenz des Refresh Tokens da dieser leider abläuft
- [ ] Beschreibung für `evcc token <ford>` um Code in Token zu verwandeln
- [ ] Nutzung VIN, wie ist unklar

_Originally posted by @andig in https://github.com/evcc-io/evcc/issues/13949#issuecomment-2132235143_

Chris591, 2024-05-26:
an die Vin bin ich auch noch nicht ran gekommen. Nur der Nickname kommt in frage im Augenblick.
Oder man müsste die ID nutzen - da würde es dann auch nach einem umbenennen oder so keine Fehler geben.
sollte man aber abfragen also zum token speichern oder?

Chris591, 2024-05-27:
Fahrzeug konnte ich über die UI einrichten - nur den RFID token kann man nicht einrichten.
mal schauen ob es stabil und über Neustarts läuft.

andig, 2024-05-27:
Was hat ein RFID Token mit Ford zu tun?

Chris591, 2024-05-27:
nichts nur mit dem Auto an sich - zum identifizieren - keine Ahnung wo das in Zukunft hin sollte.

andig, 2024-05-27:
Was ist denn das konkrete Problem?

Chris591, 2024-05-27:
naja die Freischaltung der Wallbox - Aber eigentlich wird diese ja eh nur freigeschaltet wenn das Auto erkannt wurde oder?

Chris591, 2024-05-27:
ach eine Sache  - der Status steht als Status: drinnen. ist aber für mich kein Problem nur ob es so sein sollte ist die Frage.
[image]

Chris591, 2024-05-27:
aber es ist auch nicht angeschlossen!

andig, 2024-05-27:
> naja die Freischaltung der Wallbox - Aber eigentlich wird diese ja eh nur freigeschaltet wenn das Auto erkannt wurde oder?

Ich verstehe nur Bahnhof. Bitte aussagekräfties Issue.

> aber es ist auch nicht angeschlossen!

Bitte 

    evcc vehicle --log trace

Chris591, 2024-05-27:
ich will das keiner die Wallbox ohne RFID nutzen kann - ist öffentlich zugänglich. deswegen hab ich dieses in der evcc.yaml unter vehicles
```
  identifiers:
    - aabbccdd # RFID token ID
```
Dies geht aber über die UI Konfiguration noch nicht oder?
Aber die Wallbox sollte ja eh nicht freischalten wenn ein unbekanntes Auto angeschlossen ist oder hab ich da einen falsch Vorstellung?

andig, 2024-05-27:
Das hat nichts mit Ford zu tun- OT.

Chris591, 2024-05-27:
```
sudo docker exec evcc evcc -l trace vehicle db:1
[main  ] INFO 2024/05/27 17:05:06 evcc 0.126.4 (5a4220b8)
[main  ] INFO 2024/05/27 17:05:06 using config file: /etc/evcc.yaml
[db    ] INFO 2024/05/27 17:05:06 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/05/27 17:05:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT * FROM `settings` 60 <nil>
[mqtt  ] INFO 2024/05/27 17:05:06 connecting evcc-852809414 at tcp://10.x.x.x:1883
[mqtt  ] DEBUG 2024/05/27 17:05:06 tcp://10.x.x.x:1883 connected
[db    ] TRACE 2024/05/27 17:05:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT * FROM `config_details` WHERE `config_details`.`config_id` = 1 8 <nil>
[db    ] TRACE 2024/05/27 17:05:06 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 1 <nil>
[ford  ] TRACE 2024/05/27 17:05:06 POST https://dah2vb2cprod.b2clogin.com/914d88b1-3523-4bf6-9be4-1b96b4f6f919/oauth2/v2.0/token?p=B2C_1A_signup_signin_common
[ford  ] TRACE 2024/05/27 17:05:07 grant_type=refresh_token&refresh_token=***
--
{"access_token":"***","id_token":"***","token_type":"Bearer","not_before":1716822307,"expires_in":1200,"expires_on":1716823507,"resource":"c1e4c1a0-2878-4e6f-a308-836b34474ea9","id_token_expires_in":1200,"profile_info":"***","scope":"https://dah2vb2cprod.onmicrosoft.com/fordconnect/access openid offline_access","refresh_token":"***","refresh_token_expires_in":7776000}
[ford  ] TRACE 2024/05/27 17:05:07 GET https://api.mps.ford.com/api/fordconnect/v3/vehicles
[ford  ] TRACE 2024/05/27 17:05:09 {"status":"SUCCESS","vehicles":[{"vehicleId":"***","make":"F","modelName":"Mustang Mach-E","modelYear":"2021","color":"LUCID RED TC/RED CARPET TC","nickName":"Mustang","modemEnabled":true,"vehicleAuthorizationIndicator":1,"serviceCompatible":true,"engineType":"BEV"}]}
[ford  ] TRACE 2024/05/27 17:05:09 GET https://api.mps.ford.com/api/fordconnect/v3/vehicles/***
[ford  ] TRACE 2024/05/27 17:05:12 {"status":"SUCCESS","vehicle":{"vehicleId":"***","make":"F","modelName":"Mustang Mach-E","modelYear":"2021","color":"LUCID RED TC/RED CARPET TC","nickName":"Mustang","modemEnabled":true,"lastUpdated":"05-26-2024 18:01:47","vehicleAuthorizationIndicator":1,"serviceCompatible":true,"engineType":"BEV","vehicleDetails":{"batteryChargeLevel":{"value":94.0,"distanceToEmpty":409.0,"timestamp":"05-26-2024 17:53:27"},"mileage":24350.3,"odometer":39188.0},"vehicleStatus":{"tirePressureWarning":false,"deepSleepInProgress":false,"firmwareUpgradeInProgress":false,"remoteStartStatus":{"status":"ENGINE_STOPPED","duration":0,"timeStamp":"05-14-2024 12:26:27"},"chargingStatus":{"value":"NotReady","timeStamp":"05-26-2024 17:23:40","chargeStartTime":"05-26-2024 12:31:00","chargeEndTime":"05-26-2024 15:34:00"},"plugStatus":{"value":false,"timeStamp":"05-26-2024 17:23:40"},"ignitionStatus":{"value":"OFF","timeStamp":"05-26-2024 17:53:27"},"doorStatus":[{"vehicleDoor":"REAR_LEFT","value":"CLOSED","vehicleOccupantRole":"PASSENGER","timeStamp":"05-26-2024 17:30:27"},{"vehicleDoor":"UNSPECIFIED_FRONT","value" [... cut ...] atus":{"value":"NOTSET","timeStamp":"05-26-2024 17:23:29"}},"vehicleLocation":{"speed":0.0,"direction":"EAST","timeStamp":"05-26-2024 17:23:40","longitude":"8.72465","latitude":"50.243808"},"vehicleCapabilities":{"remoteLock":"Available","remoteUnlock":"Available","remoteStart":"Available","remoteStop":"Available","boundaryAlerts":"Unavailable"}}}
Soc:           94%
Capacity:      91.0kWh
Charge status: 
Range:         409km
Odometer:      39188km
Position:      x.xxx,y.yyy
Features:      []
```

Chris591, 2024-05-27:
"plugStatus":{"value":false,"timeStamp":"05-26-2024 17:23:40"}

Chris591, 2024-05-27:
wie gesagt ist auch nicht verbunden

andig, 2024-05-27:
Fixed in master

Chris591, 2024-05-27:
weiß nicht ob das mit Ford oder der UI Konfiguration zutun hat.
[image]
Hier hab ich ihn Mach-E genannt.
[image]
Hier wird aber der Nickname angezeigt.

Chris591, 2024-05-28:
leider hab ich im log nichts mehr sehen können - aber er war offline und hat geladen - als ich in die Konfiguration gegangen bin hat er nach kurzer alles angezeigt vom Auto aber am LP stand er immer noch als Offline und hat keinen SoC angezeigt.
nach einem Neustart von EVCC lief alles wieder.

Chris591, 2024-05-28:
> Fixed in master

confirmed - Status wird jetzt als "A" angezeigt wenn nicht verbunden.

Chris591, 2024-05-29:
der Error ist schon wieder da.
`[main  ] ERROR 2024/05/29 10:00:17 creating vehicle db:1 failed: cannot create vehicle type 'template': cannot create vehicle type 'ford-connect': cannot get vehicles: unexpected status: 429 (Too Many Requests)`
das ist die Zeit wo der docker container ein update über Watchtower bekommt.
restart der containers oder über die UI löst das Problem. Vielleicht hat hier einer eine Idee woran das liegen könnte.

djrepose, 2024-05-29:
wie kann ich den genau die tokens generieren das erste mal? irgendwie geht das bei mir nicht egal was ich bei evcc token danach eingebe?

Wondermusic, 2024-05-30:
Hallo zusammen,

ich weiß nicht, ob das direkt hier hin gehört, aber wie komme ich an die Daten, die ich eintragen muss?
Registriert bin ich bei "developer.ford.com/", aber wo bekomme ich die Client ID & Secret sowie die Tokens her?

andig, 2024-05-30:
Es gehört nicht hier hin. Mach bitte eine Diskussion auf.

Max112233445566, 2024-05-30:
> weiß nicht ob das mit Ford oder der UI Konfiguration zutun hat. ![Bild 27 05 24 um 22 27](https://private-user-images.githubusercontent.com/39920986/334206583-42c52119-cccf-4471-9709-f2bf0db10004.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcwNjY1ODAsIm5iZiI6MTcxNzA2NjI4MCwicGF0aCI6Ii8zOTkyMDk4Ni8zMzQyMDY1ODMtNDJjNTIxMTktY2NjZi00NDcxLTk3MDktZjJiZjBkYjEwMDA0LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUzMFQxMDUxMjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02ZjAyNTQ2OWRmMGU2ZWE0YWRkYmM5NzExZmRlNmFjNDFjNWVhYjBkY2QyYmYxYjY2MDZjYmUzMjE2ZjhlMGZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.iITDP1cyq4FGAeQJXtXq1W29AR42cQKkrsunnN8ZDfA) Hier hab ich ihn Mach-E genannt. ![Bild 27 05 24 um 22 27](https://private-user-images.githubusercontent.com/39920986/334206654-d6039903-a231-434e-8fe9-f4a37eab89b5.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcwNjY1ODAsIm5iZiI6MTcxNzA2NjI4MCwicGF0aCI6Ii8zOTkyMDk4Ni8zMzQyMDY2NTQtZDYwMzk5MDMtYTIzMS00MzRlLThmZTktZjRhMzdlYWI4OWI1LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUzMFQxMDUxMjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YTVmZDlkMmM5MWU0ZjFjMDY4NGE5NWEzMDhiZDQ1YmE1YjE0NDMxYjRiMTMzMTlhOTliNjgyZDNlOGNhOTY4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.bs2I0S_qIlF5Scnamox_bIRrpkrGhMM6E-3dT2CJa4g) Hier wird aber der Nickname angezeigt.

Ist bei mir auch so, EVCC nimmt ihr die Bezeichnung aus der Ford API und nicht aus der evcc.yaml.

{
    "status": "SUCCESS",
    "vehicle": {
        "vehicleId": "ff8080818f9b12b5018f9ececa1c0021",
        "make": "F",
        "modelName": "Kuga - TD (CX482 EU)",
        "modelYear": "2020",
        "color": "Frozen White",
        **"nickName": "Dennis",**

andig, 2024-05-30:
Das wäre ein Bug falls in der yaml ein anderer Title vergeben wäre

Max112233445566, 2024-05-30:
@andig ja, es ist ein Bug, Title ist ungleich nickName

miniluma, 2024-06-03:
Wahrscheinlich bin ich zu doof: Ich habe alle bekannten Infos im yaml eingetragen und erhalte das:

```
evcc token ev5 -c ./evcc.yaml -l debug
[main  ] INFO 2024/06/03 12:40:46 evcc 0.126.5
[main  ] INFO 2024/06/03 12:40:46 using config file: ./evcc.yaml
[main  ] FATAL 2024/06/03 12:40:46 1 error(s) decoding:

* '' has invalid keys: accesstoken, clientid, clientsecret, template

```

Eine Idee, was ich falsch mache? Mein Ziel ist es, den RefreshToken zu erzeugen.

hackfred64, 2024-06-03:
ich grabe mich auch gerade durch...
Installiert habe ich 0.126.5
Bei Ford habe ich mich registriert, clientid und clientsecret habe ich also.
Sobald ich eine Konfiguration mit "evcc configure" zu erstellen versuche komme ich bis zur Eingabe des access Token und des refresh Token. Die Hilfe verweist mich an der Stelle auf "https://docs.evcc.io/docs/devices/vehicles#ford-connect"
Damit komme ich aber nicht weiter (ist noch nicht implementiert). Kommen die beiden Token von Ford oder evcc? Mit "evcc token" im cli komme ich nicht weiter, da lande ich auch bei "has invalid keys"...
Ich habe gesehen für Tesla gibt es eine Seite tesla.evcc.io wo diese beiden Token generiert werden was mir aber maximal als Hinweis darauf dient, dass die beiden Token auch für Ford durch evcc erstellt werden.

andig, 2024-06-03:
Nightly nehmen…

miniluma, 2024-06-03:
> Nightly nehmen…

Dann kommt `[main  ] FATAL 2024/06/03 16:17:05 oauth2: "invalid_grant" "AAD...: The provided JWE is not a valid 5 segment token.\r\nCorrelation ID: c3dff...\r\nTimestamp: 2024-06-03 14:17:05Z\r\n"`

andig, 2024-06-03:
Dann bitte neues Issue mit trace log. Das hat- mit Verlaub- alles nichts mit den weiteren Themen hier zu tun!

hackfred64, 2024-06-03:
> Nightly nehmen…

covered already by  #13949
thks!

hackfred64, 2024-06-03:
as @andig closed #13949 and I am not able to provide further feedback there:

`# docker exec -it evcc evcc token

[main  ] INFO 2024/06/03 18:43:22 evcc 0.126.5 (9ea73de7)
[main  ] INFO 2024/06/03 18:43:22 using config file: /etc/evcc.yaml
? Please enter your client id: xxxxxxxxxx
? Please enter your client secret: xxxxxxxxxxxxxxx
? Please enter your authorization code: 
`
Again same questions as asked earlier today: where (and by whom and how) do I get this authorization code from?
Above is the output of "evcc token" using the nightly build. It may well be I am missing something because of lacking documentation still to come.

But at the minute I cannot confirm this has been closed!

Max112233445566, 2024-06-03:
@hackfred64 
It's similar to Postman, please take a look:
https://github.com/evcc-io/evcc/discussions/14117#discussioncomment-9608262

hackfred64, 2024-06-03:
Thx! So I need to take the workaround via Postman?!
I will give it a try later today and keep you posted!

hackfred64, 2024-06-03:
right now stuck with the https://fordconnect.cv.ford.com/ registration. Once having entered my Ford Pass credentials, then accepting my vehicle to be connected, I'm ending at a localhost:3000 port rejecting my access. Might be related to my Linux OS, needs further deep dive ...

hackfred64, 2024-06-03:
OK, at the end I managed my way through it! Thanks @Max112233445566, with his help I finally made it.
Definately not something everybody is able to work with.
So at the end I can confirm: this is closed! Just looking forward for helpful documentation, I'm happy to volunteer to help here ;-).

hackfred64, 2024-06-03:
where can I, or would have been able to, find any documentation on the `evcc token` command? Would be a first start for me ...
It doesn't help answering questions by simply repeating: use the command `evcc token` with a nightly build (and then closing the thread).

BTW: I'm just wondering if the release version 0.126.5 would not already have been working anyway.

andig, 2024-06-03:
Locking here due all the off-topic comments. Please use discussions.

closed 2025-01-06 by andig