evcc-io/evcc #14074: Wärmepumpe nicht als EVCharger im HEMS (SMA) darstellen

Tombra1889, 2024-05-26:
### Describe the bug

via SEMP wird auch bei einer Wärmepumpe folgendes übertragen:

```
<DeviceInfo>
<Identification>
<DeviceId>F-########-bbbbbbbbbbbc-00</DeviceId>
<DeviceName>Heat pump</DeviceName>
<DeviceType>EVCharger</DeviceType>
<DeviceSerial>############-#</DeviceSerial>
<DeviceVendor>github.com/evcc-io/evcc</DeviceVendor>
</Identification>
```

### Steps to reproduce

Gerät als "Heating" definieren.

### Configuration details

```yaml
features:
   - integrateddevice
   - heating
```

### Log details

```text
<DeviceInfo>
<Identification>
<DeviceId>F-########-bbbbbbbbbbbc-00</DeviceId>
<DeviceName>Heat pump</DeviceName>
<DeviceType>EVCharger</DeviceType>
<DeviceSerial>############-#</DeviceSerial>
<DeviceVendor>github.com/evcc-io/evcc</DeviceVendor>
</Identification>
```

### What type of operating system are you running?

Linux

### Version

0.126.4

andig, 2024-05-26:
Was wäre denn korrekt?

Tombra1889, 2024-05-26:
Wenn ich es richtig recherchiert habe gibt es folgende:
```
AirConditioning
Charger
DishWasher
Dryer
ElectricVehicle
EVCharger
Freezer
Fridge
Heater
HeatPump
Motor
Pump
WashingMachine
Other
```

andig, 2024-05-26:
Such Dir was aus- erstmal lässt sich das nur aus den Features ableiten.

Tombra1889, 2024-05-27:
Bei mir würde HeatPump gut passen!

andig, 2024-05-28:
Sieht aus als wäre das leider nixht so einfach, siehe PR. Daher derzeit wontfix.

closed 2024-05-28 by andig