evcc-io/evcc #14078: Lädt nicht mehr mit voller Leistung

micw, 2024-05-27:
### Describe the bug

Moin,

ich habe am Wochenende auf die aktuelle Nightly gewechselt, damit die Ladeanzeige für Skoda wieder geht. Heute fällt mir auf, dass trotz Überschuss nicht mehr mit der verfügbaren Leistung geladen wird.

### Steps to reproduce

1. Laden mit PV
2. reichlich Überschuss
...

### Configuration details

```yaml
log: debug

        interval: 10s

        site:
          - title: myhome
            meters:
              grid: grid
              pv:
              - pvWest1
              - pvWest2
              - pvOst
              - pvGarage
              battery:
              - battery1
            residualPower: 100

        mqtt:
          broker: 192.168.1.3:1883
          topic: evcc
          user: sensor
          password: ****
          topic: evcc

        meters:
          - name: grid
            type: custom
            power:
              source: mqtt
              topic: gridmeter/sensor/powertotal/state
              timeout: 30s
            energy:
              source: mqtt
              topic: gridmeter/sensor/energyimporttotal/state
              timeout: 600s
          - name: pvWest1
            type: custom
            power:
              source: mqtt
              topic: solar/116182009632/0/power
              timeout: 30s
            energy:
              source: mqtt
              topic: solar/116182009632/0/yieldtotal
              timeout: 3600s
          - name: pvWest2
            type: custom
            power:
              source: mqtt
              topic: solar/114183090265/0/power
              timeout: 30s
            energy:
              source: mqtt
              topic: solar/114183090265/0/yieldtotal
              timeout: 3600s
          - name: pvOst
            type: custom
            power:
              source: mqtt
              topic: solar/pvOst/power
              timeout: 30s
            energy:
              source: mqtt
              topic: solar/pvOst/yieldtotal
              timeout: 3600s
          - name: pvGarage
            type: custom
            power:
              source: mqtt
              topic: solar/pvGarage/power
              timeout: 30s
            energy:
              source: mqtt
              topic: solar/pvGarage/yieldtotal
              timeout: 3600s
          - name: battery1
            type: template
            template: victron-energy
            usage: battery
            host: 192.168.3.200
            port: 502

        vehicles:
          - name: Leaf
            title: Leaf
            type: template
            template: nissan
            user: ****
            password: ****
            vin: ****
            capacity: 40
            phases: 1

          - name: Enyaq
            title: Enyaq
            type: template
            template: enyaq
            user: ****
            password: ****
            capacity: 80
            phases: 3

        chargers:
          - name: Garage
            type: template
            template: fronius-wattpilot
            host: 192.168.1.164
            password: ****

          - name: Carport
            type: template
            template: fronius-wattpilot
            host: 192.168.1.165
            password: ****

        loadpoints:
          - title: Garage
            charger: Garage
            vehicle: Leaf
            mode: pv
            maxCurrent: 16
            phases: 0
          - title: Carport
            charger: Carport
            vehicle: Enyaq
            mode: pv
            maxCurrent: 16
            phases: 0
```

### Log details

```text
[site  ] DEBUG 2024/05/27 13:23:57 ----
[lp-1  ] DEBUG 2024/05/27 13:23:57 charge power: 0W
[lp-1  ] DEBUG 2024/05/27 13:23:57 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/05/27 13:23:57 charge power: 3660W
[lp-2  ] DEBUG 2024/05/27 13:23:57 charge currents: [5 4.9 4.9]A
[site  ] DEBUG 2024/05/27 13:23:57 pv power: 8639W
[site  ] DEBUG 2024/05/27 13:23:57 battery soc: 95%
[site  ] DEBUG 2024/05/27 13:23:57 battery power: -2089W
[site  ] DEBUG 2024/05/27 13:23:57 grid meter: -2578W
[site  ] DEBUG 2024/05/27 13:23:57 site power: -4567W
[lp-2  ] DEBUG 2024/05/27 13:23:57 detected active phases: 3p
[lp-2  ] DEBUG 2024/05/27 13:23:57 !! session: chargeRater.chargedEnergy=9.1 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/05/27 13:23:57 charger status: C
[lp-2  ] DEBUG 2024/05/27 13:23:57 soc estimated: 72.55% (vehicle: 72.00%)
[lp-2  ] DEBUG 2024/05/27 13:23:57 vehicle soc: 73%
[lp-2  ] DEBUG 2024/05/27 13:23:57 vehicle soc limit: 100%
[lp-2  ] DEBUG 2024/05/27 13:23:57 vehicle range: 328km
[lp-2  ] DEBUG 2024/05/27 13:23:57 pv charge current: 13.6A = 7A + 6.62A (-4567W @ 3p)
[lp-2  ] DEBUG 2024/05/27 13:23:57 max charge current: 13A
[site  ] DEBUG 2024/05/27 13:24:07 ----
[lp-1  ] DEBUG 2024/05/27 13:24:07 charge power: 0W
[lp-1  ] DEBUG 2024/05/27 13:24:07 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/05/27 13:24:07 charge power: 3710W
[lp-2  ] DEBUG 2024/05/27 13:24:07 charge currents: [5.1 5 4.9]A
```

### What type of operating system are you running?

Linux

### Version

0.126.4 (fbf59d12)

micw, 2024-05-27:
Hier noch ein Screenshot:
[image]

micw, 2024-05-27:
Achso, Wechsel von 1/3 phasig wird gemacht aber es lädt immer jeweils am unteren Ende der Leistung.

andig, 2024-05-27:
Die 13A werden ja frei gegeben, dein Auto nimmt trotzdem nur 5A. Vmtl. Limit im Auto.

closed 2024-05-27 by andig

micw, 2024-05-27:
Danke für den Tipp. In der Wallbox sehe ich auch gerade 11A. Scheint wirklich ein Problem am Auto zu sein.

micw, 2024-05-27:
Lag offenbar am Ladestecker. Raus/Rein, lädt wieder volle Leistung.