evcc-io/evcc #14080: 2 Phasiges Laden

dalle21, 2024-05-27:
Es wäre schön, wenn man am Ladepunkt auch 2 Phasiges Laden Auswählen könnte. So kann der Überschuss auch bei hohen Leistungen besser verteilt werden, ohne hohe Einphasige Ströme und damit Schieflast zu erzeugen. 
Wenn ich meinen Tesla 2 Phasig lade kommt EVCC nicht mit den Leistungen bei der Stromregelung zurecht und produziert Fehler

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

closed 2024-05-27 by andig

dalle21, 2024-05-27:
>   Das ist physikalisch nicht möglich. Du kannst aber das Fzg entsprechend konfigurieren falls es nur 2 Phasen verwendet.

Hallo Andig,Warum soll das Phisikalisch nicht möglich sein? Ich lade regelmäßig 2Phasig. 2 Phasig mit 16A entsprechen auch 7,4KW. Ich habe mal ein Bild angehangen.Gruß Danilo--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

andig, 2024-05-27:
Dein Auto lädt 2p. Deine WB gibt 3p frei.

dalle21, 2024-05-27:
Ja, das ist so. Die Wallbox gibt immer alle drei frei. Aber ich kann über die Zuleitung ja bestimmen, ob 1, 2 oder alle drei Phasen an der Wallbox ankommen.So mache ich das ja auch. Wenn es schnell gehen soll 3Phasig und bei Überschuss 1 Phasig.Nun wollte ich das ganze mal 2Phasig machen. Was beim Model 3 auch funktioniert um die Schieflast zu vermeiden.Das funktioniert auch, aber leider werden in EVCC dann Fehler produziert, weil am Ladepunkt nur 1 oder 3Phasig ausgewählt werden kann.--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

dalle21, 2024-05-27:
Hier nochmal ein Foto der WB. Dort ist zu sehen, daß die Sicherung für L2 nicht eingeschaltet ist.--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

premultiply, 2024-05-27:
Das ist nicht mit dem Standard und auch mit vielen Fahrzeugen nicht vereinbar.
Bei einigen Fahrzeugen produziert ein solche Vorgehen ganz schnell Defekte mit 4- bis 5-stelligen Reparaturkosten.

Bloß nicht so machen.
Wer hat denn da überhaupt Einzelsicherungen verbaut?
Es sind dabei immer verkoppelte Automaten für alle Phasen zu verwenden.

Die zwei Phasen musst du am Fahrzeug definieren und sie sind nur für die interne Berechnung relevant.
Es scheint allerdings dass du kein zweiphasiges sondern ein dreiphasiges Fahrzeug hast.

dalle21, 2024-05-27:
Ok, wenn das nicht so vorgesehen ist, lassen wir das.Ich kann aber mein Model 3 mir 1, 2 oder 3 Phasen laden.Bei nur einer Phase kann er dann 32A. Bei 2 oder 3 dann 16A.--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.